theyosh/TerrariumPI #242: XU4 Support

xu4user, 2019-01-21:
I have an XU4. I would like to try to install the software and get it going, but what other hardware do I need in order to control, temp, lights, misting, humidity, etc. I have a BioPod that the motherboard went out on and wanted to see if I could get this up and going and not have to deal with the non-support from the company. Can anyone provide a step by step on how to hook up the other hardware needed? Thank you.

theyosh, 2019-01-21:
@xu4user Hi, out of the box it will unfortunately NOT work. My software depends on certain Raspberry pi packages and modules for python.

Long answer:
I see that is supports Ubuntu/debian. So that means there is apt and .deb packages. Also python is available. So that should/could work.

But then the python code needs some GPIO modules which are Raspberry PI only. And with your XU4 it will not work. It has GPIO support, so that is nice, and that should work. But again, where is the python library different. I do not support WiringPI. Which is used on the XU4.

Then, there is also a possibility to work without GPIO, and use USB or Wifi power switches and bluetooth sensors. But then, you need to have at least network and bluetooth support on the board which is not the default.

So, I think I should be able to get it running with my technical background, but that will take a lot of time. Also supporting multiple hardware boards is rather difficult for one man.

I have seen the BioPod website, but it is hard to figure out how they control the hardware and if that is somehow compatible with my software. But that is hard to judge.

So, if you need a step by step manual to get it to work with a XU4 and a BioPod, I cannot help you. Both are unknown hard systems for me. Sorry.

xu4user, 2019-01-21:
Do you have step by step for the pi then? All hardware and necessary steps?

stuartabrown, 2019-01-22:
@xu4user check the hardware guide at https://github.com/theyosh/TerrariumPI/wiki/Hardware.  Otherwise I search the issues queue for what I'm trying to do and work through how other people have set things up.

theyosh, 2019-01-27:
@xu4user for now there is no such thing. There are just to much variants possible. So for now I would like to close this issue and you could follow @stuartabrown his medium posts https://medium.com/@jinky32

theyosh, 2019-02-02:
For now I am closing this now, as I cannot support as in place replacement. And tutorials and info can be found in the wiki

closed 2019-02-02 by theyosh