theyosh/TerrariumPI #243: HTTPS

stuartabrown, 2019-01-22:
Hi.

I've noticed that https://terrariumpi.theyosh.nl/ is showing errors in browsers now that they insist on https, is your cert expired?  

I'd like to add https to access my home setup over the internet, hopefully via LetsEncrypt / certbot and last night went through https://pimylifeup.com/raspberry-pi-ssl-lets-encrypt/ but got into trouble around section 8b and ended up with Apache taking over and I couldn't access TerrariumPI even locally (just go the standard 'It works!' page from Apache.

DO you know how we could use LetsEncypt to provide certs for TerrariumPI front end?

theyosh, 2019-01-22:
Euh, nope, domain is incorrect. it is https://terrarium.theyosh.nl

My webserver accepts everything before .theyosh.nl so, https://I_am_a_monkey.theyosh.nl give the same error ;P

SSL on the Pi is not handy, and is hard to do. Because your Pi is not directly connected with your internet. And that involves port forwarding in routers. And that is beyond this project. 

My software will by default run on port number :8090. So http://[raspberry_pi]:8090 should always work. Then I have some information about running [NGINX on port 80](https://github.com/theyosh/TerrariumPI/wiki/FAQ#use-and-enable-nginx-server-on-port-80). Apache will break that. So, to revert that, you could try: `apt purge apache2` and reboot. That would fix it for now.

Then if you still want SSL, then my first question is your Pi already accessible from the internet? Did you add port forwarding in your router? Because that is mostly a problem for most people. So there is an alternative. Make sure you have [NGINX running](https://github.com/theyosh/TerrariumPI/wiki/FAQ#use-and-enable-nginx-server-on-port-80) and use [dataplicity](https://docs.dataplicity.com/docs/host-a-website-from-your-pi)

stuartabrown, 2019-01-22:
ah ok thanks.  I got the URL from the wiki home page https://github.com/theyosh/TerrariumPI/wiki on the right hand side.

I followed your Nginx on port 80 guide and that worked perfectly :)  Port forwarding on the router was fine, and I set an ANAME on a domain I own to point to my WAN IP (it's static) and was able to connect, it's just that I couldn't set up HTTPS.

theyosh, 2019-01-22:
Ah, ok thanks. Updated the wiki.

For now, using dataplicity is easier. You will only not have your own domain. Else I have to add some extra config to the nginx config file.

stuartabrown, 2019-01-22:
no problem, will do.

closed 2019-01-22 by stuartabrown