theyosh/TerrariumPI #244: Can't access via webinterface

HarcourtA, 2019-01-23:


theyosh, 2019-01-23:
Please try: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi
Without that I cannot help. Paste the screen information here.

HarcourtA, 2019-01-23:
That just gives me: pi@raspberrypi:~/TerrariumPI $ screen -r
There is no screen to be resumed.

theyosh, 2019-01-23:
Hmmm that is strange. Then could you try the following at the location /home/pi/TerrariumPI

`python terrariumPI.py`

That should then give more information

HarcourtA, 2019-01-29:
Apologies it has taken so long for me to reply- it's been a manic week.
I get the following feedback:

pi@raspberrypi:~ $ /home/pi/TerrariumPI
-bash: /home/pi/TerrariumPI: Is a directory
pi@raspberrypi:~ $ python terrariumPI.py
python: can't open file 'terrariumPI.py': [Errno 2] No such file or directory

Kargha, 2019-01-29:
@HarcourtA You have to type 

`cd /home/pi/TerrariumPI`

to go to the /home/pi/TerrariumPI folder, and then do the 

`python terrariumPI.py`

The reason you are getting the "No such file or directory" error is because you are not in the folder where the file is located.

HarcourtA, 2019-01-29:
Thank you very much Kargha- output as follows:

Traceback (most recent call last):
  File "terrariumPI.py", line 26, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 127, in __init__
    self.get_weather)
  File "/home/pi/TerrariumPI/terrariumWeather.py", line 425, in __new__
    return weather_source(source, temperature_indicator, windspeed_indicator, callback)
  File "/home/pi/TerrariumPI/terrariumWeather.py", line 40, in __init__
    self.set_source(source,True)
  File "/home/pi/TerrariumPI/terrariumWeather.py", line 189, in set_source
    self.update(refresh)
  File "/home/pi/TerrariumPI/terrariumWeather.py", line 137, in update
    self.callback(socket=True)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 598, in get_weather
    data = self.weather.get_data()
AttributeError: 'terrariumEngine' object has no attribute 'weather'

HarcourtA, 2019-01-29:
It's a 3 model B+
Oh no! Hope it can be sorted!!

theyosh, 2019-01-29:
@HarcourtA could you subscribe to #246 ? Because there I explained a temporary solution. You have both the same error....

HarcourtA, 2019-01-31:
I did a git pull and that has seemingly fixed my issue, just connecting up the energenie pi mote now

Thanks a lot for your help!!

theyosh, 2019-02-02:
Cool closing this issue

closed 2019-02-02 by theyosh