theyosh/TerrariumPI #245: Is it possible to access the webcam stream externally?

Swiftnesses, 2019-01-24:
Hi @theyosh 

I now have the coop fully integrated into my smart home, automatic door / still images and all statuses via webhook, thank you!

Is it possible to access the webcam live stream externally, I assume it might be?

stuartabrown, 2019-01-24:
@swiftnesses you can access your TerrariumPI most easily via https://www.dataplicity.com I think and then you can see webcam.

Swiftnesses, 2019-01-25:
Hi @stuartabrown my devices are already accessible via SSH and my own domain, I need access to a raw stream...

stuartabrown, 2019-01-25:
Ok cool. In that case I'd love to learn this too.

theyosh, 2019-01-26:
Euh, the raspi_live webcam is HLS and can just be watched from the internet. But what do you mean with RAW stream?

Swiftnesses, 2019-01-27:
Right now I'm streaming via: 

http://192.168.1.48:8090/webcam/a94df035fc91a158c1f9d7a1c769ca66/stream.m3u8

Is there another url I can use @theyosh

theyosh, 2019-01-27:
You should do the following: https://github.com/theyosh/TerrariumPI/wiki/FAQ#use-and-enable-nginx-server-on-port-80

You will need to make TerrariumPI accessible through the internet. Or if you have enough network knowledge, you can make your own portforwarding. That is the only way.

Swiftnesses, 2019-01-27:
I already have it setup via NGINX and my custom domain / SSL.

I wondered if this is the only stream that is accessible outside of the web ui:

http://192.168.1.48:8090/webcam/a94df035fc91a158c1f9d7a1c769ca66/stream.m3u8

stuartabrown, 2019-01-27:
@Swiftnesses can you access that resource if you replace your LAN IP with your WAN IP or domain name?  Have you port forwarded 8090 on your router?

theyosh, 2019-01-27:
@Swiftnesses Then replace `http://192.168.1.48:8090/` with your external domain. That is all. If that is not what you are looking for, then there is no such RAW stream you need.

Also, it is still not clear what you mean with RAW stream. The HLS is as RAW as it gets...

Swiftnesses, 2019-01-27:
How do I access the HLS stream externally via another app, is the HLS stream the m3u8 url?

Swiftnesses, 2019-01-27:
I might end up running another script to convert the m3u8 stream to rtsp, as the app I'm using works better with rtsp it seems...

theyosh, 2019-01-27:
The HLS stream is accessible at: http://[your_IP_or_domain]/webcam/a94df035fc91a158c1f9d7a1c769ca66/stream.m3u8  And that is the full url that a VLC player for example can play. If you have TerrariumPI working with portforwaring or dataplicity. 

You could try to create a RTSP stream of it, but I do not think that will work better due to more port forwaring issues. That is why HLS is developed.

The Rpi Cam can only be used by 1 program. And I use a RPI cam software that comes with the Pi OS. That is using the GPU and therefore no CPU. Downside is that the camera it self is always occupied and you should us the HLS stream as a source for further processing if needed/wanted.

closed 2019-01-27 by Swiftnesses