theyosh/TerrariumPI #246: Unable to load web interface after install

whipcreambeast, 2019-01-26:
Hey everyone,

Attempted install this on a fresh image of Scratch Lite, got an exception on the installation but it installed.
The application did not start after manually running ./start.sh, so I attempted the installation again on the same image.

After that was done, got the following screen error:

2019-01-26 12:47:26,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.0.1.49:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 26, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 127, in __init__
    self.get_weather)
  File "/home/pi/TerrariumPI/terrariumWeather.py", line 425, in __new__
    return weather_source(source, temperature_indicator, windspeed_indicator, callback)
  File "/home/pi/TerrariumPI/terrariumWeather.py", line 40, in __init__
    self.set_source(source,True)
  File "/home/pi/TerrariumPI/terrariumWeather.py", line 189, in set_source
    self.update(refresh)
  File "/home/pi/TerrariumPI/terrariumWeather.py", line 137, in update
    self.callback(socket=True)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 598, in get_weather
    data = self.weather.get_data()
AttributeError: 'terrariumEngine' object has no attribute 'weather'
2019-01-26 12:47:36,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2019-01-26 12:47:36,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 10 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!

Anyone had this error?

Cheers,
Sean

ese23, 2019-01-26:
I thing this is "weather error" edit settings.cfd
there is a line with URL. Change it, for example to Yosh original gecko home :-)
i use  **sudo nano settings.cfg** to edit file. In putty ssh session.

But im not specialist.
PS i have this error when i past bad URL of norvegian weather inst.
second try and working.

theyosh, 2019-01-26:
@ese23 what url did you use? Then I can have a look. And yes, if a wrong weather url is entered, it will not start. I am looking into it to see if I can change that

ese23, 2019-01-26:
I want do copy for you, but lost in settings mayhen sry :-/

theyosh, 2019-01-26:
@whipcreambeast did the installer had problems? If so, rerun the installer once more so that you will not have any problems. Then you can remove the settings.cfg so that the default weather data is loaded. That should start the software

whipcreambeast, 2019-01-27:
Yes the installer has had problems, I am unable to get it working without it running into issues installing.
Seem to run into an issue when running opencv-python-headless.
Is there any way that I can get a log from the installer to help with troubleshooting?

FYI, I have tried both Python 2 and 3 and having is same issue.

theyosh, 2019-01-27:
I have just installed a completly new version for both python versions. No issues at all. So I get the feeling it is something outside my software.

How big is your SD card? You need at least 4GB. And what kind of Pi is it?

So lets test now only python2. That saves some complexity. Run manual:

`sudo apt-get -y install libftdi1 screen git subversion watchdog build-essential i2c-tools pigpio owserver sqlite3 vlc-nox ffmpeg libfreetype6-dev libjpeg-dev libasound2-dev sispmctl lshw libffi-dev ntp libglib2.0-dev rng-tools libcblas3 libatlas3-base libjasper1 libgstreamer0.10-0 libgstreamer1.0-0 libilmbase12 libopenexr22 libgtk-3-0 libxml2-dev libxslt1-dev python-twisted python-pip python-dev python-mediainfodll python-smbus python-pil python-opencv python-numpy python-lxml`

This will install all needed packages. But I guess that they are all installed already. Then the next step is:

`sudo pip2 install python-dateutil rpi.gpio psutil picamera pigpio requests gpiozero gevent untangle uptime bottle bottle_websocket pylibftdi pyalsaaudio pyserial python-twitter python-pushover requests[socks] Adafruit_DHT Adafruit_SHT31 luma.oled bluepy pywemo pyownet emails`

This should install all the needed python2 modules. I guess the pip2 install is failing somehow and should produce an error which can be posted here.

whipcreambeast, 2019-01-28:
Only sd card I had laying around was a 64gb, so I am using that and it is a Pi 2.

Flashed a new image of scratch lite and manually loaded the dependent packages and ran the pip 2 installations. No errors to report on either of the installations. 
Re-ran the installation using python 2, no exceptions appeared (awesome).

Still getting the same error as initially reported, and I am unable to find a setting.cfg.
I did attempt to make a file to see if this helped, the error did not change.
I have also changed the default to where I live which is Brisbane, still no change.

Is there a way to get further debugging when starting the application?

ese23, 2019-01-28:
Just... how strong power source do y have? A?
Because i have similar problems. And only if i have atached monitor, i can see errors with low power warnings (at the end of install, ofcourse). Then i putt new stronger power source and woala... Just try to help :-)

theyosh, 2019-01-28:
Ah, power problem, that could also be an issue. I do not know what the minimum should be, but it looks like 1Amp at least: https://www.pidramble.com/wiki/benchmarks/power-consumption

But I advice to use a 2+Amp adapter as you will also hook up some sensors and use bluetooth and or wifi.

whipcreambeast, 2019-01-28:
Cheers theyosh and ese23 for trying to help with this.
It was being powered by my PC, so I did try with an external 2 Amp adapter, no luck unfortunately.
Checking the dmesg, could not see any recent low voltage messages.
I also had a look at other posts and tried booting the python app manually, still no luck.

At this stage the only thing that is connected is a USB Wifi Adapter.

theyosh, 2019-01-29:
@whipcreambeast strange question: Could you try to set the CPU in turbo mode with `raspi-config` ? 

I have no clue why you have this problem. I am testing on a Rpi3 so that could be the key to this..

theyosh, 2019-01-29:
Or, do a `git pull` that will check for this strange error, and will just trying to continue... but it is a bit hard to test for me right now. So could you try this. It should still produce an error message, but should continue to work...

whipcreambeast, 2019-01-29:
Ok I quickly enabled turbo mode then pulled the latest commit. Tested and yep working now, was also able to get my local weather as well.

Cheers @theyosh, I owe you a beer.

theyosh, 2019-01-29:
There are more people with this error.... But with different error. So I keep this open for now.

Do you see some warning during startup mentioning weather problems? Should be visible in the screen session.

whipcreambeast, 2019-01-30:
2 warnings were found, my current screen output is:

2019-01-30 18:35:56,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.0.1.49:8090 ...
Failed to execute management command 'le on' (code: 17, error: Invalid Index)
2019-01-30 18:36:05,251 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
2019-01-30 18:36:05,257 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2019-01-30 18:36:10,000 - INFO    - terrariumWebserver   - Running webserver at :::8090

theyosh, 2019-01-30:
Ok, this looks ok. Rpi2 does not have Bluetooth LE (as far as I know) and you do not use OWFS hardware.

But the strange thing is that there is no weather error. If you have some time to spare, could you try again without the turbo mode on the Pi. Then we know for sure it is the turbo option that is needed for the Rpi2

theyosh, 2019-01-30:
@whipcreambeast Could you have a look at #210 ? Would be nice if you can add some pictures

theyosh, 2019-02-02:
Closing as working and closed #244

closed 2019-02-02 by theyosh