theyosh/TerrariumPI #248: Different temperatures throughout the day

stuartabrown, 2019-02-04:
I have a few related questions on this so hopefully it's OK to group them under this issue.

1) With my current thermostat I have four separate heat settings depending on the time of the day e.g. at 09.00 it is 25C, at 12:00 it is 31C, at 16:00 it is 29C and at 19:00 it is 20C.  Is it possible to replicate this within  TerrariumPI?
2) I noticed that within the yr.no XML there are four temperature values give for a day, similar to https://github.com/theyosh/TerrariumPI/issues/166 might it be possible one day to have the temperatures set according to the forecacst in the XML (ok, we'd be a day ahead of reality but no problem)
3) Ideally I don't want my heating device to switch on/off when a temperature is reached, I want it to dim (e.g. 50% or whatever) when reaching the max temp.  Am I right in thinking that this is already possible as described https://github.com/theyosh/TerrariumPI/issues/106
4) How do I got about controlling lights etc via a PWM dimmer?  Does the dimmer attach to the device plug somehow or is it connected to a relay switch in some way?

Thanks!

stuartabrown, 2019-02-04:
Thanks @theyosh :)

> Yeah, #166 was more like an idea. But it is rather hard to make at the moment. Also, I think that it is a nice idea, but will not work. What if the humidity is that high for a few days, that the sprayer is spraying 20+ times a day. That is not good. So that is way I am not making that option.

There is no humidity value, so no problem ;-) 

```
<time from="2019-02-08T03:30:00" to="2019-02-08T09:30:00" period="1">
 <!--
 Valid from 2019-02-08T03:30:00 to 2019-02-08T09:30:00 
-->
<symbol number="3" numberEx="3" name="Partly cloudy" var="03d"/>
<precipitation value="0"/>
 <!--  Valid at 2019-02-08T03:30:00  -->
<windDirection deg="291.4" code="WNW" name="West-northwest"/>
<windSpeed mps="3.2" name="Light breeze"/>
<temperature unit="celsius" value="26"/>
<pressure unit="hPa" value="1005.9"/>
</time>
```

(I think `precipitation` is just amount of rain, not humidity).  If you have configurable values for day and night could you theoretically just just have two additional (optional) values that you can choose to pre-populate with the weather location/times you have chosen (00:30:00 06:30:00  12:30:00  18:30:00 in the XML) so 18:30:00 - 00:30:00 is night, 06:30:00 - 12:30:00 is morning (temp increasing incrementally from value at 06:30:00 to value at 12:30:00 - temp difference divided by number of hours difference = temp change per hour) and 12:30:00 -   18:30:00 is afternoon.

Thanks for the PWM steer, I'll take a look.  I'm waiting for camera and UV sensors to arrive from China and it's driving me crazy not being able to do anything - trying to learn Python so maybe one day I can contribute something useful!

theyosh, 2019-02-09:
Ok for now I put your first request on the wish list:

> 1. With my current thermostat I have four separate heat settings depending on the time of the day e.g. at 09.00 it is 25C, at 12:00 it is 31C, at 16:00 it is 29C and at 19:00 it is 20C.  Is it possible to replicate this within  TerrariumPI?

Not sure how I will make it, but I think it is a neat feature to have. The list is growing though, and not sure when this is possible.

stuartabrown, 2019-02-09:
thanks @theyosh .  I know you are not looking at this functionality but at https://gist.github.com/stuartabrown/718eaee2813c4296bed11cd290da1d49 is my attempt at what I mean by grabbing the values of tomorrow's weather forecast. The values in the `tomorrowTemps` array would be the values that populate the the four separate heat settings described above.

theyosh, 2021-12-10:
Closing this issue, as with the new [release 4.1.0](https://github.com/theyosh/TerrariumPI/releases/tag/4.1.0) this is now possible.

closed 2021-12-10 by theyosh