theyosh/TerrariumPI #249: Watertank additional option

Natrixz, 2019-02-04:
Hey Yosh,

Can we perhaps get an extra option at the "Watertank" where we can provide a maximum temperature for the water and the sensor that measures it?
That way we can not only use the distance sensor to kill or enable switches, but also use the temperature to kill power to any heating elements.
Just in case it fails on....

Would be nice if it's possible! =)

theyosh, 2019-02-05:
In other words: the heating in the watertank should stop when there is not enough water and could harm the heater.... Yup, sounds reasonable

Natrixz, 2019-02-05:
Well yes that option would be very nice as well =)
Mine can withstand that, however I have seen at a friend of mine who has alot of aquariums that sometimes when a heater breaks it will fail "ON"
Meaning the heater will keep running and running and running.

So what I mean is that if the water reaches X temp, turn "off" switch Y

Hope this makes more sense ;)

theyosh, 2019-02-05:
Ah, no, that will be done in the future by creating multiple heaters. A heater inside for the animal, and one for the water tank. So you then can make sure it will never get hotter than X degrees. 

But then it needs to work with the water tank level, as a second check somehow... So the logic gets a bit more complicated..

Natrixz, 2019-02-05:
Alright =)

Well I wouldn't mind if it would start with just the temperature and maybe later add the logic for the water level as well.
Altho I do understand that it does make it a bit more complex....

theyosh, 2019-02-05:
What I have done is adding a 1Wire temp shielded sensor in the water tank for temperature measurement and added a aquarium heater to the switches which operated with the light environment part. So when the lights are on, the water tank gets heated. But this will not kill the power when it get to hot. But you could set an warning through the notification system that a sensor is to high.

But this had influence on my ultrasonic distance sensor as condensation is occurring and harming the electronics. So I do not heat it at the moment. And a cold rain shower of 20 sec will not kill my gecko :)

Natrixz, 2019-02-05:
Yeh I have done sort of the same :)
So for now I will just receive alot of emails if something go's wrong, that does signal me so I can kill the power (hopefully) in time.

Curious to see if I will be getting condensation as well!
It's currently set up on two of my terrarium's, one of wich needs the warm water to keep his defecation cycle up (sadly that's how they do in nature as well)

Ps. Thanks for the addition Andrej, however I would like to be able to remotely switch things off in case I need it.

theyosh, 2021-09-01:
Hi Natrixz,

with the new TerrariumPI 4 it is possible to have multiple temperature zones in 1 enclosure. So, now you can just create a new temperature area, which has a sensor in the water tank, and a relay for the heater in the tank. And then you can control water temperature.

So I think this is now possible :)

theyosh, 2021-12-06:
Another update on this. Look at https://github.com/theyosh/TerrariumPI/issues/630#issuecomment-986870534 . You can now have dependencies between areas. So I think you should now be able to make the setup you want. Are you able to test this?

theyosh, 2021-12-10:
Closing this issue, as with the new [release 4.1.0](https://github.com/theyosh/TerrariumPI/releases/tag/4.1.0) this is now possible.

closed 2021-12-10 by theyosh