theyosh/TerrariumPI #250: Cant access web dashboard

kronnofatal1ty, 2019-02-06:
Hi, i'm new to this projects and i followed your tutorial step by step and when i'm trying to access the web dashboard always keep saying connection refused and i'm kinda stuck here.
Can someone say what RPI version your using and which linux distro?
I'm frying my head for 3 days and can't get any answers. Thank you and btw great project ;)

kronnofatal1ty, 2019-02-06:
[image]
[image]

kronnofatal1ty, 2019-02-06:
im using Raspbian Strecht too. I've tested with RPI 3 and 3B+ same result as i posted

theyosh, 2019-02-06:
Please rerun the installer. The python module emails is missing, so the installer did not run fully.

Kargha, 2019-02-06:
That's really odd. I mean, it's clearly missing the python email module... Are you using regular Stretch or Stretch Lite? Apparently the regular stretch (with desktop and recommended software) doesn't seem to work.

kronnofatal1ty, 2019-02-06:
The oddest thing is during the installation. When trying to install the required software (with stretch lite and desktop version) the errors are the same. ssl errors and no matching distribution found errors.
I know that he keeps trying till the modules are installed. Im going to try to install manually the packages to check if it makes any difference. And going to portforwarding on my router.

kronnofatal1ty, 2019-02-06:
[image]

kronnofatal1ty, 2019-02-06:
with the stretch lite is the same errors

theyosh, 2019-02-06:
Hmmm it is a bit small, but I can see something like that python 2.7 is somewhere depricated.

So could you try the following: `sudo ./install.sh 3` the 3 will install it as python 3. But, then you also need to install the software either manual with: `sudo ./start.sh 3` or change the `/etc/rc.local` file by adding the 3 to the startup command.

kronnofatal1ty, 2019-02-06:
i did what you said with the command ./install.sh 3
Everything was running smoothly till the last 3 packages.
Again SSL errors

kronnofatal1ty, 2019-02-06:
Bad handshake: SysCallError(104, 'ECONNRESET')

kronnofatal1ty, 2019-02-06:
Im going to test that with strecht lite. ill keep u up

kronnofatal1ty, 2019-02-06:
[image]

Now i'm dealing with this. Any help?

theyosh, 2019-02-06:
Yes, this is somewhat a known issue: https://github.com/theyosh/TerrariumPI/wiki/Known-issues#python3-numpy-import-error

In short run: `sudo pip3 uninstall -y numpy`

But the installer should take care of that....

kronnofatal1ty, 2019-02-06:
When you solve a problem more appear lol
ImportError: cannot import name 'PiCamera'

kronnofatal1ty, 2019-02-06:
solved. just deleted the imports from camera

theyosh, 2019-02-06:
Again, you are missing modules. Please read: https://github.com/theyosh/TerrariumPI/issues/246#issuecomment-457920167

I can not fix this, as somehow the installation is failing. I have installed more then 100 times now, and it is always working. So the issue is somewhat outside my control

Kargha, 2019-02-06:
@theyosh Could it potentially be connected to lack of power? I know I've had problems with my PI sometimes when doing CPU and read/write intensive tasks if I haven't had a stable enough power source (it loved to turn off my wi-fi mid-downloads and such sometimes).

theyosh, 2019-02-06:
@Kargha I do not have enough knowledge about that. It sounds reasonable that heavy CPU usage will use more power. And the installation of the python packages is all compile work. So that could be an issue.

But at the same time there are lot of people having issues due to a 'broken' SD card. And swapping to a new SD card fix the issues. So I am more tending to a broken SD card than a power issue. Even when there are not yet SD card errors in the OS.

theyosh, 2019-02-09:
@kronnofatal1ty is your issue fixed? Then this issue can be closed?

closed 2019-02-16 by theyosh