theyosh/TerrariumPI #251: Exclude sensor from average calculation

jornobe, 2019-02-07:
Is it possible to somehow exclude sensors from the average calculated values shown on the dashboard. In my case there is one in the pool which is not relevant to the average temperature calculation.

theyosh, 2019-02-08:
Technically this can be made. But I am struggling with the interface. The sensor settings page is rather full already. So adding more settings, will not fit well.

So for now I am thinking of that this has to be setup outside my web interface. So it has to be a manual config change. I will make this first, as this is the easiest to do right now. And add a FAQ item explaining how to setup

AndrejWeber, 2019-02-08:
That sounds great!
But then it should also not shown on display.
For me: I have a DHT22 near the raspi, just to know the temperature of electronics cabinet. In the moment it is calculated to average of terrarium.

theyosh, 2019-02-08:
Yesh! Will add an badge to the graph that is excluded...

theyosh, 2019-02-09:
Ok, just added new code that make this possible... So here are the install instructions.

- `git pull` (on the master branch)
- `sudo ./install.sh` ( add 3 if running python 3) (New oled library is used, so new python modules needs to be installed)
- In the settings file, you can add the following line to a sensor below `[sensor_SOME_ID]`
`exclude_avg = true`
- restart Terrariumpi
- clear browser cache and reload

This will add a blue badge above the graph of the sensor is excluded. It will not show a badge on the average graphs.

theyosh, 2019-02-09:
And here is the Wiki: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-exclude-a-sensor-from-average-calculation

jornobe, 2019-02-12:
Nice! Highly appreciated!

Natrixz, 2019-02-13:
I noticed the badged don't always show up on the sensor page, I excluded two (well now just one) sensor but it doesn't have the badge

theyosh, 2019-02-16:
For me it is working like expected. The badge is there when the sensor is excluded from average.

What does the following url shows at your system: `http://[pi]:8090/api/sensors` . There it should also say exclude_avg = true

Natrixz, 2019-02-16:
That does show the badge as it should be, actually it appears to be fixed after the latest git pull and restart =)
So ignore my observation

theyosh, 2019-02-16:
No problem.

stuartabrown, 2019-02-17:
Just a note to say that the `excl. avg` badge seems to be permentant for me too in the latest release:
[image]
Badge appears but config says it shouldn't because sensor isn't excluded
[image]

Raerion, 2019-02-18:
Nice feature, immediately started using it!
For me only one of three temperatures is really relevant and now I can see this temperature diectly on the dashboard.

_Some ideas:_
Maybe if you ever decide to rework the dashboard one could think about a configurable approach (enable disable items like water consumpiton/power consumption/environment pane/audio) and add sensors to groups which will be averaged (sensor settings like: `show_on_dash = True` and `average_group = Temperature1`). I'm thinking about cases where you have something like three air temperatures, two water temperatures and maybe a cabinet tempearure one could make three groups and have three dashboard displays for this groups.

theyosh, 2019-02-19:
> _Some ideas:_
> Maybe if you ever decide to rework the dashboard (like #260 ...?) one could think about a configurable approach (enable disable items like water consumpiton/power consumption/environment pane/audio) and add sensors to groups which will be averaged (sensor settings like: `show_on_dash = True` and `average_group = Temperature1`). I'm thinking about cases where you have something like three air temperatures, two water temperatures and maybe a cabinet tempearure one could make three groups and have three dashboard displays for this groups.

Nice idea, but I am a very bad UX designer... so I think this will not happen soon or maybe never.. To make it that flexible, is a lot of programming... So I am sorry, for now, do not get your hopes that up...

theyosh, 2019-02-19:
@stuartabrown Just to be sure. After manual changing settings, you have to restart. Did you do so? Settings file only be read during startup. When changing the settings in the UI, I will reload it in code. But this exclude setting is a manual setting change and requires a restart

theyosh, 2019-02-23:
Found some scenarios where the badge would not hide properly. That is fixed now

stuartabrown, 2019-02-24:
@theyosh have pulled the latest release.  Behaviour was aas per before but if i set `exclude_avg`  to `True`, restart, then set to `false` and restart everything is ok :)

theyosh, 2019-03-04:
Closing as it is working like expected

closed 2019-03-04 by theyosh