theyosh/TerrariumPI #252: Fahrenheit/Celsius issue

Janson777, 2019-02-08:
Big thanks for TerrariumPi.  Will upload pics when I get my system finished.  I am having an issue with 1-wire temp sensors reporting results in Celsius when the system is set to Fahrenheit (refer to sys_log_errors.jpg).  The first value highlighted is actually about 79F-80F.  Not sure if I am missing something or it is a bug.

[image]
[image]
[image]
[image]

theyosh, 2019-02-08:
Ah, this looks like a bug. But could you try one more thing? When you put it to Fahrenheit and save the system setting. Then restart the software. If that will give the right value, then I have a clue where to look

Janson777, 2019-02-08:
Went through the log-out and restart process.  The errors still appear to still be there.

[image]
[image]

theyosh, 2019-02-08:
Crap, found the problem.. I had this fixed before, but in a different branch. Forgot to merge. But the fix is already there.

So you can now do a `git pull` and it should work. If you have then new errors, then please rerun the installer `sudo ./install.sh`

theyosh, 2019-02-09:
So, I can confirm that the latest code has fixed this. The only thing is that you have to reload the hole page after changing the temperature indicator.

closed 2019-02-16 by theyosh