theyosh/TerrariumPI #253: Live counter vor switched devices

AndrejWeber, 2019-02-08:
What do you thing about a resettable counter for lifetime of Lamps and heaters etc?
It is interesting how often the lamp has switched on and how much hours it burns, before it fail. 
Also UV-lamps have only a limited lifetime, before the uv intensity will get worse.
A logging is already there, why not calculate the total lifetime like total energy usage.
It is like “just one more thing” from Steve Jobs ;-)

AndrejWeber, 2019-02-08:
Tust as Idea, we can set in setting for switches a date, when a lamp behind a switch was replaced. Then you can calculate from this date. If there is no date, no calculation, or from first start

theyosh, 2019-02-09:
Yes, I have the same idea for more then a year. But this will add a hole new logic part to the system. So that is way I did not made it.

Because this is somewhat like a agenda system. And there was someone how used his Google Calendar system and iframed it in the side. There he posted when to change the lights, and other dates to remember.

But you only need to know the max amount of hours you can/will use the light. That should be added to the settings. The option to set the replacement date will then be done through the settings menu option on the top right of the graph (wrench icon).

Then you  could make some calendar based on those dates to see when lights are changed in the past.

jornobe, 2019-02-12:
A fancy agenda sounds nice but a simple solution as proposed above could be enough. Maybe for now a timer that displays total running time of the lights and temperature environments and a way to reset those counters in the settings file.

Both total number of days and total number of hours would be nice to see.

theyosh, 2019-02-12:
@jornobe for powerswitches it is partly already possible. You can select with the calander icon a different period to show. For now only day,, week, month and year. So pick a year. Then next to the name it will show the total power usage in watt and duration on...  So there is already some information available...

But it is on the wish list for a later moment to make

theyosh, 2019-05-13:
Ok, after some thinking..... I have now a new branch called 'feature/calendar' which has the following new features:

- Under the wrench icon an option for logging hardware replacement
- Under the kalender icon at the graph the new period 'last replacement' which will show the graph from the last replacement till now. Which will then give a total on time in hours next to the title.
- And when replacing hardware, option to get a reminder to replace again.
- Added a new calender at the calender menu item which shows the replacement actions and reminders.
- A new status indicator called 'calendar' which shown the next 7 days upcoming events.

This is a 'poor man' implementation. Else I have to rethink the interface completely to get all this information on screen.

And later on, I can add more features to the calendar.....