theyosh/TerrariumPI #254: Meross MSS425E Power Switches

RusuPetrut, 2019-02-09:
Hi guys ! What a fabulous job you've done here  !!!  I'm a newbie in Terrarium Pi and just play around until i understand it and after i will put this system on my aquarium . I'm excited to join the team !

I have a question/request if you don't mind , i bought before searching and finding Terrarium PI , two Power Switches "Meross MSS425E" , which i find them very handy with my current system(2 pumps , 1 heater, lights , air stone , co2 electrovalve) but i want to integrate them with Terrarium Pi . 

Those power switches are cheap (30E) and you can find them on amazon , here is a link : https://amzn.to/2SFvnTZ

I found that Meross integrates with Google Home and Alexa and somebody already try to integrate them and published a library : https://pypi.org/project/meross-iot/

Could you please provide a feedback about this ? 

Thank you for everything ! 
[image]
[image]
[image]
[image]
[image]

theyosh, 2019-02-09:
Hi,

I think that it is possible to integrate those power plugs. They act almost as WeMo power plugs. But I do not have the hardware at home, nor will I buy it (for personal reasons). So then I would ask you to do some testing and debugging for me.

If you have a Pi running, please install the python library:

`pip install meross_iot --upgrade`

Then create a `test.py` file with the following contents:
```
import time
import sys
from meross_iot.api import MerossHttpClient

if __name__=='__main__':
    httpHandler = MerossHttpClient(email="YOUR_MEROSS_CLOUD_EMAIL", password="YOUR_PASSWORD")

    # Retrieves the list of supported devices
    print("Listing Devices...")
    devices = httpHandler.list_supported_devices()
    print(devices)

    for counter, device in enumerate(devices):
        print("Playing with device: %d" % counter)
        print(device)
        # Returns most of the info about the power plug
        print("\nGetting system data...")
        data = device.get_sys_data()
        print(data)

        # Reads the historical device consumption
        print("\nReading consumption data...")
        consumption = device.get_power_consumptionX()
        print(consumption)

        # Info about the device
        print("\nGetting device trace...")
        trace = device.get_trace()
        print(trace)
        print("\nGetting device debug...")
        debug = device.get_debug()
        print(debug)

        # Returns the capabilities of this device
        print("\nRetrieving device abilities...")
        abilities = device.get_abilities()
        print(abilities)

        # Returns the current power consumption and voltage from the plug
        # (Note: this is not really realtime, but close enough)
        print("\nReading electricity...")
        electricity = device.get_electricity()
        print(electricity)
```

You need to fill in your login details for the service. If you then run the file with python it should produce some output. I need that output to see what information is available to work with.

So run: `python test.py` and make either a screenshot or copy paste the output here. By copy pasting is formatting needed.

RusuPetrut, 2019-02-10:
Hi , 

I just run "python test.py" , here is the response from the  Meross MSS425E Power Strip : 

[image]

I also format the response and you can download from here : [https://we.tl/t-Et3RKnPC4c](url)

Thank you

theyosh, 2019-02-10:
Hmm, is see that your device is missing some code support. So therefore the above test does give an error. Could you try the following:

Edit the file `test.py` and remove the following lines:

```
# Reads the historical device consumption
print("\nReading consumption data...")
consumption = device.get_power_consumptionX()
print(consumption)
```

and rerun again: `python test.py`. It should produce more output.

RusuPetrut, 2019-02-10:
Thanks for your quick reply , here is the output : [https://gist.github.com/RusuPetrut/1c327ea14766baa824b4af270ec5af9f](url)

theyosh, 2019-02-10:
Yes! Ok, now I need some time to format that information correctly and see what I can do with it. It can take a few days before I have it working.

But thanks a lot for the information!

theyosh, 2019-02-10:
Is it correct that with the last test, all power plugs where off? And do you have 4 power plug connections and 1 for usb?

RusuPetrut, 2019-02-10:
Yes , all the plugs were off at test point , because before the script modification i just played with toggle to see what channel correspond to each plug , here is the conclusion : 
[image]

Only Ch1 , Ch2 , Ch3 are necessary for toggle.

theyosh, 2019-02-10:
But does the output repeat it then 5 times? Because I can only do an on or off at a device. And toggle all at once is not what you want..

RusuPetrut, 2019-02-10:
Yes , i repet the output test `test.py` 5 times, each time I toggle one plug with the Meross Aplication. If this is what you mean

theyosh, 2019-02-10:
No, I had the idea that a device is one plug and therefore the output should be repeated. But as it turns out, a device is just the hole power device with different channels. The used example and test script did not make that clear. So there is no repeat.

But that you see a change from 1 to 0 is good and does confirm that all info is available.

theyosh, 2019-02-11:
Ok, after some thinking I think I have a working way. But I could not test it. So I am asking you to test this for me. But it needs Python 3, so I hope you have some time to set it up for Python 3:

- `cd TerrariumPI`
- `git pull`
- `git checkout issue/254`
- `sudo ./install.sh 3` ( the value 3 will install everything for using Python 3 including the new Merros IoT module)
- make sure there is no TerrariumPI running. https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-stop-terrariumpi
- `sudo ./start.sh 3` ( again the 3 is telling using Python 3)
- **clear browser cache**
- Go to the System settings, and you should see a new part called 'Merros Cloud'. Enter here you login email address and password. Save it
- When the new settings are saved, you have to restart TerrariumPI in the screen session. Do not restart the Pi itself. https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-restart-terrariumpi
- When restarted, it should show at least 4 new power switches. If not there should be errors in your screen session. https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

For now this is the easiest way to add support for those power switches. The restart after setting login credentials is needed. But once added and the system has found your powerplugs, there is no restart needed anymore and everything should work.

Also, keep in mind, I do **NOT** encrypt password in the settings. So you Merros cloud password will be stored in plain text!

RusuPetrut, 2019-02-12:
[image]

Hi ! This is what i receive after i install everything for using Python 3 `sudo ./install.sh 3`  and the systems reboots

theyosh, 2019-02-12:
Ok, you found a bug. So I fixed that now. So execute the command in the TerrariumPI folder

`git pull`

And then restart it. It should at least go further then this error

RusuPetrut, 2019-02-12:
[image]

No "Merros Cloud" in system settings .

theyosh, 2019-02-12:
Did you clear your browser cache?

RusuPetrut, 2019-02-12:
Yes

theyosh, 2019-02-12:
And are you sure that you are on the branch 'issue/254' ? Because it should show a second row with title for the meross cloud settings.

RusuPetrut, 2019-02-12:
[image]
I receive this error when i want to go on branch , do you know wy ?

theyosh, 2019-02-12:
This is strange. You have done a `git pull` which should provide the latest information. Also this branch.

If I type in `git checkout issue/2` and then a Tab, it should give all the available branches. And there I see this one also. I have no idea why you do not have it

theyosh, 2019-02-12:
oh crap... I do know why you do not see it... My installation instructions tells you not to download multiple branches.... Crap crap...

So, then it needs some more work. First, move the current TerrariumPI to a temp location or rename it.

Then do the following command:

`git clone --recursive https://github.com/theyosh/TerrariumPI.git`

This will get all history and branches in the TerrariumPI folder. Then you should be able to go to the branch issue/254

theyosh, 2019-02-12:
But that also means, that the you have to rerun the installer as well. All you actions are done on the master branch. So that means after the new git clone start again at: https://github.com/theyosh/TerrariumPI/issues/254#issuecomment-462525080 at the 3rd dot

RusuPetrut, 2019-02-12:
I manage to install it , when i try to "submit" the switch i receive this error , the system don't save this setting .

[image]
[image]

theyosh, 2019-02-12:
Made again some fixes. I hope that this does not bother you...  that you have to try this a couple of times.

So do again `git pull` and restart. Then you should be able to save the settings. When that succeeds, restart once more to activate the Meross cloud

RusuPetrut, 2019-02-12:
No problem , glad to help you . The settings are the same from the previous post 
[image]

theyosh, 2019-02-12:
Cool. So next run: `git pull` and restart.

RusuPetrut, 2019-02-12:
Yes , the authentication is working . I receive the same error
[image]

theyosh, 2019-02-12:
yesh... I just found another bug.. Again `git pull` and restart :)

RusuPetrut, 2019-02-12:
[image]

theyosh, 2019-02-12:
Are you running this with Python 3? Because that is mandatory. So you should start it with `sudo ./start.sh 3` the 3 value is important here

RusuPetrut, 2019-02-12:
Yes, always `sudo ./start.sh 3`

theyosh, 2019-02-12:
Hmm, this is a bit hard. But I made a change, and added some more debug code. So again `git pull` and restart please.

theyosh, 2019-02-12:
And, you should not have to save the settings. Only restart it, and let it run...

RusuPetrut, 2019-02-12:
[image]

theyosh, 2019-02-12:
Ok, found a stupid typo. So, `git pull` and restart. And just restart. It looks like you are saving some settings? That should not be needed any more. So, for now, the goal is that it starts without Traceback errors.

RusuPetrut, 2019-02-12:
[image]

theyosh, 2019-02-12:
And yesh, found another typo.. :) `git pull` and restart

RusuPetrut, 2019-02-12:
Ok , now all power plugs after `sudo ./start.sh 3` just go off 

[image]

theyosh, 2019-02-12:
Ok , this one is a bit hard.. I do have an idea, but not sure. So I added some extra debug. Lets see what that says. `git pull` and restart please.

theyosh, 2019-02-12:
just found the bug... make sure you do an extra `git pull`

RusuPetrut, 2019-02-12:
Now it doesn't stop , just like reading hardware state for each channel all the time : 
[image]

RusuPetrut, 2019-02-12:
In the previous `git pull` i power on all the channels , after the systems starts it automatically stop each channel one by one

RusuPetrut, 2019-02-12:
[image]

theyosh, 2019-02-12:
Hmm, about changing the state I have no clue at the moment. But I add some debug code. As I do not know sure if it does repeat that much.

So could you try to count how often the text is passing by: 'Meross switch init'

`git pull` and restart

There is also a lot of debug logging from the meross library

RusuPetrut, 2019-02-12:
[image]

theyosh, 2019-02-12:
Ok, but as far I can see now, it is working. Only the changing the power switch does give an error.

But, for now, can you see the power switches on the web interface? And it is getting late here, so I have to stop for now. So I hope that you at least 4 power switches in the web interface right now

RusuPetrut, 2019-02-12:
Add new channel: 1
Meross switch init
(<meross_iot.supported_devices.power_plugs.Mss425e object at 0x6b9b3ab0>, 1)
Add new channel: 2
Meross switch init
(<meross_iot.supported_devices.power_plugs.Mss425e object at 0x6b9b3ab0>, 2)
Add new channel: 3
Meross switch init
(<meross_iot.supported_devices.power_plugs.Mss425e object at 0x6b9b3ab0>, 3)
Add new channel: 4
Meross switch init
(<meross_iot.supported_devices.power_plugs.Mss425e object at 0x6b9b3ab0>, 4)
DONE scanning

RusuPetrut, 2019-02-12:
Yes of course , thank you for for your support ! Have a good night !

theyosh, 2019-02-12:
Ah thanks for the update... this looks pretty cool. So you have the power switches on the web interface. 

Then we have to see what happens when you try to toggle a power switch. Just toggle one channel just once.... When I have time I can see what I can do

RusuPetrut, 2019-02-12:
All switches work on web interface. 💪

theyosh, 2019-02-12:
Whaaat!!! You can toggle them!!!! NICE!!!! Party time...!

RusuPetrut, 2019-02-12:
Great work !

theyosh, 2019-02-14:
So, after two days, is it still working?

RusuPetrut, 2019-02-14:
Hi , yesterday i've done a `git pull ` and now i receive this error : 

[image]

theyosh, 2019-02-14:
Is it started with python 2 by any chance?  And you are still on branch issue/254?

RusuPetrut, 2019-02-14:
Yes , i am in branch issue/254 and python 3

RusuPetrut, 2019-02-14:
Ok , now for no reason is working again 👍

theyosh, 2019-02-14:
Hmm, I have also changed some debug code for the excessive logging of the Meross module. Is there now less verbose debug in the screen? 

But that error is a bit strange. Could maybe triggered from somewhere else...

theyosh, 2019-02-14:
Did you already saved the power switches settings such as a name and power usage info? This can maybe interfere somehow with starting up

theyosh, 2019-02-16:
Hi, could you do one more test for me? I did optimize the code a bit. So if you can do a final `git pull` on branch `issue/254` and restart the software. I expect no problems. So if you can confirm that, I will merge this with the master branch so that is available for everybody.

RusuPetrut, 2019-02-16:
Hi , sorry for this late reply , i was going to write today with test results . So , i added one more power plug in Meross App and everything stop working . I've done `git pull ` now but the system restart over and over itself 

[image]

[11 more replies]

commit 88592d9d (2019-02-12): Fixes #254
closed 2019-02-23 by theyosh