theyosh/TerrariumPI #255: Issues communicating with piMote

HarcourtA, 2019-02-09:
Seem to have an issue communicating with the Energenie PiMote. With light on an 08:00-20:00 timer under the environment heading the light won't turn on and gives the following  reading in the log 
INFO    - terrariumSwitch      - Changed power switch 'Skylight' of type 'eg-pm-rf' at address '1' from state 'False' to state 'True' (Forced:False)

Any ideas? (Forced:False) seems odd to me

stuartabrown, 2019-02-09:
Can you turn the device on/off using the switch button in the TerrariumPI user interface?

HarcourtA, 2019-02-09:
Nope, it was working but seems to have stopped responding overnight...

HarcourtA, 2019-02-09:
N.b. all the switches are coming up with (Forced:False) in the log when I try to change them

theyosh, 2019-02-09:
I guess that you need to reboot your Pi. As I cannot see if the PiMote is working without errors due to the lack of code in the PiMote modules. So that means that my software is sending the power change commands to the PiMote board.

The message Forced: False is correct. It should only be True during startup to make sure the switch is in a certain state to start with.

So, I guess that your PiMote board is causing the issues, or that the power plugs itself. So try rebooting the Pi and see what happens.

HarcourtA, 2019-02-11:
I believe this is resolved now
Probably a rookie error on my part- every time I rebooted the pi I was navigating to TerrariumPI directory and inputting the start command- now I've skipped that step all seems to be working smoothly
Thanks for your help

closed 2019-02-13 by theyosh