theyosh/TerrariumPI #256: Integrate Syslog

Natrixz, 2019-02-10:
Hi Yosh,

Would it perhaps be possible to integrate a syslog function?
That way we are able to just send all the logs and events to a syslog server and do some smart things there.
(Especially with some of us having it "open" on the internet)

theyosh, 2019-02-10:
Hi Natrix,

this is/was already possible :) But I made an example for it and some documentation: https://github.com/theyosh/TerrariumPI/wiki/FAQ#syslog-local-and-remote

So, I think you should be able to do it out with code changes from my part. I only added an example in the logging.cfg but that is not enabled by default

Natrixz, 2019-02-10:
Ah dang!

The only thing in the FAQ that doesn't make sense is the line;
'/dev/log' to ('hostip',portnr) (take care with quotes)
;)

Other then that, let's see what happens =)

theyosh, 2019-02-10:
Local: `args=('/dev/log','local6')`
Remote: `args=(('192.168.1.1',514),'local6')`

Only I have not tested remote

Natrixz, 2019-02-10:
Well let’s test it now =)
I Will let you know

Natrixz, 2019-02-10:
Good news, it works perfectly!=)

theyosh, 2019-02-10:
Cool!

Natrixz, 2019-02-10:
Is there perhaps an easy way (without debug) to enable the incorrect password logging?
Right now if you enter a wrong password on switches etc it doesn't show up into the logging

theyosh, 2019-02-10:
That is correct. There was no logging for that. Now there is logging for incorrect logins. Valid logins are NOT logged. This due to keeping IP sessions in memory which I do not have at the moment

Natrixz, 2019-02-10:
Awesome, thanks!
I will test it out right now =)

Im already happy with failed logging ;)
That way it's still easy to spot if it's beeing brute-forced =)

theyosh, 2019-02-10:
But then this can closed again :). There will be some logging missing, but that is due to just lack off logging. More logging will be added every time code changes...

Natrixz, 2019-02-11:
Cool, thanks for the addition!

closed 2019-02-11 by Natrixz