theyosh/TerrariumPI #257: UV Index Measurement

stuartabrown, 2019-02-13:
Hi @theyosh,  feel free to close this if the answer is 'what?! no way!'.

So, there's quite a lot of discussion about UV Index required by different types of reptiles which in a way equates to how far from a UV source a basking spot should be e.g. 
[image]

The VEML6075 sensors mentioned in https://github.com/theyosh/TerrariumPI/issues/90 provide a way to calculate UV Index on page 6 of the technical documentation.  Do you think it might be possible to represent the UV Index calculated by these sensors in the TerrariumPI UI?

theyosh, 2019-02-16:
Done! :P

stuartabrown, 2019-02-16:
awesome!  I'm learning how to set up the MPDMv4.1 dimmer module at the moment for my next tutorial post but will move on to the UV sensors which have arrived immediately after :)

theyosh, 2019-02-16:
But I do just calculate the values. I do not say which value is correct. So it is up to you (owner) where the place the UV sensor in order to get the right values.

theyosh, 2019-03-20:
Ok, I am closing this now. As it is working for a while now and I made a new release. If there are problems, just reopen this one. But I expect that is not needed :)

closed 2019-03-20 by theyosh