theyosh/TerrariumPI #259: HLS Stream

stuartabrown, 2019-02-14:
So today an RPi camera, UV sensors and dimmer all arrive - and now I have to start writing my newbie tutorials for these!

I've connected up the RPi camera no problem with location rpicam_live.  It seems to refresh about once a minute - is that correct or should it show me real time view?

Also reading through https://github.com/theyosh/TerrariumPI/issues/245 I'm trying to play the stream through VLC.  However if I put http://[mylanip]/webcam/a94df035fc91a158c1f9d7a1c769ca66/stream.m3u8 into a browser NGINX throws a 404.  If I open the network stream in VLC it gives an error 'VLC is unable to open the MRL....'

Any tips much appreciated

theyosh, 2019-02-16:
Hmm, it sounds like some mixup. Ok, some things to check.
1. Make sure you have not also the 'rpicam' webcam added. You can only have one webcam with the rpicam as source. So make sure you only have an 'rpicam_live' webcam
2. If the settings are correct, please restart the software once. Just to make sure. Do you then see errors in the screen session.
3. When you are logged in to the Pi with ssh , enter the command `ps fax` and look for: 
```
4277 pts/1    S+     0:00          \_ /bin/bash ./live_rpicam.sh HLS Live 1920 1080 0 /dev/shm/webcam/a94df035fc91a158c1f9d7a1c769ca66
 4278 pts/1    Sl+    0:47          |   \_ /usr/bin/raspivid -o - -b 2000000 -t 0 -w 1920 -h 1080 -rot 0 --drc low -fps 30 -g 30 -pf main -lev 4.1 -ae 16,0xff,0x808000 -a 8 -a  HLS Live @ %d/%m/%Y %
 4279 pts/1    S+     2:25          |   \_ /usr/bin/ffmpeg -hide_banner -nostdin -re -i - -c:v copy -f hls -hls_time 2 -hls_list_size 3 -hls_flags delete_segments+split_by_time -hls_segment_filename
```

stuartabrown, 2019-02-17:
closing this.  I just pulled the latest version and the issue has gone away.  I suspect it's nothing to do with changes in the code but perhaps some internal caching / old config because I had previously had a camera with the same name and deleted it.  All works as advertised now :)

closed 2019-02-17 by stuartabrown