theyosh/TerrariumPI #260: Sensor Page - showing all Sensor as Gauge

funware84, 2019-02-15:
Hi
i wish to have an over all overview of all sensor atached.
Like all curent temperatures as gauge followed by humidity pH etc. at one overview page. 

Would be nice to see all curent values at a glance.

stuartabrown, 2019-02-15:
@funware84 I don't have loads of sensors attached yet but everything I have got shows on the dashboard (home > dashboard in the left-hand menu)

funware84, 2019-02-16:
HI, the Dashboard ist showing the average of temp or humidity, I would like to have all senseor, like all 6 temp sensor an all 4 humidity etc on 1 single page

theyosh, 2019-02-16:
@funware84 but then only the gauges... I have to think about that.. As it is not 'in line' with the other pages

funware84, 2019-02-16:
that would be awsome.  only gauges was what i wanted.

funware84, 2019-02-20:
exculde is not what i wanted
more like grouping or just every single value as gauge on a additional page

funware84, 2019-02-23:
Hi,
do u have any idea to solve it?
are u working on it

theyosh, 2019-02-23:
Hi,

yes I have made this possible with the following way:

1. `git pull` to get the latest version.
2. restart the software and go the the 'System' -> 'Settings' and just save your current settings.
3. login with ssh into your pi, and manual edit `nano /home/pi/TerrariumPI/settings.cfg`
4. Change the value of `sensor_gauge_overview` from `false` to `true` and save the file
5. restart once more and clear browser cache.

Then you should have under the sensors menu the menu `all`. That should give you the requested overview I hope.

funware84, 2019-02-24:
thanks a lot,
works fine

theyosh, 2019-02-24:
Cool. Then I close this now.

closed 2019-02-24 by theyosh