theyosh/TerrariumPI #262: Power_Switch referenced before assignment - TerrariumPI won't start

Natrixz, 2019-02-16:
Traceback (most recent call last):
  File "/home/nick/TerrariumPI/terrariumPI.py", line 26, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/nick/TerrariumPI/terrariumEngine.py", line 135, in __init__
    self.__load_power_switches()
  File "/home/nick/TerrariumPI/terrariumEngine.py", line 285, in __load_power_switches
    if power_switch.get_id() in prev_state:
UnboundLocalError: local variable 'power_switch' referenced before assignment

After the latest pull it won't start anymore :(
(wanted to test the new notifications haha)

theyosh, 2019-02-16:
Damm, thanks! I do like the remark about the rollback. That makes it indeed clear where the fuckup it :D
Should work again

Natrixz, 2019-02-16:
Your welcome! =)

Fix is confirmed ;)

closed 2019-02-16 by Natrixz