theyosh/TerrariumPI #264: HC-SR04 Ultrasonic Distance sensor error

toads100, 2019-02-20:
I keep getting this error. I have double check my wiring and it looks correct. I have done a complete power down and reboot. I can't figure out why I'm still getting this. any advise would be great. Im using a 5 gallon bucket for my water tank its 18 litters and 14 inches high. The sensor still seems to work with the error.

2019-02-20 09:53:02,131 - ERROR   - terrariumEnvironment - Environment watertank sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!

toads100, 2019-02-21:
I moved to different GPIO pins and all seems good now.

theyosh, 2019-02-24:
Thanks for the feedback

toads100, 2019-02-24:
Well after a few days it seems the same error has come back. Im not sure what it could be now

ERROR - terrariumEnvironment - Environment watertank sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!

2019-02-24 08:22:55,381 - WARNING - terrariumSensor      - Measured value Noneinch from hc-sr04 sensor 'Water Level' is outside valid range 0.00inch - 15.00inch in 4.00468 seconds.

theyosh, 2019-02-24:
Buy new/different hardware? From a different supplier? The red one from sparkfun do not work ( https://www.sparkfun.com/products/13959 )

toads100, 2019-02-24:
My power supply is 4 amps out on 5 volts. This is the sensor im using

[image]
[image]

toads100, 2019-02-24:
I think I will order another sensor. Do you have one you can recommend?

theyosh, 2019-02-24:
As they are not that expensive in my opinion, just order a couple from ebay or amazon. I do not keep track where I bought my hardware. Cheap sensor of about 5 dollars I will just buy in bulk.

So that is also an advice. Always buy more sensors then needed. Backup hardware is always handy t have. I has saved me in the pasted

toads100, 2019-02-24:
We have a local electric store I’m heading to now. I’m going to buy a few and try them. I’ll report back hopefully it’s just a faulty sensor

toads100, 2019-02-25:
Ok, so I got a new sensor yesterday. And I have the same error. I did Used 10K and 4.7K resistor do you think it matters what resister goes where? I used the 4.7k resistor on the ECHO feed and the 10k from GPIO 14 to 16 and then I have my TRIG on Gpio 13

toads100, 2019-02-26:
Issue resolved. Faulty ground wire

closed 2019-02-26 by toads100