theyosh/TerrariumPI #266: Show temperature on display

Kollimat, 2019-02-24:
Hello i just got ssd1306 diplay and i want to know if there is or will be posibility to show actual temperature and humidity on display, because now im not at home so much and my mom is taking car of my snake and it would be nice for her to always see temperature on display, its little hard for her to know how to connect to the web interface and look. Thank you :)

theyosh, 2019-03-02:
Hi,

I have had more of those questions, but I am sorry, but this will introduce a lot of extra code and interfacing. Because you are now asking for temperature. Another one wants the humidity. Another wants a mix of both etc...

So for now, it will show the average values per sensor type. And that will be updated every 30 seconds. So it is possible to readout the current average temperature. But you have to wait for it maybe when there is a lot of lines and text.

theyosh, 2019-03-04:
So how about something like: https://harizanov.com/2013/02/using-my-1-8-tft-as-a-raspberry-pi-status-display/

The python code is using weather data from a different source. But that can be changed to the TerrariumPI server. So then you can just have another Pi running the display where you want. As long as they are on the same Wifi.

theyosh, 2019-03-10:
Closing this, due to outside software scope

closed 2019-03-10 by theyosh

kahuwi14, 2019-11-09:
Hey there!
Sorry for being away such a long time...

I just wrote a little Python-script to display all JSON-Output on an OLED-Display. 
It's nothing special, but it works and I would post it (as last time with the notification-script, if you'd like)!

Greetings,
kahuwi