theyosh/TerrariumPI #267: 4 Channel I2C AC Dimmer Support

Ultameciia, 2019-02-27:
Still super new to all this so I apologize if this is a dumb question. I'm looking to add a way to control lighting and heating via a dimmable interface. After some looking I came across [this](https://www.amazon.com/Channel-Dimmer-Arduino-Raspberry-Projects/dp/B06Y1TSQ4Q/ref=sr_1_fkmr0_1?keywords=dimmable+raspberry+pi+relay&qid=1551278579&s=electronics&sr=1-1-fkmr0). 

Would this be supported? If it's not, is there a process for me to do to make it work?

stuartabrown, 2019-02-27:
@Ultameciia the dimmer that is recommended is https://www.tindie.com/products/nEXT_EVO1/universal-ac-mains-dimmer-mpdmv41/ but it may be that  @theyosh can support the one you mention

Ultameciia, 2019-02-27:
I guess that would be half the question. The libraries are readily available for the dimmer I linked. I like that this one has 4 relays compared to the two on the other. If its relatively simple to integrate then I might be able to do so and then submit the code to theyosh for review and possibly integration.

AndrejWeber, 2019-02-27:
@Ultameciia 
I'm shure, @theyosh can implement this module, but it is not able to dim for example halogen Lamps 

> With this module you can control the intensity of incandescent lamps, LED dimmable bulbs, power heating elements or fan speed [...]

Ultameciia, 2019-02-27:
Right, the goal for it in my use case would be a dimmable led strip for general lighting and then control of an Arcadia DHP. The other two relays im hoping to use just as a simple on off. I assume I can do that. Dimmed at 100% would be off correct?

Ultameciia, 2019-02-28:
Sorry to flood this, this project has me all sorts of excited for the things I can do. The original 4 channel one I posted doesn't seem to have as much documentation as [this one.](https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html)

It's also cheaper by quite a bit. If anything this might be a more cost effective way to reach the same goal.

AndrejWeber, 2019-02-28:
Much more Documentation, you are right!

> Dimming`can be achieved by Pulse Skip Modulation:
Method 1 — One or more cycles (sine wave signal) are transferred to the load, while following one or several cycles are blocked.
Method 2 — Partial transferrence of each sine wave to the load.
Method 3 — Generation of modulated full sine signal of different frequency up to few hundred hertz. This method requires specialized powerfull AC generators with different modulation.
Methods 1 and 2 are the easiest to execute with the help of a Dimmer and programm code: in both cases, there is a need of circuit that detects the zero crossing and can control a triac.

That is the reason, why the already known dimmer is the better alternative. 
It has a build in detection of zero crossing.
The cheap one has to detect from another hardware and then you have to look for timing and so on... very difficult!
For normal filament light bulbs it will work also without detection, but not smooth. 
Fans or heater are much to slow, to bother any zero crossing. Only at low dimming, like 1% a fan could stop, and get hot.
A heater will not care about anything it will just burn all energy it gets. 

In my opinion, if you will not dim any halogen lamps or something else, that cares for a well waveform then I would prefer the dimmer from Amazon.

Ultameciia, 2019-02-28:
Wait, maybe I'm missing something. The sensor I linked with just the one relay does have built in Zero Cross Detection. I don't believe the first one does (the 4 relay I2C one). That's why I mentioned the second one might be a better option.

AndrejWeber, 2019-02-28:
Oh ok, it has a build in detector, but only a detector!
Now you need to calculate in real-time, every millisecond, the whole sinus wave long has an own dim value, the detector does nothing, it only gives a signal to pinout, about 100 signals every second. 
Theoretically the Raspi has to calculate the dim state. But Linux can not deal real-time processes. 
I have read about arduino can do this, but there I cannot help you. Google can do it 🤪
This Logic to calculate the right dim state in real-time is build in at MPDMv4.1

theyosh, 2019-03-02:
Hmm, for me I cannot judge if it will work. And as I understand everything well from above here, it is not the most stable and suitable dimmer device. And that I cannot find Python code to work on a Raspberry PI, I think I will not add support for this.

The dimmers that are currently support are in my opinion good enough.

closed 2019-03-04 by Ultameciia