theyosh/TerrariumPI #268: Cant add 1wire sensor

prcdslnc13, 2019-02-28:
When I try to add a DS18b20 under the 1Wire entry in sensors I just get the blue status bar but no confirmation on the web interface. 

When I am logging I get:
`2019-02-28 10:59:33,518 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2019-02-28 10:59:38,553 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 46, in webserver_headers
    return fn(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 88, in wrapper
    return func(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 282, in __update_api_call
    result['ok'] = self.__terrariumEngine.set_config(path,postdata,request.files)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1159, in set_config
    update_ok = self.set_sensors_config(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 713, in set_sensors_config
    self.__load_sensors(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 186, in __load_sensors
    self.__unit_type)
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 460, in __new__
    return sensor(sensor_id, sensor_type, address, name, callback_indicator)
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 82, in __init__
    self.update()
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 98, in update
    cached_data = self.__sensor_cache.get_sensor_data(self.get_sensor_cache_key())
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 86, in get_sensor_cache_key
    self.__sensor_cache_key = md5((self.get_type() + self.get_address()).encode()).hexdigest()
TypeError: cannot concatenate 'str' and 'NoneType' objects`

There are 2 attached to phys pin 7. Any help would be awesome

stuartabrown, 2019-03-02:
Have you ssh'd in and run 'i2cdetect -y 1' as it suggests? That should give you the address

theyosh, 2019-03-02:
Hi, you cannot add 1wire sensors throught the web interface. When you have connected them according to: https://github.com/theyosh/TerrariumPI/wiki/Hardware#ds1820

Then you should restart once, so it will scan for it during start up. Adding sensors when the software is running, will not work. You have to restart/reboot after adding hardware.

prcdslnc13, 2019-03-03:
Thanks for the clarification that cleared it up. This can be closed!

closed 2019-03-04 by theyosh