theyosh/TerrariumPI #269: Webcam Streaming from IP error

Ultameciia, 2019-03-02:
I'm trying to use a webcam that has a remote stream setup. Currently using a program to stream the feed from the webcam. I can access it on all devices via http://10.0.0.95:8081, however when I add that address I see a ton of errors from TerrariumPi. On the interface I never get a settings saved dialogue and the webcam isn't saved. In screen -r I see: 

`2019-03-01 20:48:12,435 - ERROR   - terrariumWebcam      - Error getting raw USB image from webcam 'Tokay Cam' with error message:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 518, in get_raw_data
    camera = cv2.VideoCapture(int(self.location[10:]))
ValueError: invalid literal for int() with base 10: '0.0.95:8081'
2019-03-01 20:48:13,180 - ERROR   - terrariumWebcam      - Error getting raw USB image from webcam 'Tokay Cam' with error message:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 518, in get_raw_data
    camera = cv2.VideoCapture(int(self.location[10:]))
ValueError: invalid literal for int() with base 10: '0.0.95:8081'
2019-03-01 20:48:13,909 - ERROR   - terrariumWebcam      - Error getting raw USB image from webcam 'Tokay Cam' with error message:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 518, in get_raw_data
    camera = cv2.VideoCapture(int(self.location[10:])
`

Is there another method to do this? I'm at a bit of a loss.

theyosh, 2019-03-02:
What kind of stream does it produce? And I think you have not added as a remote camera in TerrariumPI? Because the code suggests that it is trying to connect to it as an USB webcam, which will not work.

Ultameciia, 2019-03-02:
The stream is MJPEG. Ive added it under camera settings. Under settings and location I have the IP there instead of /dev/video0.

stuartabrown, 2019-03-02:
Are you using that camera twice in TerrariumPI? Perhaps once for static and then setting it up again for streaming? I tried to do that as well but you can only use once

Ultameciia, 2019-03-02:
No, its a logitech c270. I orginally had it at /dev/video0 but didnt have a powered usb hub so the performance was sub optimal. I then set it up on an old laptop with yawcam. Its an mjpeg stream (also have an ootion to stream via javascript, but dont know too much about that).

theyosh, 2019-03-03:
When you can watch the stream on http://10.0.0.95:8081 are you able to right click on the image and then select 'show image only?' Because you need the direct image link. And I think you have now entered the website and not the stream image url.

Ultameciia, 2019-03-04:
I thought you might be right, so I tried to drill down the exact source. This is the stream of errors.

`pi@raspberrypi:~ $ screen -r
2019-03-03 21:12:27,487 - ERROR   - terrariumWebcam      - Error getting raw USB image from webcam 'Tokay Cam' with error message:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 518, in get_raw_data
    camera = cv2.VideoCapture(int(self.location[10:]))
ValueError: invalid literal for int() with base 10: '6.73.185:8081/video.mjpg?q=30&fps=33&id=0.46729548020019696&r=1551669079613'
2019-03-03 21:12:28,307 - ERROR   - terrariumWebcam      - Error getting raw USB image from webcam 'Tokay Cam' with error message:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 518, in get_raw_data
    camera = cv2.VideoCapture(int(self.location[10:]))
ValueError: invalid literal for int() with base 10: '6.73.185:8081/video.mjpg?q=30&fps=33&id=0.46729548020019696&r=1551669079613'`
.
The numbers are a tad different for IP but that's just due to the port forwarding I have set up so it points to a public IP instead of an internal IP. The actual stream source is [here](http://73.6.73.185:8081/video.mjpg?q=30&fps=33&id=0.46729548020019696&r=1551669079613) This link is live so you can see the source for yourself.

theyosh, 2019-03-04:
At this time, that source is not supported. I did not test that, and it is not supported right now.

I could make it work, but it is faster, if you could just provide a single .jpg file from the webcam. Not motion jpeg. A single url to a jpg should work.

Secondly, this motion jpeg livestream cannot be streamed live with my software. Because of it own HTTP server and port forwarding. And using the Pi as an encoder for live encoding is hard. And will not work either if the RPI cam is used also.

So the easiest right now, it to try to get a still image from the stream, and add that as a http source in my software.

theyosh, 2019-03-04:
Turns out not to be that hard... It is now supported.

theyosh, 2019-03-04:
Ok, do a `git pull`, and restart. This will trigger the installer due to new libs. So also rerun the installer. After that do not reboot and let the software run. It should pick up your webcam stream. At least I am currently using that for testing.

Ultameciia, 2019-03-04:
Ok I'll get that going and report back!

Ultameciia, 2019-03-04:
We're all good to go now! Thanks for your fast work, this really is an amazing project!

closed 2019-03-04 by Ultameciia