theyosh/TerrariumPI #270: Idea for new feature for secure monitore terrarium status.

kaju666, 2019-03-03:
Opening raspberry to public is very dangerous from security point of view.
So i think it will be nice feature to add some type of feature to view terririum status outside your network.
I saw your API and it have all needed informations.  But i'm not a programmer and can't do anything with that.

I thinking about simple script which is generating image. And on that image you have values of your terrarium like temperature, webcam image, light status. And that image is posted every e.g 30 minutes on some image hosting platform.

What do you think about that feature and is this hard to do?

stuartabrown, 2019-03-03:
Hi - the standard way to do this with no programming knowledge is to use https://www.dataplicity.com/.  You can then access your RPi securely over the public internet.

kaju666, 2019-03-03:
why do you think it is secure?
i know that page similar is to remonte.it

But you still opening 80 port for whole internet.
And i think it's very easy to bypass the login for terrarium pi, it's not https.

Kargha, 2019-03-03:
In fact, HTTPS wouldn't really make much of a difference. All it does is prevent people from getting too much information from the traffic, it's still open to bruteforce/dictionary/whatever type of attack you want to do. 

As for the port you open, that entirely depends on what you decide to setup for the port-forwarding. You can keep the internal port to 80, external port to anything you want. You can also make the machine not respond to various types of pings and what not as well using iptables, which helps hide it as well (since many automatic scanners will just ping a whole IP range and see what responds, and if it gets one, it will use that for further probing/attacks).

I'd say the main issue with it doing screen dumps or the like would be that if an alarm goes off (heat for example, or a sensor stops working), it can take up to half an hour before you'd even be able to notice. That and figuring out what URL you'd have to go to for the service.

Other ways to solve this could be, if you have a static IP, alternatively something like DynDNS, would be to have it create a reverse connection to your external machine (i.e. your laptop) either through reverse shell (and write a python script to query the API for data for example) or reverse tunnel. But it might be a bit over the top for something like this.

jornobe, 2019-03-04:
From most devices and operating systems you can run an OpenVPN client.

I have a RPi as access gateway running OpenVPN and DynDNS. From my IOS devices I just connect to VPN first and then access the TerrariumPi page internally over https or ssh.

For a quick look what's going on I have a script updating the MOTD  every 5 minutes so the "latest" temperature and humidity values are shown immediately after opening an SSH connection to the RPi running TerrariumPi.

theyosh, 2019-03-04:
@jornobe way cool! can you post that MOTD script here? Would be a nice addition

jornobe, 2019-03-04:
[image]

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

jornobe, 2019-03-04:
It's nothing fancy right :-) Just doing an owread and redirecting it to an output file, but it does what I wanted to achieve... 

#/bin/bash

echo Temperature CHE left: > ./check.results.txt
owread 28.56B9F0080000/latesttemp  >> ./check.results.txt
echo >> ./check.results.txt
echo Temperature CHE right:  >> ./check.results.txt
owread 28.5EE480080000/latesttemp  >> ./check.results.txt
echo >> ./check.results.txt
echo Temperature left: >> ./check.results.txt
owread 26.E58825020000/latesttemp >> ./check.results.txt
echo >> ./check.results.txt
echo Temperature right:  >> ./check.results.txt
owread 26.847725020000/latesttemp >> ./check.results.txt
echo >> ./check.results.txt
echo Humidity left: >> ./check.results.txt
owread 26.E58825020000/humidity >> ./check.results.txt
echo >> ./check.results.txt
echo Humidity right: >> ./check.results.txt
owread 26.847725020000/humidity >> ./check.results.txt
echo >> ./check.results.txt

clear
cat /home/pi/dino > /etc/motd
column -c 80 ./check.results.txt >> /etc/motd

jornobe, 2019-03-04:
I shared this for the idea behind not so much for the script itself ;-)

theyosh, 2019-03-04:
Thanks, and yes, it is the idea... Love the dino ascii art!

Kargha, 2019-03-04:
@jornobe that's a nice way to do it! And I love the Dino art. I do however curse you for getting the Jurassic park theme song stuck in my head :-p

theyosh, 2019-03-04:
But to get back on to the question. I am afraid that this is something outside the scope of my software. As I do provide a REST API for getting all kinds of information and doing actions, it should be possible to integrate. But that does involve programming.

Security is something that I cannot fix. As I made the software for inhouse use first. Then I made it possible to add HTTP authentication in front of it. So using it with datapilicty and add HTTP authentication, it should be secured enough in my opinion. As Dataplicity will take care of the SSL and therefore also securing the HTTP authentication headers, it is pretty save. But again, I use plain text authentication and the password are stored in plain text. So that is not secure!

And there are more requests like these. #266 Has the same issue. If you need a custom view or notifications, it is up to you to create it yourself. The tools are there.

kaju666, 2019-03-08:
ok thanks for explaining.

@theyosh can you tell us how did you make and always up to date image of your terrarium?
[image]
and i see it's uplaoded to github servers.

theyosh, 2019-03-10:
@kaju666 I have my terrarium accessible through my home internet connection. And  the picture you see is from my RPI cam above my terrarium.

Then I just picked the most zoomed out image url and posted that on github. Github will cache it, and update it when needed. This can be done, due to the fact that I have my terrariumpi software open for the internet.

theyosh, 2019-03-10:
But, there is also a RAW image icon at the webcams. That will give you always the latest picture. But it lacks date and time information.

kaju666, 2019-03-10:
Ok so i spend some time with python and bash and there it is :)
https://github.com/kaju666/TerrariumPI-Send-Webcam-Image-With-Data-To-Public

Maybe someone also want to make similiar so this will help.

It looks like this: http://kaju.ovh/cam1/terrarium.jpg

theyosh, 2019-03-11:
Cool! I add your script to my wiki :) https://github.com/theyosh/TerrariumPI/wiki/Integrations
This is a nice and simple example of picking you needed data and mix it together as you want. Nice and simple!

theyosh, 2019-03-20:
@kaju666 Then I am closing this issue now. Thanks for the idea and the result. And cool to see that you can use the remote data from my software.

closed 2019-03-20 by theyosh

kaju666, 2021-10-03:
I updated my script to v4 version.
Also demo is now here:
https://kaju.ovh/terra/terrarium.jpg