theyosh/TerrariumPI #271: API exposes sensitive data

Kargha, 2019-03-05:
The /api/config part of the TerrariumPI exposes a great deal of sensitive information, including:

- TerrariumPI authentication username
- SMTP Username, Password & Server
- ALL external API keys (pushover, telegram, twitter, etc)
- Meross Username and Password (which could potentially let an attacker gain access to any other smart device on the same Meross account. Not sure how severe it is as I don't use Meross myself.)

As such, anyone with an open installation (i.e. requiring authentication only on changes), has the above information available to the public. Sensitive information available through the API should require authentication, or perhaps even have it excluded from the API output.

theyosh, 2019-03-05:
Hmm, that is not ok! :) So, then I would suggest to authenticate for every setting page.

Other option is to enable authentication for all actions. That would also fix this.

But I think it is better safe then sorry, So I will add authentication on all settings pages...

Kargha, 2019-03-05:
@theyosh Yea, if the site requires authentication for all API calls, that'd help (unless you're referring to enforcing authentication for regular page access as well). If you require it for any page access, I guess it would kind of lower "QoL" (Quality of Life) for anyone running it purely in an internal environment. What are your thoughts?

theyosh, 2019-03-05:
hehe.. I only had to change one line of code :P

theyosh, 2019-03-05:
Then I will close this. :)

closed 2019-03-05 by theyosh