theyosh/TerrariumPI #272: Strange DHT22 error (they Not THIS Bad are they?)

wes1007, 2019-03-06:
been having some intermittent issues with DHT22 and DS18B20 temperature/Humidity sensors. Cable lengths are all shorter than 3.5m. The DS18B20 seem to be fine though as i get readings all the time.

THe DHT22s however seem to get none readings randomly. Sometimes come back and sometimes dont. HardReboots seem to help where i generally get 3 readings out of 4. then it starts going downhill. Runing sudo reboot seemed to drop all the DHT22s. 

Im concerned about the updates exceeding the timeout of 30s. Not sure if that has something to do with it.

DHT22 - InternalTemp/Humidity - Pin 13
DHT22 InternalTemp/Humidity Pin12

Logs below:

**Hard Reset 1**
`2019-03-06 18:43:38,045 - INFO    - terrariumEnvironment - Update done in 0.00447 seconds. Waiting for 14.99553 seconds for next update
2019-03-06 18:43:35,607 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-06 18:43:35,489 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-06 18:43:34,677 - WARNING - terrariumEngine      - Updating took to much time. Needed 43.68376 seconds which is 13.68376 more then the limit 30
2019-03-06 18:43:34,675 - ERROR   - terrariumEngine      - Updating is having problems keeping up. Could not update in 30 seconds for 11 times!
2019-03-06 18:43:33,888 - WARNING - terrariumSensor      - Measured value None% from dht22 sensor 'IntakeHumidity' is outside valid range 0.00% - 100.00% in 10.10896 seconds.
2019-03-06 18:43:23,657 - WARNING - terrariumSensor      - Measured value None% from dht22 sensor 'Internal-Humidity' is outside valid range 0.00% - 100.00% in 10.10840 seconds.
2019-03-06 18:43:23,043 - INFO    - terrariumEnvironment - Update done in 0.00453 seconds. Waiting for 14.99547 seconds for next update
2019-03-06 18:43:13,418 - INFO    - terrariumSensor      - Updated w1 sensor '1-WaterTemp' from 23.75C to 23.81C in 0.91758 seconds
2019-03-06 18:43:12,378 - INFO    - terrariumSensor      - Updated w1 sensor '2-WaterTemp' from 23.44C to 23.44C in 0.92718 seconds
2019-03-06 18:43:11,328 - WARNING - terrariumSensor      - Measured value NoneC from dht22 sensor 'Internal-Temp' is outside valid range 1.00C - 42.00C in 10.10681 seconds.
2019-03-06 18:43:08,041 - INFO    - terrariumEnvironment - Update done in 0.00454 seconds. Waiting for 14.99546 seconds for next update
2019-03-06 18:43:05,605 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-06 18:43:05,487 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-06 18:43:01,099 - WARNING - terrariumSensor      - Measured value NoneC from dht22 sensor 'IntakeTemp' is outside valid range 0.00C - 45.00C in 10.10727 seconds.
2019-03-06 18:42:53,135 - INFO    - terrariumEnvironment - Update done in 0.10041 seconds. Waiting for 14.89959 seconds for next update
2019-03-06 18:42:53,086 - INFO    - terrariumSwitch      - Changed power switch '5-IntakeFanHigh/Low' of type 'gpio-inverse' at address '35' from state 'False' to state 'True' (Forced:False)
2019-03-06 18:42:53,039 - INFO    - terrariumSwitch      - Changed power switch '7-ExhaustFanHigh/Low' of type 'gpio-inverse' at address '32' from state 'False' to state 'True' (Forced:False)
2019-03-06 18:42:53,037 - INFO    - terrariumEnvironment - Environment temperature is turning on the alarm max powerswitches based on sensor
2019-03-06 18:42:50,990 - WARNING - terrariumEngine      - Updating took to much time. Needed 43.77125 seconds which is 13.77125 more then the limit 30
2019-03-06 18:42:50,989 - ERROR   - terrariumEngine      - Updating is having problems keeping up. Could not update in 30 seconds for 10 times!
2019-03-06 18:42:50,213 - WARNING - terrariumSensor      - Measured value None% from dht22 sensor 'IntakeHumidity' is outside valid range 0.00% - 100.00% in 10.10880 seconds.
2019-03-06 18:42:39,973 - WARNING - terrariumSensor      - Measured value None% from dht22 sensor 'Internal-Humidity' is outside valid range 0.00% - 100.00% in 10.19280 seconds.
2019-03-06 18:42:38,038 - INFO    - terrariumEnvironment - Update done in 0.00595 seconds. Waiting for 14.99405 seconds for next update
2019-03-06 18:42:38,035 - INFO    - terrariumEnvironment - Environment temperature alarm max is triggered to state True, but the settle time has not passed: 10.0 seconds.
2019-03-06 18:42:37,383 - INFO    - terrariumSwitch      - Changed power switch '5-IntakeFanHigh/Low' of type 'gpio-inverse' at address '35' from state 'True' to state 'False' (Forced:False)
2019-03-06 18:42:37,337 - INFO    - terrariumSwitch      - Changed power switch '7-ExhaustFanHigh/Low' of type 'gpio-inverse' at address '32' from state 'True' to state 'False' (Forced:False)
2019-03-06 18:42:35,604 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-06 18:42:35,484 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-06 18:42:29,658 - INFO    - terrariumSensor      - Updated w1 sensor '1-WaterTemp' from 23.81C to 23.75C in 0.91997 seconds
2019-03-06 18:42:28,618 - INFO    - terrariumSensor      - Updated w1 sensor '2-WaterTemp' from 23.44C to 23.44C in 0.93457 seconds
2019-03-06 18:42:27,557 - WARNING - terrariumSensor      - Measured value NoneC from dht22 sensor 'Internal-Temp' is outside valid range 1.00C - 42.00C in 10.10943 seconds.
2019-03-06 18:42:23,034 - INFO    - terrariumEnvironment - Update done in 0.00446 seconds. Waiting for 14.99554 seconds for next update
2019-03-06 18:42:17,327 - WARNING - terrariumSensor      - Measured value NoneC from dht22 sensor 'IntakeTemp' is outside valid range 0.00C - 45.00C in 10.10949 seconds.
2019-03-06 18:42:07,756 - INFO    - terrariumEnvironment - Update done in 0.00453 seconds. Waiting for 14.99547 seconds for next update
2019-03-06 18:42:07,216 - WARNING - terrariumEngine      - Updating took to much time. Needed 43.66859 seconds which is 13.66859 more then the limit 30
2019-03-06 18:42:06,444 - WARNING - terrariumSensor      - Measured value None% from dht22 sensor 'IntakeHumidity' is outside valid range 0.00% - 100.00% in 10.10730 seconds.
2019-03-06 18:42:05,603 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-06 18:42:05,482 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-06 18:41:56,208 - WARNING - terrariumSensor      - Measured value None% from dht22 sensor 'Internal-Humidity' is outside valid range 0.00% - 100.00% in 10.10841 seconds.
2019-03-06 18:41:52,231 - INFO    - terrariumEnvironment - Update done in 0.00449 seconds. Waiting for 14.99551 seconds for next update
2019-03-06 18:41:45,978 - INFO    - terrariumSensor      - Updated w1 sensor '1-WaterTemp' from 23.81C to 23.81C in 0.91250 seconds
2019-03-06 18:41:44,938 - INFO    - terrariumSensor      - Updated w1 sensor '2-WaterTemp' from 23.44C to 23.44C in 0.92896 seconds
2019-03-06 18:41:43,887 - WARNING - terrariumSensor      - Measured value NoneC from dht22 sensor 'Internal-Temp' is outside valid range 1.00C - 42.00C in 10.10754 seconds.
2019-03-06 18:41:37,338 - INFO    - terrariumEnvironment - Update done in 0.11364 seconds. Waiting for 14.88636 seconds for next update
2019-03-06 18:41:37,276 - INFO    - terrariumSwitch      - Changed power switch '5-IntakeFanHigh/Low' of type 'gpio-inverse' at address '35' from state 'False' to state 'True' (Forced:False)
2019-03-06 18:41:37,228 - INFO    - terrariumSwitch      - Changed power switch '7-ExhaustFanHigh/Low' of type 'gpio-inverse' at address '32' from state 'False' to state 'True' (Forced:False)
2019-03-06 18:41:37,227 - INFO    - terrariumEnvironment - Environment temperature is turning on the alarm max powerswitches based on sensor
2019-03-06 18:41:35,601 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-06 18:41:35,480 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
[... cut ...]
2019-03-06 18:30:59,878 - INFO    - terrariumEngine      - Done setting terrariumPI authentication
2019-03-06 18:30:59,876 - INFO    - terrariumEngine      - Setting terrariumPI authentication
2019-03-06 18:30:59,875 - INFO    - terrariumEngine      - Done Loading terrariumPI config
2019-03-06 18:30:59,850 - INFO    - terrariumEngine      - Loading terrariumPI config
2019-03-06 18:30:59,367 - INFO    - terrariumPI          - Starting terrariumPI
  65 2   Raspberry Pi 3 Model B Plus Rev 1.3 - TerrariumPI 3.9.4 - Terrarium home automation v. 3.9.4 - Github`

jornobe, 2019-03-07:
Because they are not providing constant trustable readings I have moved away from the DHT22 sensors. 

Now use the RPI3 1-wire host adaptor and SWE3 humidity/temperature sensors from Sheepwalk Electronics. Solid readings 24h a day! These sensors hook up using standard UTP cables to the 1-wire host adaptor which itself plugs onto the Pi directly.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

wes1007, 2019-03-07:
Damn. not exactly what i was wanting to hear. I have a bunch of SI7021 Temp/humidity sensors that run over I2C but my concern is that 3m of cable might not be idea with them (that and ill have to make a new breakout board :( )

I have a DHT22 running on a nodemcu tied into openhab for home automation thats been working pretty flawlessly for a while now. maybe i just got lucky with that one?

wes1007, 2019-03-07:
@theyosh anything to add to this beside chuck the DHT22s in the bin?

theyosh, 2019-03-09:
I am not sure about this. I do experience that DHTXX are not that stable. I have just one for testing, but that is failing about 30% of the time according to my logfiles. So that is 1/3 of all measurements. And that is not that stable.

Also, I do get the feeling, that it is a bit sensible to power fluctuations as well that it needs somehow to much power when it needs to measure.

So, my experience is that DHTXX are not useful. I also recommend to use something like the SWE3 from Sheepwalk. That uses it own bus and different sensors. Which are a bit more expensive, but I have it running now for couple of years already without any issues. It is way more stable.

I do had some sensors over 2 meter cable with I2C. It did work, until some water came into the sensors ;). But this is also for hard to judge. I do not have that much hardware experience. But, I would just say, try one with 3M cable. And see how it works. If it stays stable for lets say a week, you could try to use it in production.

And do you see power errors when you run `dmesg`?

Voltage normalised (0x00000000)
Under-voltage detected! (0x00050005)

theyosh, 2019-03-20:
Ok, I close this now, as there is not much that I can do about this. DHTXX is cheap and nice for some small projects. But 24/7 is just to much for those sensors....

closed 2019-03-20 by theyosh