theyosh/TerrariumPI #273: Energie vs Meross Power Boards - Pros/Cons, Degree of Support etc.

victorhooi, 2019-03-07:
I see that TerrariumPi now supports the Meross MSS425E Power Switches =).

How do these compare to the Energie EG-PMS2 and EG-PMS2-LAN, in terms of TerrariumPi support, reliability, features etc?

The Energie seem to be controllable over a local web interface, whereas the Meross devices are only cloud-controllable, is that correct? Is there no local way of controlling the Meross devices?

Might be good to document some of the pros/cons of either, or experiences in the wiki.

Or are there other ways of controlling power that people recommend.

Also - I noticed a small typo in the wiki (Heading [here](https://github.com/theyosh/TerrariumPI/wiki/Hardware#eg-pm2-1) should be EG-PM2-LAN) what's the best way of suggesting/contributing changes there?

(I'm building a new TerrariumPi setup, and need to control things like a mister unit, LED lights etc.)

theyosh, 2019-03-07:
Hi,

yes, I do support Meross MSS425E, but I have not tested it my self. So I can only explain what it should do ;) You have to enter your login details that you use in your phone app. And then my software will act as a client that can control the power switches. But that is done by sending a command to the Merros cloud, which will send the command back to the power switch. So it needs an internet connection for operating. That is a downside if you ask me.
Else you have to reverse engineer the commands that are being sent. But this could be copyrighted somehow, so I will not go that way.
But it the power switches can be controlled by the TerrariumPI software, so you can just use them for sprayers or lighting. But I have not tested this, so I cannot say how much time it will take before the command is actually executed.

All Energie works with local lan. So that is safer. I have the USB and LAN version here. The USB works always. No issues there. The LAN version has a Wifi client connected to it, to make it wireless :) But that is not that stable, and my Energie will lose connection. Therefore I have a 24/7 ping running to that IP which keeps the connections alive. But I think this has to do with my Wifi setup. So directly to the LAN it could work better. Not tested that much.

But I think both power switches needs a lot of space. And therefore I do like those better: https://denkovi.com/usb-relay-board-four-channels-for-home-automation-v2
You have to do some manual work to make all the connections and wiring, but this is much smaller and integrates better with home build systems.

And for dimming I recommend: https://github.com/theyosh/TerrariumPI/wiki/Hardware#nextevo Works like a charm

I have updated the hardware page. Thanks for spotting.

And if you are building a new setup, please take a look at: #210

Natrixz, 2019-03-07:
You actually don't support the denkovi board you have linked.
That one uses a jar file and you didn't want to incorporate it, unless that has changed? =)

(Would make me happy as well haha)

theyosh, 2019-03-07:
@Natrixz damm you are fast and correct. The right one is indeed the old one: https://denkovi.com/usb-relay-board-four-channels-for-home-automation

That stupid Java is still an issue... But that is a long time ago, so maybe we can have another look at it.

Natrixz, 2019-03-07:
Haha yeah I try and keep up to date with all development for all my terraria ;)

I would love it if your able to make the v2 work as well =)
Main benefit being that it can save the state of the relay so that if for some reason it powers down it will come back up in that state =)

victorhooi, 2019-03-07:
Interesting - yeah, it sounds like Energie or Denkovi boards are potentially more reliable.

And yes will definitely add photos to that issue when I'm done (although may be a while to plan it all out and pull together parts).

I just filed a new issue - https://github.com/theyosh/TerrariumPI/issues/274 - what do you think about the Sonoff (or other ESP8266-based switches) with the [Tasmota](https://github.com/arendst/Sonoff-Tasmota) firmware?

They seem very popular in home automation community, and the firmware is quite actively developed, and expose a MQTT, REST API, or serial interface.

theyosh, 2019-03-07:
@Natrixz do you own a V2 version? : https://github.com/theyosh/TerrariumPI/tree/denkovi_v2

You need to manually install java on your pi. And you have to download the tool yourselfs due to legal stuff from: http://denkovi.com/denkovi-relay-command-line-tool and place the contents of zip in the folder `DenkoviRelayCommandLineTool` and rename the jar to: `DenkoviRelayCommandLineTool.jar`

Then you can add the new V2 versions of Denkovi relays. It should work with the 4,8 and 16 relays versions. Just enter the power outlet number starting from 1 you want to use.

Natrixz, 2019-03-07:
Awesome!
That was fast :)

I will hook one of my v2’s up tomorrow morning and test it right away:)
I will let you know how it go’s:)

Thanks!

theyosh, 2019-03-07:
Yes, the code refactor a year ago made this possible :) Lets see what happens

Natrixz, 2019-03-08:
Will take me a little longer...there appears to be an issue with the raspbian mirror -,-

jornobe, 2019-03-08:
What I like a lot about the Energenie LAN version is that you can set a “hardware schedule” via the management tool in seconds, not minutes. 

I have set a recurring switch off command of the plug that controls my rain system in its hardware schedule. It runs every 30 seconds which makes sure that the enclosure cannot flood in case something goes wrong with the communication between the Pi and the EG-PMS2-LAN (which happened to me twice...).

Sometimes TerrariumPi needs to trigger 2 cycles of rain to have it run long enough but I prefer that over having the 320L pool (which the rain system uses as source) being emptied over the sprinklers...

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Natrixz, 2019-03-08:
> Yes, the code refactor a year ago made this possible :) Lets see what happens

Made a new issue as not to clog this one =)

closed 2019-04-22 by theyosh