theyosh/TerrariumPI #274: Sonoff (Tasmota) support via MQTT or REST API?

victorhooi, 2019-03-07:
_This is related to this [earlier issue](https://github.com/theyosh/TerrariumPI/issues/65), but I can provide some more details._

Would it be possible to add support for Sonoff devices, specifically those using the [Tasmota firmware](https://github.com/arendst/Sonoff-Tasmota)?

[Tasmota](https://github.com/arendst/Sonoff-Tasmota
) is open-source firmware for ESP8266-based smart-switches, which includes both the Sonoff devices, Shelley devices and many others brands.

Sonoff devices are cheap ($5-10), easily orderable on the internet, and can be wired up to your own plugs/devices.

Basic version - https://www.itead.cc/sonoff-wifi-wireless-switch.html

You can send commands via MQTT, via a web REST API, or over a serial bridge:

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

victorhooi, 2019-03-07:
Also, @theyosh if it helps, I can send you a Sonoff device already flashed to latest open-source Tasmota firmware, and wired up however you like?

theyosh, 2019-03-07:
Hmm, I see this: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sending-commands-with-web-requests

If that works, it is quite easy to integrate. But I have also seen: https://www.instructables.com/id/Control-Sonoff-From-Raspberry-Pi/ with ESP easy. So there are more custom builds for the same device... Have to figure out how to deal with that

victorhooi, 2019-03-08:
Interesting.

This is an article comparing the different firmwares:

https://lobradov.github.io/FOSS-Firmware-comparison-overview/

I think Tasmota might be a good starting point, as it's very popular with the home automation crowd.

Happy to help with any testing you might want to do (if you don't want to get your own Sonoff device).

theyosh, 2019-03-09:
Ok, if you are able to switch to the branch `feature/sonoff` you could test if it will work. I have tried to support three different firmware versions.

The address is in the form: http://[OPTIONAL_USER]:[OPTIONAL_PASS]@SONOFF_DEVICE 

At this time, I only expect one switch per device. But I have seen multiple power switches per device. So that will be supported later on, as I know this is working.

theyosh, 2019-03-11:
@victorhooi Able to test this? I am bit curious if it is working already. Code was not that hard to make

Ultameciia, 2019-03-21:
If he doesn't respond, I may be purchasing a few Sonoff devices in the future. If I do I can test for you.

theyosh, 2019-03-31:
@Ultameciia of course everybody is allowed to test. If you want to test, you have to download the code with the following command:

`git clone --recursive https://github.com/theyosh/TerrariumPI.git`

Make sure you have moved you existing version to a temp directory before downloading the new version. You can just copy the settings.cfg file from your old location to the new version.

Then you need to switch to the test branch with:

`git checkout issue/274`

And then restart. This should give you the new code that can be tested.

duenni, 2019-04-17:
Hi,

I try to add an Sonoff S20 with Tasmota.
As address i tried `http://hostname` and also http:://ipaddress (e.g. `http://10.10.10.10`). I hit submit but the switches don't get saved.

Log shows:
````
2019-04-17 22:23:50,218 - INFO    - terrariumEngine      - Updating terrariumPI switches
2019-04-17 22:23:49,253 - INFO    - terrariumEngine      - Loaded 0 excluding IDs: []
2019-04-17 22:23:49,252 - INFO    - terrariumEngine      - Updating terrariumPI switches
2019-04-17 22:23:44,633 - INFO    - terrariumEngine      - Loaded 0 excluding IDs: []
2019-04-17 22:23:44,631 - INFO    - terrariumEngine      - Updating terrariumPI switches
2019-04-17 22:23:41,991 - INFO    - terrariumEngine      - Loaded 0 excluding IDs: []
````

Any idea?

theyosh, 2019-04-22:
Hmm, I will look into this. Should be something small...

theyosh, 2019-04-22:
Ok, are you on the right banch? This code is not on the master branch.

So you have to follow: https://github.com/theyosh/TerrariumPI/issues/274#issuecomment-478344088

and restart. Then you should be able to add those power switches

theyosh, 2019-04-22:
My mistake. I just missed some line of code. So do a `git pull` and restart it. It should show some more debug in the screen session

duenni, 2019-04-22:
Thank you!

Added a switch with address `http://sonoff2`.
It gets added but I can not toggle it.
`log`
````
2019-04-22 12:22:11,552 - ERROR   - terrariumSwitch      - Failed changing power switch 'sonoff2' of type 'sonoff' at address 'http://sonoff2' from state 'None' to state 'True' (Forced:False)
2019-04-22 12:22:06,065 - INFO    - terrariumEngine      - Done updating terrariumPI switches. Found 1 switches in 0.120 seconds
2019-04-22 12:22:06,060 - INFO    - terrariumSwitch      - Loaded power switch 'sonoff2' with values: power 0.0W and waterflow 0.0L/s
2019-04-22 12:22:06,059 - ERROR   - terrariumSwitch      - Failed changing power switch 'sonoff2' of type 'sonoff' at address 'http://sonoff2' from state 'None' to state 'False' (Forced:True)
2019-04-22 12:22:05,947 - INFO    - terrariumSwitch      - Initialising 'sonoff' power switch object
2019-04-22 12:22:05,946 - INFO    - terrariumEngine      - Loaded 0 excluding IDs: []
2019-04-22 12:22:05,945 - INFO    - terrariumEngine      - Updating terrariumPI switches
````

`screen`
````
Test Tasmota
http://sonoff2/cm?cmnd=Power
Result
{u'POWER': u'OFF'}
'_sre.SRE_Match' object has no attribute '__getitem__'
2019-04-22 12:22:06,059 - ERROR   - terrariumSwitch      - Failed changing power switch 'sonoff2' of type 'sonoff' at address 'http://sonoff2' from state 'None' to state 'False' (Forced:True)
'_sre.SRE_Match' object has no attribute '__getitem__'
2019-04-22 12:22:11,552 - ERROR   - terrariumSwitch      - Failed changing power switch 'sonoff2' of type 'sonoff' at address 'http://sonoff2' from state 'None' to state 'True' (Forced:False)
'_sre.SRE_Match' object has no attribute '__getitem__'
2019-04-22 12:22:17,530 - ERROR   - terrariumSwitch      - Failed changing power switch 'sonoff2' of type 'sonoff' at address 'http://sonoff2' from state 'None' to state 'True' (Forced:False)
````

theyosh, 2019-04-22:
another git pull? Made a small change

duenni, 2019-04-22:
This was the fastest fix I have ever seen. 😄 
Thank you, it is working now. Really cool!

`log`
````
2019-04-22 12:33:18,086 - INFO    - terrariumSwitch      - Changed power switch 'sonoff2' of type 'sonoff' at address 'http://sonoff2' from state 'True' to state 'False' (Forced:False)
2019-04-22 12:33:13,267 - INFO    - terrariumSwitch      - Changed power switch 'sonoff2' of type 'sonoff' at address 'http://sonoff2' from state 'False' to state 'True' (Forced:False)
2019-04-22 12:33:06,149 - INFO    - terrariumEngine      - Update done in 2.04940 seconds. Waiting for 27.95060 seconds for next update
2019-04-22 12:33:06,007 - INFO    - terrariumSwitch      - Updated sonoff power switch 'sonoff2' status in 0.00009 seconds
````
`screen`
````
Test Tasmota
http://sonoff2/cm?cmnd=Power
Result
{u'POWER': u'OFF'}
Toggle Sonoff
http://sonoff2/cm?cmnd=Power%200
State
{u'POWER': u'OFF'}
2019-04-22 12:33:04,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
Toggle Sonoff
http://sonoff2/cm?cmnd=Power%201
State
{u'POWER': u'ON'}
Toggle Sonoff
http://sonoff2/cm?cmnd=Power%200
State
{u'POWER': u'OFF'}
````

theyosh, 2019-04-22:
Cool! And another update, which will read out the current state of the switch. So if you do git pull and restart.

No my software will also see the power switch changes when you toggle with your mobile phone or any other remote control for you Sonoff.

duenni, 2019-04-22:
This also works. 👍 
I also did a quick test with timers.

Thanks again, this is really great work!

theyosh, 2019-04-22:
Cool!. Then the code is now also in the master branch. Other Sonoff firmware versions will be supported when people use them...

theyosh, 2019-04-22:
Then I am closing this now. As there is not much more to be done. The initial request is to support Tasmota firmware which it is. Thanks for testing everybody!

closed 2019-04-22 by theyosh