theyosh/TerrariumPI #275: Denkovi V2

Natrixz, 2019-03-08:
Exception AttributeError: "'_OwnetConnection' object has no attribute 'socket'" in <bound method _OwnetConnection.__del__ of <pyownet.protocol._OwnetConnection object at 0x6c1dfa50>> ignored
2019-03-08 11:16:37,777 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
2019-03-08 11:16:43,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
Traceback (most recent call last):
  File "/home/user/TerrariumPI/terrariumPI.py", line 31, in <module>
    terrariumWebserver.start()
  File "/home/user/TerrariumPI/terrariumWebserver.py", line 480, in start
    quiet=True)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 755, in run
    run(self, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 3129, in run
    server.run(app)
  File "/usr/local/lib/python2.7/dist-packages/bottle_websocket/server.py", line 17, in run
    server.serve_forever()
  File "/usr/local/lib/python2.7/dist-packages/gevent/baseserver.py", line 367, in serve_forever
    self.start()
  File "/usr/local/lib/python2.7/dist-packages/gevent/baseserver.py", line 305, in start
    self.init_socket()
  File "/usr/local/lib/python2.7/dist-packages/gevent/pywsgi.py", line 1490, in init_socket
    StreamServer.init_socket(self)
  File "/usr/local/lib/python2.7/dist-packages/gevent/server.py", line 146, in init_socket
    self.socket = self.get_listener(self.address, self.backlog, self.family)
  File "/usr/local/lib/python2.7/dist-packages/gevent/server.py", line 157, in get_listener
    return _tcp_listener(address, backlog=backlog, reuse_addr=cls.reuse_addr, family=family)
  File "/usr/local/lib/python2.7/dist-packages/gevent/server.py", line 252, in _tcp_listener
    sock = GeventSocket(family=family)
  File "/usr/local/lib/python2.7/dist-packages/gevent/_socket2.py", line 123, in __init__
    self._sock = _realsocket(family, type, proto)

Something appears to die :P

Natrixz, 2019-03-08:
PS. I have to run the java -jar Denkovi.jar as sudo or it won't do as instructed and give a weird error

Natrixz, 2019-03-08:
It doesn't switch the relay yet, nothing happens so far.
Any logs you want me to show?

theyosh, 2019-03-09:
Hmm,

first make sure the modules ftdi_sio and usbserial are unloaded. `sudo rmmod usbserial` and `sudo rmmod ftdi_sio`.

Then try: java -jar DenkoviRelayCommandLineTool.jar [DEVICE SERIAL NUMBER or ID] [RELAY COUNT] [RELAY NUMBER] status

Where [DEVICE SERIAL NUMBER or ID] is 0
Where [RELAY COUNT] is 4 8 or 16
Where [RELAY NUMBER] is 1 for testing

And I have created new switch types. So make sure you use the right denkovi power switch type when adding to terrariumpi. And I also add some extra debug. So could you do a `git pull` and see what that does?

Natrixz, 2019-03-09:
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ java -jar DenkoviRelayCommandLineTool.jar 0
Error: One or more parameters misssing...
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ java -jar DenkoviRelayCommandLineTool.jar 0 status
Error: One or more parameters misssing...
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ java -jar DenkoviRelayCommandLineTool.jar 0 4v2 status
Error: One or more parameters misssing...
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ java -jar DenkoviRelayCommandLineTool.jar 0 4v2 status 0
Error: This serial number is too short. MCP Relay Boards serial numbers are 10 symbols!
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ java -jar DenkoviRelayCommandLineTool.jar 0 4v2 0 status
Error: This serial number is too short. MCP Relay Boards serial numbers are 10 symbols!
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ java -jar DenkoviRelayCommandLineTool.jar 0001691995 4v2 0 status
Error: Relay Board with this serial number or ID is not found!
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ java -jar DenkoviRelayCommandLineTool.jar 0001691995 4v2 status
Error: One or more parameters misssing...
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ sudo java -jar DenkoviRelayCommandLineTool.jar 0001691995 4v2 status
Error: One or more parameters misssing...
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ sudo java -jar DenkoviRelayCommandLineTool.jar 0001691995 4v2 status 0
Error: Invalid command type!
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ sudo java -jar DenkoviRelayCommandLineTool.jar 0001691995 4v2 0 status
0
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ sudo java -jar DenkoviRelayCommandLineTool.jar 0 4v2 0 status
Error: This serial number is too short. MCP Relay Boards serial numbers are 10 symbols!
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ sudo java -jar DenkoviRelayCommandLineTool.jar 0001691995 4v2 0 status

theyosh, 2019-03-09:
Not sure how to read all this..  So then I need to know what the working command is for you. So could you just try according this page to get it working: http://denkovi.com/denkovi-relay-command-line-tool

So that you can pass the right command to me, and then I try to incorperate it.

Natrixz, 2019-03-09:
The working command for me is;
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ sudo java -jar DenkoviRelayCommandLineTool.jar 0001691995 4v2 0 status
0
That returns the status as 0 when it's off and as 1 when it's one

Using;
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ sudo java -jar DenkoviRelayCommandLineTool.jar 0001691995 4v2 0 1
Will turn switch 1 on
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ sudo java -jar DenkoviRelayCommandLineTool.jar 0001691995 4v2 0 0
will turn switch 1 off

theyosh, 2019-03-09:
could you also paste the output of the status where you can read out the serial id? Would help me for parsing the right data

Natrixz, 2019-03-09:
sudo java -jar DenkoviRelayCommandLineTool.jar list
Serial number [Description] [ID]
FTDI devices:
MCP devices:
0001691995 [Microchip Technology Inc. MCP2200 USB Serial Port Emulator null] [id=0]

theyosh, 2019-03-09:
Thanks a lot. And is sudo always needed :( ?

Natrixz, 2019-03-09:
Sadly; yes :(
Otherwise it gives some weird error

theyosh, 2019-03-09:
And I see that you also enter '4v2' where the documentation says that '4' should be enough. Could you check this?

Natrixz, 2019-03-09:
Natrixz@terrarium02:~/TerrariumPI/DenkoviRelayCommandLineTool $ sudo java -jar DenkoviRelayCommandLineTool.jar 0001691995 4 0 status
Error: FTDI Relay Boards serial numbers are exact 8 symbols. Please check if this serial number is MCP relay board!

4v2 is always needed for the v2's is appears

theyosh, 2019-03-09:
New code, new round, new bugs :P

Natrixz, 2019-03-09:
2019-03-09 16:19:31,458 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
Set power switch state cmd:
['sudo', 'java', '-jar', 'DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar', '0', '4', '1', '0']
[sudo] password for natrixz:
Set state done!
Set power switch state cmd:
['sudo', 'java', '-jar', 'DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar', '0', '4', '2', '0']
Set state done!
Set power switch state cmd:
['sudo', 'java', '-jar', 'DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar', '0', '4', '3', '0']
Set state done!
Set power switch state cmd:
['sudo', 'java', '-jar', 'DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar', '0', '4', '4', '0']
Set state done!
2019-03-09 18:47:33,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
global name 'address' is not defined
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 766, in gevent._greenlet.Greenlet.run
  File "/home/nick/TerrariumPI/terrariumEngine.py", line 561, in __engine_loop
    display_message = ['%s %s' % (_('Uptime'),terrariumUtils.format_uptime(system_data['uptime']),),
UnboundLocalError: local variable 'system_data' referenced before assignment
2019-03-09T18:47:33Z <Greenlet at 0x6c2d0a48: <bound method terrariumEngine.__engine_loop of <terrariumEngine.terrariumEngine object at 0x76a1b9d0>>> failed with UnboundLocalError

I had to fill in my sudo password but it still throws the errors

theyosh, 2019-03-09:
Ok, what will happen when you add the following line to: `/etc/sudoers.d/terrariumpi`

`pi ALL=(ALL) NOPASSWD: java -jar DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar`

And reboot. I hope this will make it run with sudo without password

Natrixz, 2019-03-09:
Crap, I pasted it in too quickly and didn't notice the full path missing...have to fix my sudoers.d file now lol

theyosh, 2019-03-09:
Sorry, I had not tested it. So I fixed in my installer now. And it seems that my old hardware is also supported. So I can do some testing my self... but as you said that it is already working when sudo password is added, then I expect that with the fixed sudoers file, that it should work.

Natrixz, 2019-03-09:
Yeh it will take me a bit to fix it, damnit haha
(I never set a root password so it's impossible to login with root -,-)
Need to boot it on my linux system and fix the sudo'ers file by hand lol

Natrixz, 2019-03-09:
Alright; it's fixed, let's see if it works with the correct line in the file =)

Natrixz, 2019-03-09:
It's still asking for a password lol

Natrixz, 2019-03-09:
Alright, I see the issue;
It's calling it as root, so can you remove the sudo from the commandline? =)
Let's try that

theyosh, 2019-03-09:
Ok, this will take some more time then... strange.. and my device is not supported. But I also saw this:

Error: Invalid relay board type! Currently supported types are: 4 - for USB 4 Relay FT245RL chipset, 8 - for USB 8 Relay FT245RL chipset, 16 - for USB 16 Relay FT232RL chipset, 4v2 - for USB 4 Relay MCP2200 chipset and 4v8 for USB 8 Relay MCP2200 chipset!

That explains the 4v2 value

Natrixz, 2019-03-09:
Yeh indeed, i stumbled across that too :P 
Weird it still asks for the password tho...let's see if I can figure out why

theyosh, 2019-03-09:
I have an idea.... but again... playing with sudoers is tricky.

1. You need to start the java tool with root rights, therefore sudo.
2. The terrariumpi software is running as normal user (pi).
3. So pi needs to do a sudo command without a password. Then I will get to:

`/usr/bin/sudo /usr/bin/java -jar DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar`

And in my code, I have to add a full path to sudo. And then I am wondering if more parameters on the command line will work...

Natrixz, 2019-03-09:
Nope that doesn't work, I tried it lol

theyosh, 2019-03-09:
Add '*' to the command line after .jar. That worked for me

theyosh, 2019-03-09:
`pi  ALL=(ALL) NOPASSWD: /usr/bin/java -jar DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar *`

Natrixz, 2019-03-09:
Yep that works.
just adding /usr/bin/java works too but is very open in case the Pi does ever get compromised

theyosh, 2019-03-09:
So, another update. The installer will add new lines to the sudoers file. So that should be fixed.

And it should also see the different boards using v2 if needed. Could you test that? Just update the code with git pull and restart.

Natrixz, 2019-03-09:
Pulled the latest update and restarted; so far so good and it appears to work fine!

Natrixz, 2019-03-09:
Quick question; isn't it faster/easier to do denkovi.jar all status
so you get all the status back at once and then take out the correct reading per switch?
(Then you don't have to do 4/8/16 individual status pulls)

theyosh, 2019-03-09:
It is faster to read out once, but the problem then rises that I have to cache and share the data between different objects in code. I am currently doing that for sensors, but that is something I can test. And adding caching without testing is hard. But of course, it is way more efficient to do that. Main goal is now to get it working on the most easiest way to implement.

So I just did another code update. Could you do a git pull once more and restart?

Natrixz, 2019-03-09:
Still works fine =)

theyosh, 2019-03-09:
yes :). But that will not be today I think. I would like to think a bit more about how to do that.

Natrixz, 2019-03-09:
No worries!
I will leave it hooked up to one of my test Pi's =)

Already happy that it works like it does now ;)

theyosh, 2019-03-10:
Added status caching... untested :)

Natrixz, 2019-03-10:
It's currently not switching the relay's anymore, so appears broken :P

theyosh, 2019-03-11:
I added some debug code again. But you should see some errors....?

Natrixz, 2019-03-11:
Running set hardware state cmd: ['/usr/bin/sudo', '/usr/bin/java', '-jar', 'DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar', '0001691995', '4v2', '0', '1']
Get hardware state cache data
Get cache key 09b28e73b518f69edbb275674f0b230a
None
Get cache key 09b28e73b518f69edbb275674f0b230a
IS running: False
Get cache key 09b28e73b518f69edbb275674f0b230a
Get cache key 09b28e73b518f69edbb275674f0b230a
Running cmd: ['/usr/bin/sudo', '/usr/bin/java', '-jar', 'DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar', '0001691995', '4v2', 'all', 'status']
Clear caching
Get cache key 09b28e73b518f69edbb275674f0b230a
Clear caching DONE!
Got data: *0000*

Natrixz, 2019-03-11:
Get cache key 09b28e73b518f69edbb275674f0b230a
Get cache key 09b28e73b518f69edbb275674f0b230a
Final state data at address : 1
Running set hardware state cmd: ['/usr/bin/sudo', '/usr/bin/java', '-jar', 'DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar', '0001691995', '4v2', '0', '1']
Clear caching
Get cache key 09b28e73b518f69edbb275674f0b230a
Clear caching DONE!
Get hardware state cache data
Get cache key 09b28e73b518f69edbb275674f0b230a
None
Get cache key 09b28e73b518f69edbb275674f0b230a
IS running: False
Get cache key 09b28e73b518f69edbb275674f0b230a
Get cache key 09b28e73b518f69edbb275674f0b230a
Running cmd: ['/usr/bin/sudo', '/usr/bin/java', '-jar', 'DenkoviRelayCommandLineTool/DenkoviRelayCommandLineTool.jar', '0001691995', '4v2', 'all', 'status']
Got data: *1000*

theyosh, 2019-03-11:
Euh, I did change that the entered address is lowered by one. I expect that values from 1 till 4 including is entered. And the code uses 0 for first switch. At least that is the documentation saying... you experience differently?

Natrixz, 2019-03-11:
Well for some reason the address layout is all messed up....
I thought it was caching but clearing the cache didn't help.
Switching address 
0 will switch 3
1 will switch 1
2 will switch 1
3 will switch 2

theyosh, 2019-03-11:
Ok, changed some back. And saw some strange issue with printing. So could be that it will work now again.

I am looking for the line: 'Final state data at address' and then some numbers.

theyosh, 2019-03-11:
But, could you use the numbers 1 till 4? And not starting at zero? Because starting at 1 is more human and more understandable for non tech people

Natrixz, 2019-03-11:
Order is correct now, BUT it's still switching off after a few seconds

theyosh, 2019-03-11:
Could you please post a full run debug? Like a screenshot? Because all this small pieces off info is confusing... And I need the debug code to see what is happening.

Natrixz, 2019-03-11:
[image]
[image]
Im not clicking switch number 2

theyosh, 2019-03-11:
YESH! En we doen het lekker in NL :P Alleen de eerste power switch gaf de juiste status terug. Switch 2 t/m 4 dus niet, en dat geeft rare effecten. Heb dat nu aangepast en dat zou nu beter moeten werken.

Natrixz, 2019-03-11:
Oke is ff kijken wat er gebeurt na een git pull =)

Natrixz, 2019-03-11:
Alle schakelaars functioneren nu zoals het hoort en ze blijven ingeschakeld =)

theyosh, 2019-03-11:
Mooi.... en er zit nu dus ook caching in van de statussen... 1 keer per 30 seconden worden deze opgehaald tenzij er geklikt wordt. Extern schakelen duurt 30 seconden voordat dat opgepakt wordt nu

theyosh, 2019-03-11:
Ah, ik weet niet hoe veel tijd het kost om dat java te starten.... ik zal ff wat timer code toevoegen

Natrixz, 2019-03-11:
Als ik extern schakel zie ik dat de software ook nog een keer 2 0, 1 0, etc uitvoert.
Kun je om tijd te besparen niet gewoon de cache leeg gooien en invullen met de huidige waarde?
(of een optie om het gewoon keihard te overrulen en terug te zetten naar de value zoals in de cache?)

theyosh, 2019-03-11:
Ok, er zit nu een update timer in... bij mij duurt het 5 sec voordat ik een error krijg dat mijn board niet ondersteund is.... maar dat is wel stevig... 5 sec

Natrixz, 2019-03-11:
Java meuk....ik git pull nu :)

Natrixz, 2019-03-11:
Krijg overigens een hele berg errors na een restart
2019-03-11 20:42:05,108 - ERROR   - terrariumSwitch      - Error setting hardware state for switch type denkovi_v2_4, with error: 'terrariumPowerSwitchDenkoviV2_4' object has no attribute '_terrariumPowerSwitchDenkoviV2__cache'
2019-03-11 20:42:11,258 - ERROR   - terrariumSwitch      - Error setting hardware state for switch type denkovi_v2_4, with error: 'terrariumPowerSwitchDenkoviV2_4' object has no attribute '_terrariumPowerSwitchDenkoviV2__cache'
2019-03-11 20:42:17,408 - ERROR   - terrariumSwitch      - Error setting hardware state for switch type denkovi_v2_4, with error: 'terrariumPowerSwitchDenkoviV2_4' object has no attribute '_terrariumPowerSwitchDenkoviV2__cache'
2019-03-11 20:42:23,619 - ERROR   - terrariumSwitch      - Error setting hardware state for switch type denkovi_v2_4, with error: 'terrariumPowerSwitchDenkoviV2_4' object has no attribute '_terrariumPowerSwitchDenkoviV2__cache'
2019-03-11 20:42:29,698 - ERROR   - terrariumSwitch      - Error setting hardware state for switch type denkovi_v2_4, with error: 'terrariumPowerSwitchDenkoviV2_4' object has no attribute '_terrariumPowerSwitchDenkoviV2__cache'
2019-03-11 20:42:29,000 - INFO    - terrariumWebserver   - Running webserver at :::8090

theyosh, 2019-03-11:
hmmm... nieuwe pull

Natrixz, 2019-03-11:
Errors zijn weg, na een reboot gaan alle switches terug naar "uit" status

theyosh, 2019-03-11:
Dat is default... bij het opstarten gaat alles uit. En dan moeten ze de status aannemen op basis van de instellingen. Maar het werkt nu toch nog steeds?

Natrixz, 2019-03-11:
Oke dan heb ik niets gezegd xD
Jep alles werkt nu prima! =)

[9 more replies]

closed 2019-03-15 by Natrixz