theyosh/TerrariumPI #276: Meross MSS425E

Ultameciia, 2019-03-10:
Went to add a Meross MSS435E today and I'm having some issues. I've got the plugs up and running in the app and they work like a dream. However bringing it into TerrariumPI is a bit rough. I've added my login credentials to the settings page and according to #254 I should reboot the software and see the switches correct? After rebooting I see no switches. Nothing in screen -r shows anything happening with authentication or the switches. 

If I try to manually add it I select mss435e and for address use 1, 2, or 3 and get this:

`Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 46, in webserver_headers
    return fn(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 88, in wrapper
    return func(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 282, in __update_api_call
    result['ok'] = self.__terrariumEngine.set_config(path,postdata,request.files)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1156, in set_config
    update_ok = self.set_switches_config(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 740, in set_switches_config
    self.__load_power_switches(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 293, in __load_power_switches
    callback=self.toggle_power_switch)
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 753, in __new__
    raise terrariumPowerSwitchTypeException('Power switch of type {} is unknown. We cannot controll this power switch.'.`

I've git pulled and updated so I'm as current as I can be but still no luck.

I've also tried using sudo ./start.sh 3 but that throws an error and TerrariumPi never actually starts. I'm not sure if that command is depreciated or points to a bigger issue with my install. The error when I try running sudo ./start.sh 3 is:

`Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 17, in <module>
    import terrariumLogging
  File "/home/pi/TerrariumPI/terrariumLogging.py", line 17, in <module>
    from terrariumNotification import terrariumNotification
  File "/home/pi/TerrariumPI/terrariumNotification.py", line 4, in <module>
    import RPi.GPIO as GPIO
ImportError: No module named 'RPi'
`

I've also checked to see if python-rpi.gpio was installed and it is.

theyosh, 2019-03-11:
Hi,

Yes, you need to install and run everything in Python3 to get this working: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35

So rerun the installer with parameter 3 before starting it with parameter 3.

Ultameciia, 2019-03-11:
Oh man, I completely missed that. If that paragraph was a snake itd have bit me. Good thing I keep geckos. Ill mark it closed.

Ultameciia, 2019-03-11:
Ran into some issues with Python 3.5, figured I'd just reopen this instead of make a new one. After attempting to install the python 3.5 modules using sudo ./install.sh 3 I'm getting the following errors. 

`2019-03-11 16:29:32,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.0.0.79:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 17, in <module>
    import terrariumLogging
  File "/home/pi/TerrariumPI/terrariumLogging.py", line 86, in <module>
    logging.config.fileConfig('logging.cfg')
  File "/usr/lib/python3.5/logging/config.py", line 84, in fileConfig
    handlers = _install_handlers(cp, formatters)
  File "/usr/lib/python3.5/logging/config.py", line 148, in _install_handlers
    h = klass(*args)
  File "/home/pi/TerrariumPI/terrariumLogging.py", line 76, in __init__
    terrariumpi_config = terrariumConfig()
  File "/home/pi/TerrariumPI/terrariumConfig.py", line 47, in __init__
    self.__upgrade_config(version)
  File "/home/pi/TerrariumPI/terrariumConfig.py", line 55, in __upgrade_config
    if int(self.get_system()['version'].replace('.','')) >= current_version:
  File "/home/pi/TerrariumPI/terrariumConfig.py", line 517, in get_system
    data = self.__get_config('terrariumpi')
  File "/home/pi/TerrariumPI/terrariumConfig.py", line 499, in __get_config
    for config_part in self.__config.items(section):
  File "/usr/lib/python3.5/configparser.py", line 855, in items
    return [(option, value_getter(option)) for option in d.keys()]
  File "/usr/lib/python3.5/configparser.py", line 855, in <listcomp>
    return [(option, value_getter(option)) for option in d.keys()]
  File "/usr/lib/python3.5/configparser.py", line 852, in <lambda>
    section, option, d[option], d)
  File "/usr/lib/python3.5/configparser.py", line 393, in before_get
    self._interpolate_some(parser, option, L, value, section, defaults, 1)
  File "/usr/lib/python3.5/configparser.py", line 443, in _interpolate_some
    "found: %r" % (rest,))
configparser.InterpolationSyntaxError: '%' must be followed by '%' or '(', found: '%40group.calendar.google.com&ctz=America%2FChicago'
`

theyosh, 2019-03-12:
Hmm, it looks like the google calendar url is causing this issue. Quick fix is to remove it from the settings.

You can do that with `nano settings.cfg` and find the url 'external_calendar_url =' and make sure it is cleared after '='.

Then restart. It should work again. And I need some time to figure out why the google url gives problems. I see there are '%' chars in the url. That is certain a special case for the configuration file. So that needs to be looked at.

Ultameciia, 2019-03-12:
Ok, so without meross credentials I can now start TerrariumPi with sudo ./start.sh 3. However when I input my Meross credentials (made sure they were the right ones as well) I get the following errors.

`2019-03-12 14:23:55,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2019-03-12 14:23:55,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 21 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2019-03-12 14:24:06,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.0.0.79:8090 ...
2019-03-12 14:24:11,310 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2019-03-12 14:24:16,342 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 26, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 135, in __init__
    self.__load_power_switches()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 293, in __load_power_switches
    callback=self.toggle_power_switch)
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 751, in __new__
    return powerswitch(switch_id, address, name, prev_state, callback)
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 666, in __init__
    super(terrariumPowerSwitchMSS425E,self).__init__(switch_id, address[1], name, prev_state, callback)
IndexError: string index out of range
`

Ultameciia, 2019-03-12:
Checked settings.cfg and removed the MSS425E entry. This allowed me to boot TerrariumPi successfully.

theyosh, 2019-03-12:
Ok, just to be clear. To use MSS425E, only enter your meross credentials at the system settings. Then reboot the software and then it should have your powerswitches.

Do not manually add power switches to the system. The meross switches should be scanned during startup.

Ultameciia, 2019-03-12:
That's what I read in the other Meross thread. I feel like the scanning isn't happening for me at startup, there's no mention of any errors regarding meross in debug. Nor any mention of scanning. It checks OWFS, then bluetooth, then starts the webserver. 

I was only testing it manually to see what happened.

Ultameciia, 2019-03-15:
So I've been going back over this because I really would like to get this switch working. I verified that the meross_iot pip was installed. I thought maybe this was why they weren't loading. Even learned about about backwards compatibility with python 3.5 and 2.7 when I tried python -c "import meross_iot". Got really confused when it said it wasn't installed. Then realized I was an idiot and used the python3 command.

I went ahead and enabled debugging according to your steps and looked at what happens during boot. Everything initializes correctly, but no switches are ever added. I can see where it says "Starting up terrariumPi switches, but then I get nothing. After Loaded 0 excluded IDs: [] I can see it Loading previous power switch states. After that it says Done starting up terrariumPi switches. Found 0 switches in 6.779 seconds. 

I've got my credentials added. I've checked settings.cfg and they are there and correct. Is there any I can do to try to force them to load?

Ultameciia, 2019-03-15:
`2019-03-15 04:09:31,177 - INFO    - terrariumEngine      - Start terrariumPI engine log
2019-03-15 04:09:31,176 - INFO    - terrariumEngine      - Start terrariumPI webcams
2019-03-15 04:09:31,091 - WARNING - terrariumSensor      - Measured value 22.0F from dht11 sensor 'Ambient Temperature' is erratic compared to previous value 22.0F in 0.00010 seconds.
2019-03-15 04:09:31,016 - INFO    - terrariumEngine      - Start terrariumPI engine
2019-03-15 04:09:31,015 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-15 04:09:31,013 - INFO    - terrariumWebserver   - Running webserver at :::8090
2019-03-15 04:09:30,950 - INFO    - terrariumTranslations - Loaded TerrariumPI with 116 translations
2019-03-15 04:09:30,947 - INFO    - terrariumTranslations - Initialize TerariumPI translations
2019-03-15 04:09:30,920 - INFO    - terrariumEngine      - TerrariumPI engine is running
2019-03-15 04:09:30,919 - INFO    - terrariumAudio       - Player loaded with 0 audio files and 0 playlists. Starting engine.
2019-03-15 04:09:30,908 - INFO    - terrariumEngine      - Done loading terrariumPI webcams. Found 1 webcams in 0.398 seconds
2019-03-15 04:09:30,908 - INFO    - terrariumWebcam      - Done updating webcam 'Tokay Cam' at location http://73.6.73.185:8081/video.mjpg?q=30&fps=33&id=0.5390821830078989&r=1551709852200 in 0.38460 seconds
2019-03-15 04:09:30,541 - INFO    - terrariumEnvironment - Update done in 0.00714 seconds. Waiting for 14.99286 seconds for next update
2019-03-15 04:09:30,532 - INFO    - terrariumEnvironment - Starting engine
2019-03-15 04:09:30,521 - INFO    - terrariumWebcam      - Initialized remote webcam 'Tokay Cam' on location http://73.6.73.185:8081/video.mjpg?q=30&fps=33&id=0.5390821830078989&r=1551709852200
2019-03-15 04:09:30,510 - INFO    - terrariumEngine      - Loading terrariumPI webcams
2019-03-15 04:09:30,507 - INFO    - terrariumEnvironment - Done loading terrariumPI Environment 0.027 seconds
2019-03-15 04:09:30,480 - INFO    - terrariumEnvironment - Loading terrariumPI Environment system
2019-03-15 04:09:30,477 - INFO    - terrariumEngine      - Done loading terrariumPI doors. Found 0 doors in 0.001 seconds
2019-03-15 04:09:30,476 - INFO    - terrariumEngine      - Loading terrariumPI doors
2019-03-15 04:09:30,474 - INFO    - terrariumEngine      - Done starting up terrariumPI switches. Found 0 switches in 6.779 seconds
2019-03-15 04:09:23,698 - INFO    - terrariumEngine      - Loading previous power switch states from the last 2 minutes
2019-03-15 04:09:23,697 - INFO    - terrariumEngine      - Loaded 0 excluding IDs: []
2019-03-15 04:09:23,695 - INFO    - terrariumEngine      - Starting up terrariumPI switches
2019-03-15 04:09:23,694 - INFO    - terrariumEngine      - Done loading terrariumPI sensors. Found 2 sensors in 11.521 seconds
2019-03-15 04:09:23,691 - INFO    - terrariumSensor      - Updated dht11 sensor 'Ambient Temperature' from 0.00F to 22.00F in 0.00020 seconds
2019-03-15 04:09:23,690 - INFO    - terrariumSensor      - Loaded dht11 temperature sensor 'Ambient Temperature' on location 7.
2019-03-15 04:09:23,687 - INFO    - terrariumSensor      - Updated dht11 sensor 'Ambient Humidity' from 0.00% to 75.00% in 0.54366 seconds
2019-03-15 04:09:23,142 - INFO    - terrariumSensor      - Loaded dht11 humidity sensor 'Ambient Humidity' on location 7.
2019-03-15 04:09:23,127 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
2019-03-15 04:09:12,181 - INFO    - terrariumBluetoothSensor - Scanning 5 seconds for MiFlora bluetooth devices
2019-03-15 04:09:12,179 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2019-03-15 04:09:12,172 - INFO    - terrariumEngine      - Loading terrariumPI sensors
2019-03-15 04:09:12,172 - INFO    - terrariumEngine      - Done loading terrariumPI weather data
2019-03-15 04:09:12,171 - INFO    - terrariumWeather     - Done loading YR.no weather data in 1.457350492477417 seconds. Hour forcecast items: 48, week forecast items: 39.
2019-03-15 04:09:10,714 - INFO    - terrariumWeather     - Update YR.no data from ONLINE refreshing cache.
2019-03-15 04:09:10,712 - INFO    - terrariumWeather     - Initialising 'YR.no' weather object
2019-03-15 04:09:10,709 - INFO    - terrariumEngine      - Loading terrariumPI weather data
2019-03-15 04:09:10,709 - INFO    - terrariumEngine      - Done loading terrariumPI PI volume indicator
2019-03-15 04:09:10,707 - INFO    - terrariumEngine      - Loading terrariumPI PI volume indicator
2019-03-15 04:09:10,707 - INFO    - terrariumEngine      - Done loading terrariumPI PI windspeed indicator
2019-03-15 04:09:10,705 - INFO    - terrariumEngine      - Loading terrariumPI PI windspeed indicator
2019-03-15 04:09:10,705 - INFO    - terrariumEngine      - Done loading terrariumPI PI distance indicator
2019-03-15 04:09:10,703 - INFO    - terrariumEngine      - Loading terrariumPI PI distance indicator
2019-03-15 04:09:10,702 - INFO    - terrariumEngine      - Done loading terrariumPI PI temperature indicator
2019-03-15 04:09:10,701 - INFO    - terrariumEngine      - Loading terrariumPI PI temperature indicator
2019-03-15 04:09:10,700 - INFO    - terrariumEngine      - Done loading terrariumPI PI power setting
[... cut ...]
2019-03-15 04:09:10,658 - INFO    - terrariumEngine      - Done setting terrariumPI authentication
2019-03-15 04:09:10,654 - INFO    - terrariumEngine      - Setting terrariumPI authentication
2019-03-15 04:09:10,652 - INFO    - terrariumEngine      - Done Loading terrariumPI config
2019-03-15 04:09:10,628 - INFO    - terrariumEngine      - Loading terrariumPI config
2019-03-15 04:09:10,114 - INFO    - terrariumPI          - Starting terrariumPI`

Here is the relevant bits for startup. Everything after that is just updates.

theyosh, 2019-03-15:
Euh, it is a bit hard to figure this out due not having the hardware. So could you do a `git pull` which will add some more info logging.

And make sure you are running it with python 3. If you had installed it for python2 first, then the auto startup will still be at python 2. So look at `/etc/rc.local` file and make sure that the TerrariuPI startup file has the value '3' at the end to make sure you are using Python3.

When running with Python2, the Meross cloud software will be ignored.

Ultameciia, 2019-03-15:
I've modified my rc.local to run the start.sh 3 command. After doing a git pull I see Logged into Meross Cloud Successfully in the log. But then it still finds 0 switches.

`2019-03-15 15:09:56,967 - INFO    - terrariumEngine      - Done starting up terrariumPI switches. Found 0 switches in 6.110 seconds
2019-03-15 15:09:55,961 - INFO    - terrariumSwitch      - Logged into Meross cloud successfull.
2019-03-15 15:09:50,860 - INFO    - terrariumEngine      - Loading previous power switch states from the last 2 minutes
2019-03-15 15:09:50,859 - INFO    - terrariumEngine      - Loaded 0 excluding IDs: []
2019-03-15 15:09:50,857 - INFO    - terrariumEngine      - Starting up terrariumPI switches`

theyosh, 2019-03-15:
Hmmm, I see it now that you mention: MSS435E that version is not supported yet. Only MSS425E is at the moment. And I cannot find MSS435E so I can you describe your device? Because now it will be ignored

Ultameciia, 2019-03-15:
Oh woops, that's a typo in the first post. It is a MSS425E. 

https://www.amazon.com/gp/product/B07DMFZGVL/ref=ppx_yo_dt_b_asin_title_o02_s01?ie=UTF8&psc=1

That's the exact one I ordered and use currently.

theyosh, 2019-03-15:
Ok, could you do another git pull and restart? It should produce a lot of output in the screen session, I need that data...

Ultameciia, 2019-03-15:
`
2019-03-15 15:53:27,604 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
2019-03-15 15:53:27,182 - INFO    - terrariumEnvironment - Update done in 0.01540 seconds. Waiting for 14.98460 seconds for next update
2019-03-15 15:53:12,177 - INFO    - terrariumEnvironment - Update done in 0.01408 seconds. Waiting for 14.98592 seconds for next update
2019-03-15 15:53:08,589 - INFO    - terrariumEngine      - Update done in 10.98456 seconds. Waiting for 19.01544 seconds for next update
2019-03-15 15:53:08,427 - WARNING - terrariumSensor      - Measured value 25.0F from dht11 sensor 'Ambient Temperature' is erratic compared to previous value 25.0F in 0.53049 seconds.
2019-03-15 15:53:07,774 - WARNING - terrariumSensor      - Measured value None% from dht11 sensor 'Ambient Humidity' is outside valid range 0.00% - 100.00% in 10.12045 seconds.
2019-03-15 15:52:58,286 - INFO    - terrariumEngine      - Webcam update(s) done in 0.10098 seconds. Waiting for 29.89902 seconds for next update
2019-03-15 15:52:57,602 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
2019-03-15 15:52:57,174 - INFO    - terrariumEnvironment - Update done in 0.01410 seconds. Waiting for 14.98590 seconds for next update
2019-03-15 15:52:42,158 - INFO    - terrariumEnvironment - Update done in 0.01439 seconds. Waiting for 14.98561 seconds for next update
2019-03-15 15:52:30,966 - INFO    - terrariumEngine      - Update done in 3.36500 seconds. Waiting for 26.63500 seconds for next update
2019-03-15 15:52:30,805 - WARNING - terrariumSensor      - Measured value 25.0F from dht11 sensor 'Ambient Temperature' is erratic compared to previous value 25.0F in 3.05665 seconds.
2019-03-15 15:52:28,276 - INFO    - terrariumEngine      - Webcam update(s) done in 0.62787 seconds. Waiting for 29.37213 seconds for next update
2019-03-15 15:52:27,649 - INFO    - terrariumEngine      - Start terrariumPI engine log
2019-03-15 15:52:27,648 - INFO    - terrariumEngine      - Start terrariumPI webcams
2019-03-15 15:52:27,632 - WARNING - terrariumSensor      - Measured value 95.0% from dht11 sensor 'Ambient Humidity' is erratic compared to previous value 95.0% in 0.00010 seconds.
2019-03-15 15:52:27,600 - INFO    - terrariumEngine      - Start terrariumPI engine
2019-03-15 15:52:27,599 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2019-03-15 15:52:27,597 - INFO    - terrariumWebserver   - Running webserver at :::8090
2019-03-15 15:52:27,533 - INFO    - terrariumTranslations - Loaded TerrariumPI with 116 translations
2019-03-15 15:52:27,531 - INFO    - terrariumTranslations - Initialize TerariumPI translations
2019-03-15 15:52:27,510 - INFO    - terrariumEngine      - TerrariumPI engine is running
2019-03-15 15:52:27,509 - INFO    - terrariumAudio       - Player loaded with 0 audio files and 0 playlists. Starting engine.
2019-03-15 15:52:27,501 - INFO    - terrariumEngine      - Done loading terrariumPI webcams. Found 1 webcams in 0.379 seconds
2019-03-15 15:52:27,500 - INFO    - terrariumWebcam      - Done updating webcam 'Tokay Cam' at location http://73.6.73.185:8081/video.mjpg?q=30&fps=33&id=0.5390821830078989&r=1551709852200 in 0.36986 seconds
2019-03-15 15:52:27,149 - INFO    - terrariumEnvironment - Update done in 0.00799 seconds. Waiting for 14.99201 seconds for next update
2019-03-15 15:52:27,139 - INFO    - terrariumEnvironment - Starting engine
2019-03-15 15:52:27,129 - INFO    - terrariumWebcam      - Initialized remote webcam 'Tokay Cam' on location http://73.6.73.185:8081/video.mjpg?q=30&fps=33&id=0.5390821830078989&r=1551709852200
2019-03-15 15:52:27,122 - INFO    - terrariumEngine      - Loading terrariumPI webcams
2019-03-15 15:52:27,118 - INFO    - terrariumEnvironment - Done loading terrariumPI Environment 0.032 seconds
2019-03-15 15:52:27,087 - INFO    - terrariumEnvironment - Loading terrariumPI Environment system
2019-03-15 15:52:27,084 - INFO    - terrariumEngine      - Done loading terrariumPI doors. Found 0 doors in 0.002 seconds
2019-03-15 15:52:27,082 - INFO    - terrariumEngine      - Loading terrariumPI doors
2019-03-15 15:52:27,080 - INFO    - terrariumEngine      - Done starting up terrariumPI switches. Found 0 switches in 6.599 seconds
2019-03-15 15:52:26,066 - INFO    - terrariumSwitch      - Logged into Meross cloud successfull.
2019-03-15 15:52:20,484 - INFO    - terrariumEngine      - Loading previous power switch states from the last 2 minutes
2019-03-15 15:52:20,483 - INFO    - terrariumEngine      - Loaded 0 excluding IDs: []
2019-03-15 15:52:20,481 - INFO    - terrariumEngine      - Starting up terrariumPI switches
2019-03-15 15:52:20,479 - INFO    - terrariumEngine      - Done loading terrariumPI sensors. Found 2 sensors in 13.189 seconds
2019-03-15 15:52:20,477 - INFO    - terrariumSensor      - Updated dht11 sensor 'Ambient Temperature' from 0.00F to 25.00F in 0.00019 seconds
2019-03-15 15:52:20,476 - INFO    - terrariumSensor      - Loaded dht11 temperature sensor 'Ambient Temperature' on location 7.
2019-03-15 15:52:20,473 - INFO    - terrariumSensor      - Updated dht11 sensor 'Ambient Humidity' from 0.00% to 95.00% in 8.12042 seconds
2019-03-15 15:52:12,351 - INFO    - terrariumSensor      - Loaded dht11 humidity sensor 'Ambient Humidity' on location 7.
[... cut ...]
2019-03-15 15:52:05,289 - INFO    - terrariumEngine      - Done setting terrariumPI authentication
2019-03-15 15:52:05,286 - INFO    - terrariumEngine      - Setting terrariumPI authentication
2019-03-15 15:52:05,284 - INFO    - terrariumEngine      - Done Loading terrariumPI config
2019-03-15 15:52:05,253 - INFO    - terrariumEngine      - Loading terrariumPI config
2019-03-15 15:52:04,821 - INFO    - terrariumPI          - Starting terrariumPI
2019-03-15 15:51:51,969 - INFO    - terrariumPI          - Shutdown terrariumPI`

Ultameciia, 2019-03-15:
Not any different. I did see this in screen -r however.

`/usr/local/lib/python3.5/dist-packages/gevent/pywsgi.py:908: ResourceWarning: unclosed file <_io.BufferedReader name='/home/pi/TerrariumPI/log/terrariumpi.log'>
  for data in self.result:
/home/pi/TerrariumPI/terrariumWebserver.py:218: ResourceWarning: unclosed file <_io.BufferedReader name='/home/pi/TerrariumPI/log/terrariumpi.log'>
`

theyosh, 2019-03-15:
euh, I need from screen -r. Logfile does not give the needed information. So, when you did a git pull, then use `screen -r` and press Ctrl+C to restart. That will give the needed info. Make you window as big as possible, as you cannot scroll in the screen session

Ultameciia, 2019-03-15:
`2019-03-15 16:01:09,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.0.0.79:8090 ...
2019-03-15 16:01:14,570 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2019-03-15 16:01:19,599 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
Devices
[]
2019-03-15 16:01:32,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
`

theyosh, 2019-03-15:
I am afraid you do not understand it. You are posting the logfile. That is useless. 

What do you see when you do a `screen -r` There should be the information needed

theyosh, 2019-03-15:
Or, if you trust me, you could change you Meross credentials, and mail them to me for testing at: github.com@theyosh.nl Then I can do testing with your hardware. But again, you have to trust me on this if you want to do this....

Ultameciia, 2019-03-15:
That was directly copied from screen -r. I'll post a screen shot so you can see. 

[image]

theyosh, 2019-03-15:
Ah, that says there are no devices in your account. Do you have added your Meross device to the cloud already. In other words, can you control it with your mobile phone?

Ultameciia, 2019-03-15:
Yes it's in use currently to control my MistKing and lights. I'll send you over the login credentials. Plug 2 is my mistking and plug 3 are lights. But plug 1 is free to test with.

theyosh, 2019-03-15:
Hmm, it looks like it is a different device unfortunately. When I read it out with some different code I get:

```[
  {
    "reservedDomain": "us-smart.meross.com",
    "channels": [
      {},
      {
        "type": "Switch",
        "devIconId": "device001",
        "devName": "Switch 1"
      },
      {
        "type": "Switch",
        "devIconId": "device001",
        "devName": "Mistking"
      },
      {
        "type": "Switch",
        "devIconId": "device001",
        "devName": "Lights"
      },
      {
        "type": "USB",
        "devIconId": "device001",
        "devName": "USB"
      }
    ],
    "uuid": "1811133407490729086134298f1d4717",
    "bindTime": 1552245253,
    "onlineStatus": 1,
    "skillNumber": "",
    "fmwareVersion": "2.1.2",
    "hdwareVersion": "2.0.0",
    "domain": "us-iot.meross.com",
    "region": "us",
    "userDevIcon": "",
    "deviceType": "sp425ew",
    "iconType": 1,
    "devIconId": "device020_us",
    "subType": "us",
    "devName": "Tokay Tank"
  }
]
```

It reports back as a **sp425ew** type. And I am using a 3rd party library and that does not support this device. So I cannot support it. 

You can change the password back to what you want. Unfortunately for now I cannot fix this.

Ultameciia, 2019-03-15:
Dang, it was marketed as an MSS425E. Looks like I was bamboozled. Ill open an issue on the meross_iot github and see if I can get it supported.

theyosh, 2019-03-15:
That is an option. And I think it could work, but he has the same problem as me, we do not have the hardware to test.

But, if you do, then it is handy to post the device output I posted above here. That is what it reports when I ask for the connected devices. It could be that it works like an existing one, and then it is only a couple lines of code.

But then again, I am not that surprised, it is made in Chine ;) From what I have seen from Chinese stuff, they made code just to work. And when there are changes, they hide them, and just say, nothing changed. But then they changed a lot under the hood. So they do not be that clear with versions numbers. 

Because if you can look at: https://user-images.githubusercontent.com/17450475/52535934-d11b9e00-2d5c-11e9-9305-3f5fcb44f037.png

You see there some mentions of timerules. So that device can be scheduled. I did not see that in the output of your device. And how about software updates?

But, I hope they can be supported. Would be nice

Ultameciia, 2019-03-15:
I feel that. I'm a Network Administrator by trade. I see crappy Chinese programming every day. It definitely makes my life more difficult. Ha.

It does have schedules. I use them currently to control the lights and mistking. It's similar to schedules on things like the Phillips Hue smart light bulbs. I'll get the issue made over there and then I'll report back if I get anywhere.

theyosh, 2019-03-20:
Ok, then for now I am closing this. If there is an update in the other python library, my software will pick it up with the installer. So then I should only add the special type to my software. So I would ask you then to re-open this issue when this is done.

Ultameciia, 2019-03-31:
He got back to me, however Im not sure where to proceed from here. Ill drop a link for you theyosh.

https://github.com/albertogeniola/MerossIot/issues/35

theyosh, 2019-04-01:
Hmmm, you have to edit a python file from his package. So, could you try the following:

`sudo nano /usr/local/lib/python3.5/dist-packages/meross_iot/device_factory.py`

and change:

```
    elif device_type.lower() == "mss425e":
        return Mss425e(token, key, user_id, **device_specs)
    else:
        return None
```

to:

```
    elif device_type.lower() == "mss425e":
        return Mss425e(token, key, user_id, **device_specs)
    elif device_type.lower() == "sp425ew":
        return Mss425e(token, key, user_id, **device_specs)
    else:
        return None
```

I added two new lines that will make your version working like the 'original' mss425e.

Then restart the software, and enter the meross crendentials and reboot once more. If this does not help, then there is more work needed from the merossiot library

Ultameciia, 2019-04-01:
I got somewhere making the edits to device_factory.py. I'm seeing the following when I boot TerrariumPi now, just no switches. Maybe it's an easy fix?

`Connected with result code {'session present': 0}
Subscribing to topics...
Succesfully subscribed!
Succesfully subscribed!
--> {"payload": {}, "header": {"timestamp": 1554156890, "from": "/app/193134-86e56921addcd9f7ffce8a6e61be0c43/subscribe", "messageId": "696755b60cbdb00f1c8cf602211979fd", "payloadVersion": 1, "namespace": "Appliance.System.All", "sign": "4c267ae78559201585f2b1f89c9a5526", "method": "GET"}}
/app/193134-86e56921addcd9f7ffce8a6e61be0c43/subscribe --> b'{"header":{"messageId":"696755b60cbdb00f1c8cf602211979fd","namespace":"Appliance.System.All","method":"GETACK","payloadVersion":1,"from":"/appliance/1811133407490729086134298f1d4717/publish","timestamp":1554156890,"timestampMs":320,"sign":"4c267ae78559201585f2b1f89c9a5526"},"payload":{"all":{"system":{"hardware":{"type":"sp425ew","subType":"us","version":"2.0.0","chipType":"mt7682","uuid":"1811133407490729086134298f1d4717","macAddress":"34:29:8f:1d:47:17"},"firmware":{"version":"2.1.2","compileTime":"2018/11/12 09:50:07 GMT +08:00","wifiMac":"60:3d:26:b3:e6:7e","innerIp":"10.0.0.17","server":"iot.meross.com","port":2001,"userId":193134},"time":{"timestamp":1554156890,"timezone":"America/Chicago","timeRule":[[1552204800,-18000,1],[1572764400,-21600,0],[1583654400,-18000,1],[1604214000,-21600,0],[1615708800,-18000,1],[1636268400,-21600,0],[1647158400,-18000,1],[1667718000,-21600,0],[1678608000,-18000,1],[1699167600,-21600,0],[1710057600,-18000,1],[1730617200,-21600,0],[1741507200,-18000,1],[1762066800,-21600,0],[1772956800,-18000,1],[1793516400,-21600,0],[1805011200,-18000,1],[1825570800,-21600,0],[1836460800,-18000,1],[1857020400,-21600,0]]},"online":{"status":1}},"digest":{"togglex":[{"channel":0,"onoff":1,"lmTime":1554116401},{"channel":1,"onoff":0,"lmTime":1553986762},{"channel":2,"onoff":0,"lmTime":1554151562},{"channel":3,"onoff":1,"lmTime":1554153712},{"channel":4,"onoff":0,"lmTime":1552249797}],"triggerx":[{"channel":2,"id":"7ymc4cbv5bqnnj1k","count":1}],"timerx":[{"channel":3,"id":"bc91e3heh65uwaas","count":1},{"channel":3,"id":"jx3a62o9ywrkhj4v","count":1},{"channel":2,"id":"775yhygnw7pg4gst","count":1},{"channel":2,"id":"lu1gx7xi60wd0xek","count":2},{"channel":2,"id":"8l1koq5fjkcwmzh4","count":1},{"channel":2,"id":"0noknbx5795bdky4","count":1},{"channel":2,"id":"domq18ga2i8s0xn1","count":1},{"channel":2,"id":"ya6cqwtzeyntqq47","count":1},{"channel":2,"id":"jgv2c737hvzgacsq","count":1}]}}}}'
Scan error in terrariumPowerSwitchMSS425E
name 'self' is not defined`

theyosh, 2019-04-03:
Hmm... I think I found it. Do a `git pull` and restart. That should at least make it start again.

Ultameciia, 2019-04-03:
It worked! Successfully added and I can control it.

Ultameciia, 2019-04-03:
Wait, spoke too soon. While I can control the switches now and they function. They only stay on for 4 or 5 seconds then turn off automatically. Here's the screen -r output during that time. 

`800,-18000,1],[1857020400,-21600,0]]},"online":{"status":1}},"digest":{"togglex":[{"channel":0,"onoff":0,"lmTime":1554323129},{"channel":1,"onoff":0,"lmTime":1554322619},{"channel":2,"onoff":0,"lmTime":1554322897},{"channel":3,"onoff":0,"lmTime":1554323129},{"channel":4,"onoff":0,"lmTime":1554322675}],"triggerx":[{"channel":2,"id":"7ymc4cbv5bqnnj1k","count":2}],"timerx":[{"channel":3,"id":"bc91e3heh65uwaas","count":2},{"channel":3,"id":"jx3a62o9ywrkhj4v","count":2},{"channel":2,"id":"775yhygnw7pg4gst","count":2},{"channel":2,"id":"lu1gx7xi60wd0xek","count":3},{"channel":2,"id":"8l1koq5fjkcwmzh4","count":2},{"channel":2,"id":"0noknbx5795bdky4","count":2},{"channel":2,"id":"domq18ga2i8s0xn1","count":2},{"channel":2,"id":"ya6cqwtzeyntqq47","count":2},{"channel":2,"id":"jgv2c737hvzgacsq","count":2}]}}}}'
--> {"payload": {}, "header": {"payloadVersion": 1, "sign": "efeac9ce00ba7746199f035202f58773", "namespace": "Appliance.System.All", "messageId": "1ca789ef8562dbf8f1f912d6cbcc57ee", "timestamp": 1554323156, "from": "/app/193134-86e56921addcd9f7ffce8a6e61be0c43/subscribe", "method": "GET"}}
/app/193134-86e56921addcd9f7ffce8a6e61be0c43/subscribe --> b'{"header":{"messageId":"1ca789ef8562dbf8f1f912d6cbcc57ee","namespace":"Appliance.System.All","method":"GETACK","payloadVersion":1,"from":"/appliance/1811133407490729086134298f1d4717/publish","timestamp":1554323156,"timestampMs":93,"sign":"efeac9ce00ba7746199f035202f58773"},"payload":{"all":{"system":{"hardware":{"type":"sp425ew","subType":"us","version":"2.0.0","chipType":"mt7682","uuid":"1811133407490729086134298f1d4717","macAddress":"34:29:8f:1d:47:17"},"firmware":{"version":"2.1.2","compileTime":"2018/11/12 09:50:07 GMT +08:00","wifiMac":"60:3d:26:b3:e6:7e","innerIp":"10.0.0.17","server":"iot.meross.com","port":2001,"userId":193134},"time":{"timestamp":1554323156,"timezone":"America/Chicago","timeRule":[[1552204800,-18000,1],[1572764400,-21600,0],[1583654400,-18000,1],[1604214000,-21600,0],[1615708800,-18000,1],[1636268400,-21600,0],[1647158400,-18000,1],[1667718000,-21600,0],[1678608000,-18000,1],[1699167600,-21600,0],[1710057600,-18000,1],[1730617200,-21600,0],[1741507200,-18000,1],[1762066800,-21600,0],[1772956800,-18000,1],[1793516400,-21600,0],[1805011200,-18000,1],[1825570800,-21600,0],[1836460800,-18000,1],[1857020400,-21600,0]]},"online":{"status":1}},"digest":{"togglex":[{"channel":0,"onoff":0,"lmTime":1554323129},{"channel":1,"onoff":0,"lmTime":1554322619},{"channel":2,"onoff":0,"lmTime":1554322897},{"channel":3,"onoff":0,"lmTime":1554323129},{"channel":4,"onoff":0,"lmTime":1554322675}],"triggerx":[{"channel":2,"id":"7ymc4cbv5bqnnj1k","count":2}],"timerx":[{"channel":3,"id":"bc91e3heh65uwaas","count":2},{"channel":3,"id":"jx3a62o9ywrkhj4v","count":2},{"channel":2,"id":"775yhygnw7pg4gst","count":2},{"channel":2,"id":"lu1gx7xi60wd0xek","count":3},{"channel":2,"id":"8l1koq5fjkcwmzh4","count":2},{"channel":2,"id":"0noknbx5795bdky4","count":2},{"channel":2,"id":"domq18ga2i8s0xn1","count":2},{"channel":2,"id":"ya6cqwtzeyntqq47","count":2},{"channel":2,"id":"jgv2c737hvzgacsq","count":2}]}}}}'
--> {"payload": {"togglex": {"onoff": 1, "channel": 3}}, "header": {"payloadVersion": 1, "sign": "ce1494e1559459f816d91b1dd6bf1908", "namespace": "Appliance.Control.ToggleX", "messageId": "85f145c8f73864c2cc793fd5541a35da", "timestamp": 1554323157, "from": "/app/193134-86e56921addcd9f7ffce8a6e61be0c43/subscribe", "method": "SET"}}
/app/193134-86e56921addcd9f7ffce8a6e61be0c43/subscribe --> b'{"header":{"messageId":"85f145c8f73864c2cc793fd5541a35da","namespace":"Appliance.Control.ToggleX","method":"SETACK","payloadVersion":1,"from":"/appliance/1811133407490729086134298f1d4717/publish","timestamp":1554323157,"timestampMs":100,"sign":"ce1494e1559459f816d91b1dd6bf1908"},"payload":{}}'
--> {"payload": {}, "header": {"payloadVersion": 1, "sign": "c9c9283c51971f7e9a0f38ac528f43ff", "namespace": "Appliance.System.All", "messageId": "03bdbce7c09500bf7947851a7fc52cdd", "timestamp": 1554323157, "from": "/app/193134-86e56921addcd9f7ffce8a6e61be0c43/subscribe", "method": "GET"}}
/app/193134/subscribe --> b'{"header":{"messageId":"fc27acdc0c030cb5898c3acc3f0888ce","namespace":"Appliance.Control.ToggleX","method":"PUSH","payloadVersion":1,"from":"/appliance/1811133407490729086134298f1d4717/publish","timestamp":1554323157,"timestampMs":204,"sign":"15b6ea4826533795df21ec545145ee12"},"payload":{"togglex":{"channel":3,"onoff":1,"lmTime":1554323157}}}'
UNKNOWN msg received by 1811133407490729086134298f1d4717
/app/193134-86e56921addcd9f7ffce8a6e61be0c43/subscribe --> b'{"header":{"messageId":"03bdbce7c09500bf7947851a7fc52cdd","namespace":"Appliance.System.All","method":"GETACK","payloadVersion":1,"from":"/appliance/1811133407490729086134298f1d4717/publish","timestamp":1554323157,"timestampMs":309,"sign":"c9c9283c51971f7e9a0f38ac528f43ff"},"payload":{"all":{"system":{"hardware":{"type":"sp425ew","subType":"us","version":"2.0.0","chipType":"mt7682","uuid":"1811133407490729086134298f1d4717","macAddress":"34:29:8f:1d:47:17"},"firmware":{"version":"2.1.2","compileTime":"2018/11/12 09:50:07 GMT +08:00","wifiMac":"60:3d:26:b3:e6:7e","innerIp":"10.0.0.17","server":"iot.meross.com","port":2001,"userId":193134},"time":{"timestamp":1554323157,"timezone":"America/Chicago","timeRule":[[1552204800,-18000,1], [... cut ...] mpMs":781,"sign":"b8d88a30cb22ea1847a884ee2866cc65"},"payload":{}}'
/app/193134/subscribe --> b'{"header":{"messageId":"ab7224712aae0072fc842e445d5d8745","namespace":"Appliance.Control.ToggleX","method":"PUSH","payloadVersion":1,"from":"/appliance/1811133407490729086134298f1d4717/publish","timestamp":1554323159,"timestampMs":322,"sign":"2063df7e281aacf7923af8e0e45fc520"},"payload":{"togglex":{"channel":3,"onoff":0,"lmTime":1554323158}}}'
UNKNOWN msg received by 1811133407490729086134298f1d4717
`

Ultameciia, 2019-04-08:
@theyosh Looks like the Meross IOT library was updated to support any meross device on the market. Should be good news for the users here. The owner pinged me the other day on my open issue there.

https://github.com/albertogeniola/MerossIot/issues/35

theyosh, 2019-04-08:
Ah cool! That means if you want to test it, you should run the command:

`sudo pip3 install meross_iot --upgrade`

And that should then update the meross code and should/could work. I cannot test it. Make sure you restart my terrariumpi software after the update

theyosh, 2019-04-08:
Hmmm, I see that will not work without code change from my side.... Will have to look into it a bit more. But if you could just try to update and see what happens..

Ultameciia, 2019-04-08:
I'll give it a shot, I'm at work so this gives me a reason to ignore this Clearpass server I'm working on.

Ultameciia, 2019-04-08:
It is recognized and adds the surge protector just fine. However every 20 seconds or so it turns off all plugs. I've tried it in manual mode and with a timer set to it.

theyosh, 2019-04-08:
Hmm, I think you should not use any timers for now. Also clear the timers that you have created with the app on the phone. So you can test it clean, and know for sure that there should nothing interfering. Then debugging will be easier.

Ultameciia, 2019-04-08:
Ill give it a go and report back. If anything else I can always reshare credentials if you need something to test with.

Ultameciia, 2019-04-08:
With all routines disabled on the app I'm still seeing the switches shutting off after being on for a short period of time. In the log I can see where it's turned on, but then when the system updates the switch it turns everything back to the off state. Here's a bit of the log that shows where I turned it on and then the update portion. When it updates that switch it turns it off.

`019-04-08 17:03:02,297 - INFO    - terrariumEngine      - Update done in 14.99346 seconds. Waiting for 15.00654 seconds for next update
2019-04-08 17:03:02,159 - INFO    - terrariumSwitch      - Updated mss425e power switch 'USB' status in 0.45929 seconds
2019-04-08 17:03:01,598 - INFO    - terrariumSwitch      - Updated mss425e power switch 'Empty' status in 0.46770 seconds
2019-04-08 17:03:01,028 - INFO    - terrariumSwitch      - Updated mss425e power switch 'Mistking' status in 1.47044 seconds
2019-04-08 17:02:59,456 - INFO    - terrariumSwitch      - Updated mss425e power switch 'Lights' status in 0.67834 seconds
2019-04-08 17:02:59,454 - INFO    - terrariumSwitch      - Power switch changed from True state to new False state
2019-04-08 17:02:59,414 - INFO    - terrariumSwitch      - Changed power switch 'Lights' of type 'mss425e' at address '3' from state 'True' to state 'False' (Forced:False)
`

theyosh, 2019-04-10:
Hmm, I have no idea at the moment. It could be that reading the current state does not work, and then my software will shutdown the power switch as a default. But to know for sure, it needs to be debugged. Which is a bit hard.

Maybe I have time in the weekend to do some testing. If you are willing to change the login back to the one you sent me, I can do some testing at Saturday or Sunday.

Ultameciia, 2019-04-10:
Yeah i can change them. Ill change it back to the credentials I sent you before if itll let me change back to the previous password.

theyosh, 2019-04-14:
I think I found it.. just 1 line of code :( So do a git pull, and change the login details back to yours. It should now work also for you.

Ultameciia, 2019-04-14:
Man, gotta love it when it's just one little line of code. Great work, it's all working perfectly here!

theyosh, 2019-04-16:
If it is still working, maybe you should confirm that also at the meross_iot library. And then we can close this issue once more :P

closed 2019-04-22 by theyosh