theyosh/TerrariumPI #278: Assistance needed

EoinH, 2019-03-27:
Hi,

I've been successfully using this system for a number of months and all has been well.  Then a few days ago the web interface on the pi stopped responding and I have not been able to restore it.  I can still ssh into the pi.  The FAQ says that for debugging that i should run 'screen -r' to get to a new terminal however it just says "there is no screen to be resumed".

I'm not that technically proficient in Linux yet so I'm a but stuck for further troubleshooting steps.  I tried doing an upgrade over the top to see if that resolved anything and I've checked that none of the disks are full but I'm at a loss as to how to get it back up and running without wiping the whole thing and starting from scratch.  

Any assistance would be greatly appreciated!

Thanks,
Eoin,

Ultameciia, 2019-03-28:
If screen -r says there's nothing there it's most likely that terrariumPi isn't running.

From the terminal type cd TerrariumPI

Then type sudo ./start.sh (if you're using python 3 then it'd be sudo ./start.sh 3)

You'll see it telling you it's starting then type screen -r to see the log.

EoinH, 2019-03-28:
Hi and thanks for the reply.  When I try this step it starts the installer
which I follow through and reboot and then it's the same again....

Ultameciia, 2019-03-29:
Running start.sh runs the installer? That's a bit odd. Hmm

theyosh, 2019-03-29:
Yes, the start command will trigger the installer when there is an update. So that can be expected. But that should only happen once.

So, the following commands will do the same as the installer, but then manual:

run `sudo ./install.sh` as pi user. And see if there are errors during the installation. If not, just reboot.
If there are errors, rerun the installer. It should be able to install without errors. Make sure your network is working as it is needed to install the modules.

If after the reboot, there is still no screen session with `screen -r` then you could run the software manual with the command `python terrariumPI.py` as normal user. That should give you some more information.

EoinH, 2019-04-01:
Thanks again for the reply!  So I have followed those steps and went
through the installer again and there were no errors.  So after the reboot
screen -r had the same result so I ran the python script as a normal user.
After a moment it came back with the following response.
Traceback (most recent call last):
  File "terrariumPI.py", line 17, in <module>
    import terrariumLogging
  File "/home/pi/TerrariumPI/terrariumLogging.py", line 17, in <module>
    from terrariumNotification import terrariumNotification
  File "/home/pi/TerrariumPI/terrariumNotification.py", line 22, in <module>
    import twitter
  File "/usr/local/lib/python2.7/dist-packages/twitter/__init__.py", line
55, in <module>
    from .api import Api                        # noqa
  File "/usr/local/lib/python2.7/dist-packages/twitter/api.py", line 31, in
<module>
    from requests_oauthlib import OAuth1, OAuth2
  File
"/usr/local/lib/python2.7/dist-packages/requests_oauthlib/__init__.py",
line 5, in <module>
    from .oauth2_auth import OAuth2
  File
"/usr/local/lib/python2.7/dist-packages/requests_oauthlib/oauth2_auth.py",
line 2, in <module>
    from oauthlib.oauth2 import WebApplicationClient, InsecureTransportError
  File
"/usr/local/lib/python2.7/dist-packages/oauthlib/oauth2/__init__.py", line
17, in <module>
    from .rfc6749.endpoints import AuthorizationEndpoint
  File
"/usr/local/lib/python2.7/dist-packages/oauthlib/oauth2/rfc6749/endpoints/__init__.py",
line 13, in <module>sudo
    from .metadata import MetadataEndpoint
  File
"/usr/local/lib/python2.7/dist-packages/oauthlib/oauth2/rfc6749/endpoints/metadata.py",
line 22, in <module>
    from .. import grant_types
  File
"/usr/local/lib/python2.7/dist-packages/oauthlib/oauth2/rfc6749/grant_types/__init__.py",
line 8, in <module>
    from .authorization_code import AuthorizationCodeGrant
  File
"/usr/local/lib/python2.7/dist-packages/oauthlib/oauth2/rfc6749/grant_types/authorization_code.py",
line 16, in <module>
    from .base import GrantTypeBase
  File
"/usr/local/lib/python2.7/dist-packages/oauthlib/oauth2/rfc6749/grant_types/base.py",
line 12, in <module>
    from oauthlib.uri_validate import is_absolute_uri
ValueError: bad marshal data (digit out of range in long)

Thanks again for the assistance!
Eoin.

theyosh, 2019-04-01:
Hmm, looks like broken SD card, or to small SD card. I have seen those errors in the past, with only one solution. Reinstall your SD card and start over.

Make sure your SD card is 4GB or bigger.

EoinH, 2019-04-02:
Its a 16Gb card and i've checked that none of the partitions are full so maybe a corruption.  I will start fresh with a new card.  I'm now wishing there was a config backup/restore option!

theyosh, 2019-04-02:
Make a copy of the following files:

settings.cfg
history.db
description.txt
notifications.cfg

Those files can then be restored on a newly installed version.

EoinH, 2019-04-10:
So I finally got around to trying this option and it brought back to original problem where the screen session wouldn't start.  Anyway I deleted the 4 files and rebooted and it's back to normal.  I'll just put the settings back manually and it should be fine.  And I'll back up those 4 files regularly also :)  Thanks again for the help - I'd have been lost otherwise!

theyosh, 2019-04-22:
No problem. Made a FAQ out of it: https://github.com/theyosh/TerrariumPI/wiki/FAQ#which-files-to-backup

closed 2019-04-22 by theyosh