theyosh/TerrariumPI #279: MSS420F assistance,

wolfmaster74, 2019-03-30:
Hi I have just recieved The mss420f surger protector, i have setup up meross cloud and work with my phone, but cant get it to be detected by terrarium, i input user details and reboot and no switches appear i try manually addding and they will not work, i did try and debug but i dont see any meross error on the main logging page ? any ideas or other ways to test? 
many thanks loving the project

Ultameciia, 2019-03-31:
I don't believe the mss420f is supported. If you haven't setup TerrariumPI to use Python3.5 it won't work either. First setup TerrariumPi to use Python 3.5 using this https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35

But even then I don't think it'll work since I don't see that being supported.

theyosh, 2019-04-01:
@wolfmaster74 I am afraid it is not supported. I depend on https://github.com/albertogeniola/MerossIot

Maybe https://github.com/theyosh/TerrariumPI/issues/276#issuecomment-478746313 could also work for you.

theyosh, 2019-04-22:
As there is an update of the merrios_iot code, you could try to add this power surge as a 'MSS425E' type power switch. If that does not work, I cannot support it.

closed 2019-05-04 by theyosh