theyosh/TerrariumPI #280: Support Request: BrightPi Camera Light

fenyx13, 2019-04-04:
Hi TheYosh,
Would it be possible to control BrightPi from TerrariumPi?

BrightPi: https://uk.pi-supply.com/products/bright-pi-bright-white-ir-camera-light-raspberry-pi
BrightPi GitHub: https://github.com/PiSupply/Bright-Pi
Python library: https://github.com/jritter/brightpi

Thanks.

theyosh, 2019-04-22:
Hmm, the camera itself is not the issue. But how would you see this? Because the code can control all the leds individual. But I do not think I will support that. That will add to much extra code.

So, what I can think off, is that it has so toggle with the mail lights in opposite state. So light on, IR off. And lights off, IR on.

So, some out of the box thinking here. What if you add them as two independent dimmers switches? As the example code suggest, they can be dimmed. So then you have a Bright white and IR dimmer for the camera. And then at the environment page, you can use the bright white dimmer for daylight power switch, and the IR dimmer for the night power switches. As they are dimmers, they will go slowly on and off.

theyosh, 2019-05-16:
I ordered a few of them. So I can see what is possible.

theyosh, 2019-08-18:
I have added support for this. For now ONLY white lights or IR lights are supported.

Enter address value 70 for normal white. Or 70,ir for IR lights. Using both will give errors due to device sharing. I have to look into that.

theyosh, 2019-10-01:
Any update on this?

fenyx13, 2019-10-02:
Sorry for not replying earlier. In the middle of moving to a new place. 
I will find my PI tomorrow and test it out

theyosh, 2020-03-14:
Closing, due to no response. Can be opened any time when there are updates

closed 2020-03-14 by theyosh