theyosh/TerrariumPI #281: UI options

duenni, 2019-04-21:
Hi,

when closing the "Environment" tab in the dashboard, it will reappear on reloading the dashboard. Also the graphs don't fill the space and you just get some blank space when closing it. It would be nice if the graphs would expand/maximize when the Environment tab is closed.
Also: the menu on the left can be collapsed when clicking the "burger menu" button on top. This option is also resetted when reloading and the menu reappears.
Is it possible to make both things persistent (close the menu and will stay closed even on reload plus the same for the Dashboard graphs)?
Thanks in advance!

theyosh, 2019-04-22:
Hi,

I know that the dashboard is not that perfect, but I am not a UI developer. So I picked a template system from https://colorlib.com/polygon/gentelella/index.html and there it is doing the same thing.

Making those settings persistence is not handy, because how to reset those? Then I need to hack into the template system.

What I can do is add an extra setting on the system page where you can specify not to show the environment summary on the dashboard. That way the graphs can go to 100% width. The burger menu stays as is, that is 100% hack in the UI which I have no knowledge of..

duenni, 2019-04-22:
> Hi,
>
> I know that the dashboard is not that perfect, but I am not a UI developer. So I picked a template system from https://colorlib.com/polygon/gentelella/index.html and there it is doing the same [...]
 Hmm, I don't know if a reset is neccessary. It could just stay collapsed until you re-toggle it to be expanded. So the state the menu is in currently (collapsed or expanded) just is persistent until the next toggle. I will look into gentelella, maybe this is possible or I can open an issue there. 😉 

> What I can do is add an extra setting on the system page where you can specify not to show the environment summary on the dashboard. That way the graphs can go to 100% width.

Would be really cool. 👍 
Thanks!

theyosh, 2019-04-22:
do a `git pull` and restart. You should now have an option to change the dashboard a bit.

duenni, 2019-04-22:
Works like a charm.
Thanks! Really appreciate it. Great piece of software. 👍

theyosh, 2019-04-22:
Hmm, just add a cookie for this. Should remember the menu state when you change it.

duenni, 2019-04-22:
So for the collapsed menu:
Found [this issue](https://github.com/ColorlibHQ/gentelella/issues/112#issuecomment-422180149). I don't know how you would integrate it properly, so I just did a quick and dirty test and copied this to `/views/inc/footer.tpl` which now looks like this:
````
  <!-- Load the terrariumPI JS script-->
  <script type="text/javascript" src="/static/js/terrariumpi.js"></script>
  <script type="text/javascript">
    globals.current_version = '{{version}}';
    globals.temperature_indicator = '{{temperature_indicator}}';
    globals.distance_indicator = '{{distance_indicator}}';
    globals.volume_indicator = '{{volume_indicator}}';
    globals.language = '{{lang}}';
    globals.horizontal_legend = {{horizontal_graph_legend}};
    function menuToggle(){
        if ($('body').hasClass('nav-md')) {
          $("#sidebar-menu").find('li.active ul').hide();
          $("#sidebar-menu").find('li.active').addClass('active-sm').removeClass('active');
        } else {
          $("#sidebar-menu").find('li.active-sm ul').show();
          $("#sidebar-menu").find('li.active-sm').addClass('active').removeClass('active-sm');
        }
        $('body').toggleClass('nav-md nav-sm');
        setHeight();
    }

    function setHeight(){
    $('.right_col').css('min-height', $(window).height());
    var bodyHeight = $('body').outerHeight();
    var footerHeight = $('body').hasClass('footer_fixed') ? -10 : $('footer').height();
    var leftColHeight = $('.left_col').eq(1).height() + $('.sidebar-footer').height();
    var contentHeight = bodyHeight < leftColHeight ? leftColHeight : bodyHeight;
    contentHeight -= $('.nav_menu').height() + footerHeight;
    $('.right_col').css('min-height', contentHeight);
    }

    $(document).ready(function(){
       menuToggle();
    });
  </script>
</body>
````

This works for now.

duenni, 2019-04-22:
Your answer just overlapped mine. Man, you are too fast. 😄

theyosh, 2019-04-22:
And it is almost the same... I hope it will work. It is a bit jumpy, because it will still load with full menu, and then will just click the hamburger menu.

I do have some free time to fix small issues...

duenni, 2019-04-22:
Reverted my changes and tested yours. Looks good to me. 👍

theyosh, 2019-05-20:
Nice, then I close it now. But I almost forgotten... your new feature will costs some photos :) #210 Could you have a look at that 'issue'?

closed 2019-05-20 by theyosh

duenni, 2019-05-23:
> Nice, then I close it now. But I almost forgotten... your new feature will costs some photos :) #210 Could you have a look at that 'issue'?

I'd love to but I'm afraid right now I have nothing to show. I'm researching for components and so on and I started with the technical stuff. Had a Raspi laying around, everything I have right now is a Raspi with some sensors and a Miflora. 😄 Sorry.