theyosh/TerrariumPI #282: Siri Shortcuts

djarrard, 2019-04-22:
This isn’t an issue, but I thought I’d post it for anyone else in the community who might find it useful. 

I’ve been experimenting with controlling Terrarium Pi devices using Siri Shortcuts. The Terrarium Pi API makes it very easy to automate a variety of procedures at the tap of a button or just by telling Siri. It’s a little more tedious than just using python, but the simplicity of tapping a button on your Apple device is just awesome. 

In my system, I have the lighting environment setup but sometimes I need to override those settings. My terrarium is in a guest room, and guests don’t appreciate the light coming on at sunrise. Or maybe I’m up extra late and want to turn on the night light. I can use the interface but its multiple steps. I have to open the site, turn manual mode on, and then adjust my light. I then have to remember to go back and turn manual mode back off to resume the normal schedule. I built a shortcut to help automate this process.  

When tapping the button (or launching via Siri), it sets the switch to manual mode and toggles the switch. This means it works to turn a light on or off. Since I have multiple lights, I created a shortcut for each light. Of course this leaves them in manual mode, so I created a Resume Schedule shortcut that turns off manual mode on all listed switches. One tap, and everything goes back on schedule. 

I’m using a rigged up humidifier controlled by a relay switch. I’m testing its effectiveness in increasing relative humidity so I can program the timers appropriately in Terrarium Pi. I created a shortcut that allows me to run that humidifier for a specified period of time. When running the shortcut, it runs through a series of logical branches. 

If the humidifier is not already running, the shortcut asks for the amount of time to cycle the humidifier, sets the humidifier switch to manual mode and then turns it on. After the cycle, it turns off the humidifier and optionally sets manual mode back off to resume any schedules it’s associated with in Terrarium Pi. 

If the humidifier is already running, it provides an option to force it off, setting manual mode to off or on in the process. 

The overall result is that I have a single button that I can use to toggle the humidifier on or off without having to fuss around with the manual mode options. 

I’ve pasted the links to the shortcuts below. Each shortcut contains configuration steps to wire it against your system. All you have to supply to it are your system’s Terrarium Pi API URL, credentials, and the switch ID(s) you want to manipulate. 

Toggle Light
https://www.icloud.com/shortcuts/1f2e52cf642f493f886ca2b11ea1f290

Cycle Humidifier
https://www.icloud.com/shortcuts/a0d1787bd8fd4ea0b639ba596726fda5

Resume Schedule
https://www.icloud.com/shortcuts/0435c0c99c8b49928a5434b329e37094

theyosh, 2019-04-22:
Euh, so you are saying my software is Siri compatible? :P That is very awesome!

Just a technical question: Do you have to open my software to the internet in your router/firewall? Or is the Siri device doing it all local?

djarrard, 2019-04-22:
It’s doing it all local. The shortcuts are essential scripts executed from your iPhone.  As long as your iPhone is on the same WiFi network as the RPi, it will work. Running it outside of your local network would only work if you made your RPi public facing via some sort of forwarding, and then I would strongly suggest using SSL with a proper cert.

AndrejWeber, 2019-04-22:
@djarrard
Great stuff!
Do you know, if there is something eqal for Android?

@theyosh
This way we could also integrate our own Hardware-Buttons. There is only needed to link a gpio to start an own side loader python-code which will access your api.

djarrard, 2019-04-22:
@AndrejWeber
Unfortunately I don’t think there is an exact equivalent. There’s probably some things you could do with IFTTT to string together with other apps, but my guess is those things all run in the cloud and wouldn’t work for local addresses. Siri shortcuts is integrated directly into the OS so it does everything locally. 

And I was thinking the exact same thing with hardware buttons. That’s actually why I started playing with “software buttons” in Shortcuts. I wanted to see what was possible with the API. I had to study it in Fiddler for a little while but it looks like you can automate anything you do from the interface using http requests from a python script.

djarrard, 2019-04-22:
I tested the hardware button method. I replicated the shortcut in a python script and hooked up one of those tiny buttons that come with a raspberry pi kit. Works like a charm to toggle the lights on and off. The hard part will honestly be mounting the buttons on the enclosure. Not super useful on the breadboard :)

AndrejWeber, 2019-04-22:
I have some old computer tower-Buttons, I will mount them at the front of my terrarium.
I will 3D Print a cover for display and buttons together. 
May be you can share your python code to us?
I am not able to write at my own, but I can edit it for my hardware.

djarrard, 2019-04-23:
@AndrejWeber Yeah I can share the script samples. You might have to give me a day or two to clean them up properly and document deployment steps.

@theyosh Would you prefer that I create a separate repo for the hardware button scripts or do you want me to share them within this repo? Note that the scripts manipulate attached switches via the API as opposed to direct GPIO manipulation. This keeps everything nice and federated with your software.

djarrard, 2019-04-23:
Here's what I've got for a simple switch that enables Manual Mode and toggles a switch on and off.  The first thing you'll need to do is update all of the parameters that are in the Parameters section and make sure they match your system and configuration. Keep in mind that this script will leave the switch in Manual Mode. (To get around that, I ran a second button that sets all of my switches back to manual mode off.) You can test the script by running it as the pi user, however, the script won't persist after you close your terminal session by default.

Once you've tested and made sure everything works, you'll want to have the script run at startup under a headless user. One way to do that is to modify the rc.local file. To do this go through the following steps.

1. Logged in as pi, execute `sudo nano /etc/rc.local`
2. Toward the bottom,  you'll see an `exit 0`. Just above that line, add a command to execute your script. Be sure to leave an & at the end of the line to force it to fork the process.

example: `sudo python /home/pi/scripts/lightButton.py &`

3. Save rc.local and reboot. The script should automatically run in an endless loop on startup.

Here's the basic script:

```import requests
from requests.auth import HTTPBasicAuth
from gpiozero import Button
import time

#Parameters
button = Button(26) #<<<< Use GPIO Number, not physical number
apiURL = "http://localhost:8090/api"
username = "admin"
password = "password"
switchID = "1555179447" #<<<< Go to /api/switches to find Switch ID

#Endless loop
while True:
    try:
        button.wait_for_press()
    
        #URL Construction
        switchURL = "{0}/switches/{1}".format(apiURL,switchID)
        toggleURL = "{0}/switch/toggle/{1}".format(apiURL,switchID)
        manualURL = "{0}/switch/manual_mode/{1}".format(apiURL,switchID)

        #Get Switch Status
        sR = requests.get(switchURL)
        name = sR.json()['switches'][0]['name']
        power = sR.json()['switches'][0]['state']
        manualMode = sR.json()['switches'][0]['manual_mode']

        #Turn On Manual Mode
        if manualMode == False:
           requests.post(manualURL,auth=(username,password))
           print("Manual Mode enabled...")
        else:
           print("Manual Mode is already enabled...")

        #Toggle Switch
        requests.post(toggleURL,auth=(username,password))
        if power == True:
           print("{0} has been turned off.".format(name))
        else:
            print("{0} has been turned on.".format(name))

        time.sleep(1)
        
    except KeyboardInterrupt:
        print("Interrupt received, stopping...")

```

jornobe, 2019-04-24:
Thanks for sharing! Will be using your examples to create my own tasks in Shortcuts. 

This is truly awesome!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

djarrard, 2019-05-18:
For those of you who are using shortcuts or just want to access your setup from outside your network, I found that everything also works over a VPN connection. If you have a newer, higher end router, there's usually an option on your router to setup a VPN server. So when I'm travelling, I just turn on the VPN connection from my iphone and i can pull up the web page or run my shortcuts. It's a nice compromise for making your setup available from outside your network without having to worry about a lot of the security aspects and technicalities of port forwarding.

theyosh, 2019-05-20:
@djarrard Cool!. As an alternative, I have found: https://www.zerotier.com/ That should also work.

For now, I leave it open for discussion for this nice trick Siri trick. And will add it later to the FAQ.

djarrard, 2019-05-21:
Thanks @theyosh!

One other thing I just discovered. On iOS there’s an app called Pythonista that lets you run python scripts from your iphone. Seems to work really well. Anyway, there’s a Siri shortcuts integration that will let you launch a python script in pythonista using a Siri shortcut. So if you prefer to script out the procedure in python instead of using the shortcuts builder, that’s an option too. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

theyosh, 2020-05-07:
@djarrard He, I came across this: https://github.com/homebridge/homebridge

But it is not really clear what it can do and how technical you need to be. But I get the idea that installing that on the Pi, and write a custom plugin, you can talk to my software :P

I do not own Apple hardware, so I cannot test. Anybody here interested to see if it is something usefull