theyosh/TerrariumPI #283: Logs and Warnings don't reflect Fahrenheit settings

djarrard, 2019-04-24:
It looks like this issue was reported and resolved in #252, but I'm seeing it happen on my system. I did an update today just to make sure, but it still happens.

Basically, being an American, and since we like to do things that don't make a lot of sense, I have my temperature settings set to Fahrenheit. In the logs, however, we get readings like this:

'Measured value 23.5F from w1 sensor 'Enc. Mid Temp' is outside valid range 50.00F - 100.00F in 0.90387 seconds.'

The Fahrenheit equivalent is 74.3, which is in the valid range, but it gets logged as a warning so that little orange indicator keeps getting larger. It doesn't break anything, but that indicator starts stressing me out :)

djarrard, 2019-04-24:
I also found that this happens under System > Status > CPU Temp. My settings are in F, and the needle on my gauge is always slammed to the right because it's plotting those temps on a chart made for C. The indicated number has been converted to Fahrenheit correctly, but the chart's range seems to be expecting a C range. 120 would be way too high in Celsius, but it's just fine in Fahrenheit.

Bonus: For my terrarium temps, I prefer Fahrenheit, but I prefer Celsius for CPU temps. It would be nice to be able to use both instead of picking a global temperature format.

theyosh, 2019-04-25:
Hmm, those Fahrenheit values where all wrong.. so was the inch and gall calculations. Sounds like most people are European and do not had to convert values. Now it should work better.

Also the CPU temperature looks a bit better now. But having two temperature units is not possible. That will add to much code for just 1 temperature value. So it is all or nothing for now.....

But thanks for finding this. And may I ask you to look at #210 ?

djarrard, 2019-04-26:
Thanks! I can confirm that it is fixed after the update. Thanks for doing that.

I looked at #210 for you

closed 2019-05-04 by theyosh