theyosh/TerrariumPI #284: Environment not switching lamp

danijoo, 2019-05-02:
Hello,

This might be more of a layer 8 problem but I have problems with configuring the environment to correctly switch my lamp. In the attached Screenshots you can see that I configured the environment to switch the lamp on during day (by weather). However even though the current time is inside the day period in the overview, the lamp is shown off. What am I doing wrong?

Best Regards 
Daniel
[image]
[image]
[image]

djarrard, 2019-05-03:
Might be a silly question on my part, but have you made sure the switch is not in manual mode? If it is, the environment won't do anything to its state while that's turned on. You can check that by going to Switches > Status and checking to make sure there's no red Manual Mode indicator on that switch.

This environment part is working for me, but my settings look a bit different. To test if it's a bug in the parameters, try setting maximum hours to 24, minimum hours to 0, and the hours shift to 0. If it starts working then, that tells us there might be a bug or issue with how the parameters are being interpreted.

Lastly, are you seeing anything in System > Logs related to that switch?

djarrard, 2019-05-03:
One more thing, is your system time accurate to your location? You can check by opening a terminal session and typing `date`. Make sure that lines up to your local time or your schedules are going to be off.

danijoo, 2019-05-03:
Timestamp was fine. Im not sure what the actual issue was but I was able to fix is after all by removing all switches from the interface and rebooting. :)

closed 2019-05-03 by danijoo

theyosh, 2019-05-04:
@danijoo Thanks for helping out.
@danijoo I have heard form more people that after initial installation, not everything works from the start. And a restart will fix that. Then it never happens again. So I did not fix nor investigated this.

And your light settings are correct. Have fun with the software! And may I ask you to look at #210 ?

danijoo, 2019-05-04:
Hey, small update: I think the problem was part of meross_iot. After checking the logs in this i realized they are full of reoccuring ```"The client is connecting already, let's wait a bit..."``` messages sent from meross_iot module (Im using the MSS425E). So maybe it wasnt able to connect to the socket to switch lights and somehow got into a deadlock while trying to reconnect. I will keep an eye on the issue and report back if it happens again.

@theyosh pictures will come as soon as the setup is finished 👍