theyosh/TerrariumPI #285: Inconsistent naming for timer on/off

danijoo, 2019-05-04:
Hi,

I noticed that the tooltip for the timer in the environment tab does not match the gray hint message. The gray hint asks for a duration *in seconds* while the tooltip in english asks for a time in minute. From a quick glance at the source it appears that values should indeed be in seconds, not minutes. Is this correct?

Best wishes.
Daniel

theyosh, 2019-05-04:
HI,

hmm, not sure if I understand you correctly but the the duration value should be minutes:
https://github.com/theyosh/TerrariumPI/blob/master/terrariumUtils.py#L383

[image]
So I changed the text to reflect minutes. Making time schedules of seconds is not handy, because the software works with 15 seconds interval and the sensors and switches at 30 seconds interval.

So when I need to check every second for actions, or I have to make some fancy system that queues all the actions etc... So, for now, the smallest timer interval is 1 minute. Actually, it can work with 0.5 minutes ;)

danijoo, 2019-05-04:
The tooltip (right) and hint text (left) say different things which is a bit confusing. I agree that minutes is fine. Always calculating seconds is a bit cumberstone.

[image]

theyosh, 2019-05-04:
Euh, I just updated the code. So did you do a `git pull`? Then you have also to clear your browser cache. Then you will see the update. The templates are heavily cached to reduce the SD card wear.

danijoo, 2019-05-04:
perfect thanks! I would have done a PR but wasnt exactly sure if the true value is minutes or seconds.

closed 2019-05-04 by danijoo