theyosh/TerrariumPI #286: MSS425E stops working after a few hours

danijoo, 2019-05-04:
When I use a Meros MSS425E to control switches, It works for the first few hours but then suddenly stops working. In the system log below I can see a lot of messages from the ```meros_iot``` module that make me think its some kind of connectivity issue or possibly a deadlock where TerrariumPi tries to connect too often? The Rasperry is definitly connected to the Internet and the MSS425E is connected, too. I can still switch it on/off with my smartphone from the Meross App but the a manual switch from Terrarium Pi is also not possible. 
A reboot solves the issue, but only for a few hours.

Please let me know if I can do anything to help get this fixed.

Excerpt from the system log:
```
2019-05-04 20:19:30,739 - ERROR   - terrariumSwitch      - Failed changing power switch 'heat lamp' of type 'mss425e' at address '1' from state 'True' to state 'False' (Forced:False)
2019-05-04 20:19:28,215 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00004 seconds. Waiting for 29.99996 seconds for next update
2019-05-04 20:19:22,484 - INFO    - terrariumAudio       - Update done in 0.00006 seconds. Waiting for 29.99994 seconds for next update
2019-05-04 20:19:22,463 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:19:22,456 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:19:20,728 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:19:20,722 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:19:10,445 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:19:10,437 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:19:10,429 - INFO    - terrariumEnvironment - Environment light is turning off the alarm min powerswitches based on weather
2019-05-04 20:19:10,402 - WARNING - terrariumEnvironment - Update took to much time. Needed 228.47848 seconds which is 213.47848 more then the limit 15
2019-05-04 20:19:10,375 - ERROR   - terrariumSwitch      - Failed changing power switch 'heat lamp' of type 'mss425e' at address '1' from state 'True' to state 'False' (Forced:False)
2019-05-04 20:19:00,707 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:19:00,699 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:19:00,362 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:19:00,356 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:18:58,209 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00005 seconds. Waiting for 29.99995 seconds for next update
2019-05-04 20:18:52,479 - INFO    - terrariumAudio       - Update done in 0.00009 seconds. Waiting for 29.99991 seconds for next update
2019-05-04 20:18:42,686 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:18:42,680 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:18:40,344 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:18:40,336 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:18:28,668 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:18:28,661 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:18:28,205 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00005 seconds. Waiting for 29.99995 seconds for next update
2019-05-04 20:18:22,475 - INFO    - terrariumAudio       - Update done in 0.00005 seconds. Waiting for 29.99995 seconds for next update
2019-05-04 20:18:22,324 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:18:22,318 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:18:16,649 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:18:16,643 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:18:08,306 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:18:08,299 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:18:02,629 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:18:02,623 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:17:58,199 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00005 seconds. Waiting for 29.99995 seconds for next update
2019-05-04 20:17:56,287 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:17:56,281 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:17:52,470 - INFO    - terrariumAudio       - Update done in 0.00005 seconds. Waiting for 29.99995 seconds for next update
2019-05-04 20:17:42,610 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:17:42,604 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:17:42,267 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:17:42,261 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:17:28,194 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00005 seconds. Waiting for 29.99995 seconds for next update
2019-05-04 20:17:24,592 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:17:24,586 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:17:22,465 - INFO    - terrariumAudio       - Update done in 0.00005 seconds. Waiting for 29.99995 seconds for next update
2019-05-04 20:17:22,249 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:17:22,243 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:17:10,576 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:17:10,571 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:17:04,231 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:17:04,225 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:16:58,560 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:16:58,555 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:16:58,189 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00004 seconds. Waiting for 29.99996 seconds for next update
2019-05-04 20:16:52,460 - INFO    - terrariumAudio       - Update done in 0.00005 seconds. Waiting for 29.99995 seconds for next update
2019-05-04 20:16:50,214 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:16:50,208 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:16:46,543 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:16:46,535 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:16:38,198 - INFO    - protocol             - The client is connecting already, let's wait a bit...
2019-05-04 20:16:38,192 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
[... cut ...]
2019-05-04 20:15:21,950 - INFO    - protocol             - Current client connection status is: ClientStatus.CONNECTED
2019-05-04 20:15:21,941 - INFO    - terrariumEnvironment - Environment light is turning off the alarm min powerswitches based on weather
2019-05-04 20:15:21,915 - WARNING - terrariumEnvironment - Update took to much time. Needed 228.32620 seconds which is 213.32620 more then the limit 15
2019-05-04 20:15:21,887 - ERROR   - terrariumSwitch      - Failed changing power switch 'heat lamp' of type 'mss425e' at address '1' from state 'True' to state 'False' (Forced:False)
```

theyosh, 2019-05-08:
Hmm, those messages are from the meross_iot library. Not sure why, but I saw he has made a lot of updates. So for you, you could try to rerun the installer, which will update all python modules. Maybe a new version is available that fixes this...

I will also look in my code, but I cannot test it, so it will be hard for me to debug this.

danijoo, 2019-05-09:
I am already on the newest version of it. Maybe there were some breaking changes due to these updates. I will try to look into it and see if I can find a way to fix it on the weekend.

Ultameciia, 2019-05-10:
@danijoo I'd make an Issue on the meross_iot github. He sometimes takes a bit to reply, but he's super nice and has a MSS425E that can be tested with. Any changes he makes to Meross_iot would be reflected here.

danijoo, 2019-05-12:
@Ultameciia I doubt the issue is with meross_iot. I think its an implementation issue in TerrariumPi.

I now downgraded meross_iot from ```HEAD``` to version ```0.2.1.0``` and the issue appears to be fixed.

theyosh, 2019-05-12:
Hmm, not sure, but I am just installing according to the manual: https://github.com/albertogeniola/MerossIot#installation

So, if you have to lower the version, that means there is a bug in the newer version. And I am not able to test. The example code that is provide by the library, does not handle repeating getting data. It only runs once, and than all the connections are closed. But I run in a deamon mode and will keep the connections open. Maybe that is causing an issue, but I am not able to test or debug this... :(

theyosh, 2019-05-20:
Ok, I now see that the code has been changed the way it works. And I am pretty sure that I am not compatible with that change. So, for now, you should pin it to the latest 2.X version.

so, running manual `sudo pip3 install meross-iot==0.2.2.3` should do the trick. I will pin this version also to the installer.

theyosh, 2019-07-05:
For now, I leave it this way, as I do not have the hardware to test the new library. Should be done by someone that has the hardware at home and programming skills

closed 2019-07-05 by theyosh