theyosh/TerrariumPI #287: Request; Multiple Light settings

Natrixz, 2019-05-06:
Hi Yosh,

Is it perhaps possible to add more Light options?
For example;
I use 2(or 3O non-dimable lights in most of my setups, it would be nice if it's possible to switch on 1-2-3 of the lights with an interval.
Say;
Using weather/sunrise as a base for switching it, then switch light 1, 30/60 minutes later switch 2, 30/60 minutes later switch on 3.
Same for the off switches =)

I would understand if that's not possible right now, but if it can be; that would be awesome ;)

theyosh, 2019-05-08:
Hi,
if you depend on the sunrise/sunset values, it is not possible at the moment. If you use timers it is possible.

In the environment you set then 1 switch with timer options. Then the other switch will need there own timer settings, and add there some more on or off time based on the environment values.

Then you can make the schedule you want. But it will not shift when the sunrise will shift. Nor the individual timers will shift if you shift the environment timer. It is not a 'nice' solution...

Natrixz, 2019-05-09:
Yeh indeed, it's not really a nice solution haha
Right now im using my home build time delay for whenever the light switches on.
Not very configurable but atleast it delays the amount I want it too and move along with the sunrise :P

theyosh, 2019-05-20:
I'll put it in the wish list. I do like the idea..

The-Madhouse, 2020-10-07:
yes, this is something that would be cool if it could be added in the future. i'm trying to make a realistic sunrise and sunset with 3 lights, so putting light1( a normal dimmable LED e27 bulb) on first, after X time (slowly ) adding a basking light, and at the end adding the UVB bulb. and when the sun goes down first let the UVB bulb go out, then slowly let the basking light go out, and finally slowly put the LED light go out, maybe even slower because i'm going to keep leopard gecko's and what i've read they like to be active when it's getting darker, so i can extend the active time of the animals :p 

but i believe it's going to take a long time before this (if even ever?) is implimented? :p 

@theyosh is there any way i could buy you a virtual coffee or something for the awsome work you allready put in this? i realy like the way my ( still empty :p ) terrarium works for now :p

theyosh, 2020-10-14:
Yeah more options ;) But this does give me an idea. I am planning a new version, which is being programmed at right now, but that is more then just a small refactoring step. I write is again from scratch, as I am changing to webbased frameworks.

But the idea I just got is that you should be able to set a start time (on) where you could say the lights should be on (all) or start going on. Then you get the amount of slider that you have selected switches, and change the slider in time so that you can specify the time between the different switches and how they turn on (dimmer or direct).

My main problem is that I am a backend developer, and sometimes it is a bit hard to find the right user interface for settings and make clear what they do... But we will see, and this is something I will look into it.

Donations always welcome :) https://github.com/theyosh/TerrariumPI/wiki/Donations

Toekie, 2021-01-29:
Hello, since this is already about light. I'm a noob with python and cant properly read it but does your system has the sunrise/set function like the [Dawn2Dusk](https://github.com/dmkishi/Dusk2Dawn) lib? I've been working on my own system based of that lib but i also gets random hangs so i figured why not try this one since its way better, very impressive! But i need a propper dimming function.

Sincere,
T.

theyosh, 2021-02-01:
@Toekie you are refering to an Adruino project. So I cannot compare.

But when you enter the correct weather information, you will get the option to let the lights turn on and off based on sunrise and sunset. This can then be normal switches or dimmers. And you can even make the lights go on in a interval of 30 minutes if you play with the correct dimmer values.

theyosh, 2021-04-10:
Yesh!!! Party time... Finally after a few years.... it is here!! If you have time, could you have time to test this: #539