theyosh/TerrariumPI #288: AMG8833 Sensor?

res2cpu, 2019-05-15:
Good Morning

I Notice that you have added a non-contact sensor to your list, I in fact ordered one of these and was going to use an ESP32 to add it as a remote sensor (My Uromastyx has a habit of basking on the sensor).
I was wondering if it would be possible to add [AMG8833](https://learn.adafruit.com/adafruit-amg8833-8x8-thermal-camera-sensor) as well?  It's an 8x8 thermal camera, my idea was to get the highest value and report it (maybe on the outside use it as a thermal motion camera to know when he's on the backing spot).  It uses I2C. Sadly not a cheap sensor (£30 on ebay) but i do have one and could help out with testing ect.

theyosh, 2019-05-20:
Hmm, have to look into this. I have seen this: https://github.com/melopero/Melopero_AMG8833/blob/master/module/melopero_amg8833/AMG8833.py

Which does give a temperature back, but not sure if it is an average or how it is actually measured.

res2cpu, 2019-05-24:
I'll connect it up and have a play, any luck i should have some time this weekend.

theyosh, 2019-07-05:
Any updates on this?

res2cpu, 2019-07-11:
Sorry, life got in the way.  I have time this weekend to throw this on the bench.

theyosh, 2019-11-10:
Any updates on this? (second time....)

res2cpu, 2019-11-10:
Good Evening!
At last i have some answers.

I ran the [example code ](https://github.com/melopero/Melopero_AMG8833/blob/master/examples/temperature_matrix.py) which gave out a text output of the below:
`pi@TestPi:~ $ python temperature_matrix.py.1
21
[[18.0, 18.0, 18.0, 19.0, 19.0, 19.0, 19.0, 19.0], [18.0, 18.0, 19.0, 19.0, 19.0, 18.0, 19.0, 19.0], [18.0, 19.0, 21.0, 22.0, 19.0, 19.0, 18.0, 19.0], [19.0, 19.0, 25.0, 27.0, 20.0, 19.0, 19.0, 18.0], [18.0, 20.0, 24.0, 27.0, 23.0, 20.0, 19.0, 18.0], [19.0, 20.0, 25.0, 25.0, 25.0, 24.0, 19.0, 19.0], [19.0, 20.0, 23.0, 23.0, 25.0, 24.0, 21.0, 19.0], [20.0, 23.0, 23.0, 24.0, 25.0, 25.0, 21.0, 18.0]]
`
The first number (21) seems to be the average temp in 'C

It keeps looping every second giving updated values.  Hope this helps.

theyosh, 2019-11-11:
First attempt to support the sensors is done. Do a `git pull` and rerun the installer. Then restart and clear browser cache. You should be able to add a new AMG8833 sensor, which will produce a single temperature value

res2cpu, 2019-11-13:
Thank you very much.
I have done the git pull and run the installer again. The sensor is seen and I can get the first temperature from it but then it seems unhappy. I believe that my test setup has issues from other changes I have made unrelated to the software. 
I will reload it and give it another try, but it's looking good so far. I will report back as soon as I have retested.

res2cpu, 2020-01-12:
Good Afternoon

I can finally confirm that this this seem to be working correctly but only after i run the command `sudo pip install melopero-amg8833`  before running this the sensor returns noneC looking at the logs.

theyosh, 2020-01-18:
Hmm, that module is in the installer. Are you running it on puthon3? As this sensor is only supported on Python3...

theyosh, 2020-01-28:
Ok, closing now, as the module 'melopero-amg8833' is installed

closed 2020-01-28 by theyosh