theyosh/TerrariumPI #289: HELP: OWS DS18b20 sensors stop working after 12-24 hours

djarrard, 2019-05-16:
I'm using 4 2M long DS18b20 sensors (the waterproof variety) in my setup. What I find is that they report temperature perfectly for about 12-24 hours, and then completely stop working (as reflected in the logs in Terrarium PI). When this happens, if I check the /sys/bus/w1/devices directory, all of the unique directories for each sensor are completely gone. If I restart the pi, the sensors come back and everything is good again for a little while.

I've been racking my brain on things to try to fix the issue. My theory is that it's something hardware/wiring related. It's certainly not the software. So far I have tried switching from 3v3 to 5v, and even reduced the resistor size from 4k7 to 2k. These changes don't seem to make any sort of difference as the problem continues to persist. I'm NOT using parasitic mode.

The wiring is simple. I'm using a breadboard to consolidate all of the lines into three wires that go directly to the pi. I'm using GPIO 4 for the data line.

[image]

This seems a bit similar to to #212. When I run a i2c detect -y 1 it produces an empty grid. I'm not sure if that could be related. I'm a bit new to this.

Has anyone had similar issues, or have any suggestions as to how to troubleshoot? At this point, I'm at a bit of a loss.

res2cpu, 2019-05-18:
I had this exact problem on my setup, it might not be the same issue but you never know.

After x number of hours i would have all 1-wire sensors drop, i would sometimes get very odd temperature readings in the logs first followed by all 1-wire sensors dropping offline until i rebooted.

I found it to be a faulty sensor ([from Amazon](https://www.amazon.co.uk/gp/product/B07CMZQ4YG/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)), i ended up only connecting 1 sensor and running it for 12 hours, then add another, rinse and repeat until the issue started occurring again.  Replaced the last one to be added and not had an issue!

As i side note i did find that sometimes when it had happened it would cause an issue with TerrariumPI starting even after the sensor was removed (corrupt database?).

Something that might help you with your wiring is to put 3.5mm stereo headphone jacks on your sensors and use a multi-way headphone splitter,  It ensures a strong and good contact while making it easy to make changes. [Here's](https://www.amazon.co.uk/gp/product/B072361BGF/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) the one i used.

Hope this helps

vacuumboots, 2019-05-18:
I can't contribute much, but just want to say that converting 1-wire to 3.5mm headphone jacks is a brilliant hack I would never have come up with. Thanks for input res2cpu

djarrard, 2019-05-18:
Thanks, @res2cpu! I'm going to try a few things based on your suggestions. I tried removing one and Terrarium PI wouldn't start. did you have any tricks to get it to come back up after removing the sensor? Deleting/renaming the database file?

res2cpu, 2019-05-18:
Sadly i was very short on time so went for the quick fix and started fresh as i was also unsure at the time if the sd card was causing the issue (same issue on another sd card).  I'm only guessing about a database issue being the cause. I did try a reinstall of TerrariumPI over the top but it failed.

I ended up loading a fresh copy (OS and TerrariumPI), after safely shutting it down removed the SD card and made a cloned image of it using [Win32 Disk Imager](https://sourceforge.net/projects/win32diskimager/).  That way in the event of corruption i could restore it back up and be running in less then 15 min, an even faster way is to use a spare SD card and pre-image it with the backup expecting a problem and you back up and running within 2 min.

This is also the same system i use now.  Lets just say my better half is not the brightest bulb when it comes to this level of technology, but she can replace a SD card.  So whenever i make a change i take a backup of the card, clone it to a 2nd card which is in a safe place with the vivarium.  Just in case im not about she can pull the plug, change the card.  Always plan for it to fail, after all for most of us these pets are family in a way. 

I had an issue with my WiFi 2-3 weeks ago, ended up having to change the WiFi channel just before going to work.  My vivarium failed to rejoin the network or crashed ect, so i was at work not knowing if my Uromastyx was too cold/hot or if the lighting had kicked in.  Now i have the parts to fit this weekend to get both the vivariums onto wired networking, [smart plugs ](https://www.amazon.co.uk/gp/product/B07CVJYV3G/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)so i can remotely turn them both off/on independently with external monitoring of temperatures to my home automation with a watchdog running so that if for some strange reason the vivarium stops responding i get pinged.  I'm even pondering a way of automatically switching to a [2nd sd card](https://www.ebay.co.uk/itm/202312992403?ul_noapp=true).  Is this all over the top? Even i think so, but i don't think most of us would be here using computers to control our pets homes if we didn't enjoy going over the top.

Tho i would like to say that i don't in anyway believe that this software will let me down (i have had up-times of 45 days before i made changes and rebooted), i just like to make sure i try and cover every base just in case!

jornobe, 2019-05-19:
Had the same problem with 1-wire sensors attached directly to GPIO pins on the Pi. Had a scripted reboot of the Pi running every 3rd day...

Have moved completely to using the Sheepwalk 1-wire host adapter to interface with the sensors.

Since I started using this new setup I had my Pi and sensors online and working fine since January (no reboots needed).

http://www.sheepwalkelectronics.co.uk/product_info.php?products_id=67

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

theyosh, 2019-05-20:
@djarrard Hmm, those problems are nasty and hard to really find the issue. In the past I had also hardware issues, but mostly, be reconnecting/rewiring them, it suddenly works. And I am not a hardware expert.

For your 1Wire problem, I have to rely on the OS for the information. So when the are gone, my software can find it. And I have no clue why, other then either:
- Power issue (any low voltage messages when you run `dmesg` ?)
- Some wiring went bad. I have had some bad wiring which corroded due to high humidity and broke)
And then the option of wiring one at a time and wait, is the only cheap solution for now and hoping that it is indeed 1 sensor that is having issues.

Also, I had already 1-wire adapter of Sheepwalk like @jornobe . So I used that all the time. The nice thing is, that the same temperature sensors can be used. But the adapter is more stable, and looks like does handle the power usage better. I have it running now for at least 2 years 24/7. No issues at all. So for stable 1-Wire usage, I do suggest to look for a 1-wire adapter. The only thing is that I only have used 1 type of hardware. So I cannot tell if the wiring for all 1-wire adapters are the same: https://pinout.xyz/pinout/1_wire_pi_zero is also an option. And I used this for connecting all my temperature sensors: http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=23&products_id=65

@res2cpu Hmm, the software only needs Wifi for getting weather information for the timers. When that does not work, I would expect a timeout and then just continue. But then when you need the sun rise and set times, it will hang. And I have to admit, I did not test that. It is a bit hard testing without network. So, for now, I call it a 'known issue'.
For restoring, you should only have to store the following files: https://github.com/theyosh/TerrariumPI/wiki/FAQ#which-files-to-backup Not sure if you can use webdav, but then you could mount a cloud system and store there the backups..

And in the past I had database corruptions, but that should be fixed by my software. So I think you have a different corruption.

Your 'concerns' about redundancy I do understand them. But, it will get a lot more complex if I had to make such in my software. Also, my experience is that some only devices can become offline and that is hard to handle. It can be a timeout, or an error message.
But I have my terrarium (70% humidity) now running for 3 years, and only some humidity sensors died... (How ironic) but my SD card will run for at least 6 months, and I have no issues at all. But again, I use 1 wire adapter. And do NOT use DHTXX sensors, as they are failing more then working stable.
Again, SD card corruption is hard to predict. If it helps, I have changed the code already so that logging is written to temp memory and onle once a day written to the SD card. Same goes for the webcam. Only saved images are really saved. Other images are all written to the temp memory, so it will not wear on the SD card.

So, I have to advice to start using a 1-wire adapter and connect there you 1 wire sensors to it. Can be also humidity sensors.

djarrard, 2019-05-22:
A quick update here while I’m thinking about it. I’ve taken a couple actions so far. I rewired all of the wires to ensure a good connection, and while I was at it I went ahead and cut any excess lines to reduce any potential of signal degradation. The wires are 2M long but several of them needed much less distance than that to get into the enclosure. The issue popped up again a little while later. 

From what I’ve been reading on various forums, this issue can be caused by the presence of a lot of electrical noise. I don’t seem to remember having this problem until I introduced something new: speakers. I use the audio function in Terrarium Pi. Last night I unplugged the jack from the pi and disabled the audio system for now. As of this morning things are still going strong. I’ll report back in a few days as to whether this is a long term fix. I do want to clarify that if this is the fix, this **is not** an issue with Terrarium Pi at all, rather a signal noise issue. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Ultameciia, 2019-05-22:
Just to chime in, I'm seeing a pretty similar issue with my SHT20. However it causes a wider system impact. every 48 hours on the dot, I lose any readout under environment for my sensors. I can see them if I scroll down to the gauges or to the sensor page. But when this happens my switches stop following the time schedule for lights and my mistking doesn't run based on humidity(although that's most likely due to the environment not registering any sensor). A reboot fixes it, but after 48 hours it happens again.   I just chalked it up to the meross_iot pip being a bit funky with the model I use.

jornobe, 2019-05-23:
@Ultameciia - this is exactly what I had... very annoying when your humidity sensor is “stuck” in a low reading and does ‘t get updated while having the humidity environment kick in over and over and over...

As said above, running for weeks now after switching to:

http://www.sheepwalkelectronics.co.uk/product_info.php?products_id=67 http://www.sheepwalkelectronics.co.uk/product_info.php?products_id=67

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

djarrard, 2019-05-23:
Thanks everyone for all of your thoughts and suggestions. This is a great community. I wanted to update you all that so far, everything has been running for longer than it ever has without running into the problem. The fix, for me, does indeed seem to revolve around removing the speakers from the equation. Having them plugged in seems to introduce something that causes the sensors to stop responding after a while. The only two things I can think of are electronic noise or power disruptions. I have a good quality PSU so I'm leaning toward the former. I really can't explain what specifically is causing the noise. The audio jack itself, or perhaps the close proximity of the audio cable and the sensor cables. At any rate, removing the speakers for me is a small price to pay to have the sensors working reliably. I should add that my speakers do have their own power supply, so it's possible that using speakers without a dedicated power supply could produce less noise.

To makes sure it's speaker-related, I'll report back one more time after a week and close the issue if I don't run into any problems. Hopefully, this is a wrap. Thanks again for chiming in with your experiences. I'm sure this thread will help someone down the line (no pun intended).

res2cpu, 2019-05-24:
@theyosh 
Oh there was no problem with your software, when i got home everything was working fine and no problems with anything other then the WiFi dropping.  It was more a case of my thoughts on trying to cover all bases (most likely my paranoia).

One other problem i had looking back and thinking about it was then i added the mpdmv41 was similar in that all the 3.3V based sensors started to play up as soon as it powered up, i put it down to it creating noise on the 3.3V line.  To fix this i ended up powering it by a [3.3V regulator](https://www.ebay.co.uk/itm/AMS1117-LDO-DC-5V-to-3-3V-Step-Down-Converter-Regulator-800mA-Vertical-Module/142280914612) this fixed the issue and might also be the same issue for @djarrard (more load then the regulators on the raspberry pi can take).  In that case the quality of the power supply wouldn't make a difference.

Using the Sheepwalk 1-wire host adapter could fix this as it might be powered from the 5v.

djarrard, 2019-05-24:
@res2cpu perhaps you’re right. I will say I did try switching the power source to a dedicated 5V pin and saw the same behavior (the sensor is compatible with 3v3 and 5V). Since it started working again, I’ve switched it back to a shared 3v3 line. You may be right about a regulator helping with the noise when the speakers are added. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

res2cpu, 2019-05-24:
How about a USB sound-card?  That should pull power directly through the main power supply, bypassing any regulation.  Did the sound come through ok? Did it have any kind of buzz or hum which was the same at any volume?  If so it could be something called a ground loop which is very common when using switch mode power supplies.

theyosh, 2019-05-25:
I do support USB soundcards... but are the magnets in the speakers not a problem? Magnetic fields?

But I would also suggest to go for usb, as using the internal jack, you cannot use PWM stuff.. Something about shared ports on the device it self. So, go for the most cheapest usb soundcard. I tested it with a 10 dollar version. Works well.

theyosh, 2019-07-05:
Closing due to no updates and not solvable through software.

closed 2019-07-05 by theyosh

axeljerabek, 2023-12-07:
I had the same problem and wrote a watchdog. Since then, things run smooth:

https://www.wetransco.de/ds18b20-to-92-temperature-sensor-failing-write-a-watchdog/

gerben-f1, 2024-07-20:
> I had the same problem and wrote a watchdog. Since then, things run smooth:
> 
> https://www.wetransco.de/ds18b20-to-92-temperature-sensor-failing-write-a-watchdog/

 An u help me? I tryed the watchdog but doesnt work

i have 2 sensors that says 0 degrees

axeljerabek, 2024-07-21:
> > I had the same problem and wrote a watchdog. Since then, things run smooth:
> > https://www.wetransco.de/ds18b20-to-92-temperature-sensor-failing-write-a-watchdog/
> 
> An u help me? I tryed the watchdog but doesnt work
> 
> i have 2 sensors that says 0 degrees

The watchdog is only for sensors that fail after a while. If they are not working at all, the watchdog will not help you much.
Maybe you check the complete setup of the sensors again, the address and the readouts?