theyosh/TerrariumPI #291: Rpi Camera

Rushmonk, 2019-05-19:
Hi, 

my Original Pi Cam was working all ok. I just deleted it to see what settings are possible.
Now i can't re-add the camera. After several reboots and power cycles, the cam still dosen't appear. Further if I click on Save without adding anything, the progress seem to be in a loop. 

But still who can I re-add my camera?

Cheers

vacuumboots, 2019-05-19:
I've been playing around a lot with a Raspicam and two usb cameras witht the terrariumpi project this weekend. I would check "settings.cfg" in the terrariumpi project folder, remove all camera configurations from that file, power cycle the raspi with all cameras removed, then re-attach the Raspicam, turn the raspi on again, re-enable the interface with raspi-config, and re-add through the web interface. Just my two cents!

Rushmonk, 2019-05-19:
Hi, thank you for reply. But unfortunately I had no success. In my settings.cfg was no info about a camera. 
But I followed your procedure, still I've got only the testscreen picture shown, once I'm trying to add the camera.

vacuumboots, 2019-05-19:
can you post output of the following command?

raspistill -v -o test.jpg

Rushmonk, 2019-05-19:
Here you go:

"raspistill" Camera App (commit 7cbfbd38d982 Tainted)

Camera Name ov5647
Width 2592, Height 1944, filename test.jpg
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 5000, Timelapse 0
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Camera component done
Encoder component done
Starting component connection stage
Connecting camera preview port to video render.
Connecting camera stills port to encoder input port
Opening output file test.jpg
Enabling encoder output port
Starting capture -1
Finished capture -1
Closing down
Close down completed, all components disconnected, disabled and destroyed

vacuumboots, 2019-05-19:
Well that shows the raspicam itself is working fine and that the rapsberrypi can identify and use it... I would post my camera settings from my settings.cfg but I'm not sure how the id field is generated, so mine wouldn't work for yours. You might be able to find what is going wrong in /Terrariumpi/log/Terrariumpi.log

Rushmonk, 2019-05-19:
the log file trys to reach the camera, but there is no reply if it was successful or not.

Rushmonk, 2019-05-19:
I've tested the camera with Octoprint, had no issues at all. I've re-installed TerrariumPi, but the camera still don't want to operate.

theyosh, 2019-05-20:
@Rushmonk when you try to save the webcam settings, it does not work. So it looks like there is an error in the screen session. Look at: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi . And see if there are errors.

Or can you make a screenshot of the webcam settings you are using? And make sure the Pi camera can only be used once. So either live, or just shooting a single image.

Rushmonk, 2019-05-22:
@theyosh Thank you for your support. I got it up and running again, I reinstalled everything without the camera. After the first reboot, I attached the camera, rebooted again and updated the raspi-config and rebooted again. It wouldn't took it at the first attempts, but I just kept the window opened in the Browser, re-started the Pi and it was working. Pretty strange behavior, but it up now. 

Another question, you said, either live or shooting a single image, how to I determine that?

vacuumboots, 2019-05-22:
There are two modes that Raspicams can run in (but not usb cameras). 
The "location" of a raspicam can either be "rpicam" or " rpicam_live" but not both at the same time. When the location is rpicam, the camera saves a picture and a jpeg is shown on the website around every 30 seconds. When rpicam_live is chosen, a "live" stream is displayed, as an embedded m3u8. 

You may have to repeat the process you just described when switching between the two, and playing around with the resolution before you get streaming in "live" mode. Here are the settings I have for mine in my settings.cfg that are working for me:

[webcama94df035fc91a158c1f9d7a1c769ca66]
archive = disabled
archivedoor = ignore
archivelight = ignore
id = a94df035fc91a158c1f9d7a1c769ca66
is_live = True
location = rpicam_live
name = rpicam_live
resolution_height = 1080
resolution_width = 1920
rotation = 180

Rushmonk, 2019-05-23:
@vacuumboots Thank you! I'll try to change the settings in the .cfg file, as the procedure is just too complicated while the RPi is already installed and wired. Will post if just changing the settings was successful. Basically I just need to change is_Live to true, possibly also the location.

Rushmonk, 2019-05-23:
Hi,

I've got another question, maybe it's ok to ask. Or should I open a new issue?
The Wiki was a good start, but it's not working as expected.
I've a external Sensor BME280 on a Wemos D1 running ESPEasy. The Json output looks like this:
http://192.168.1.100/json?tasknr=1&view=sensorupdate

{
"TaskValues": [
{"ValueNumber":1,
"Name":"temperature",
"NrDecimals":2,
"Value":22.50
},
{"ValueNumber":2,
"Name":"humidity",
"NrDecimals":2,
"Value":70.80
}]

So I've added the Sensor like this:

http://192.168.1.100/json?tasknr=1&view=sensorupdate#TaskValues/1/Value

But I've got no data, even after 15 min? What am I doing wrong?
Further the Wiki says only one value from a remote source, that means I can not gather the information for Temp and Humidity?

Cheers!

theyosh, 2019-05-27:
@Rushmonk For now it is ok, but for the next question please open a new issue. So we can keep it nice and tidy.

I just tested your JSON and it did work for me. So could it be that the limit values of the sensors are incorrect. Also what does the log show?

Rushmonk, 2019-06-04:
Okay, thank you.

closed 2019-06-04 by Rushmonk