theyosh/TerrariumPI #292: Configurable Light Alarm

Natrixz, 2019-05-20:
Hi Yosh,

Sorry for bugging you;
Is it possible to generate alarms between certain times?
My use-case is that I have a MiFlora Light sensor in use, in case my Lights don't go on (or suddenly go off)
It apparently happens occasionally with my EG-PM2 and a reboot fixes this. (TerrariumPi will show the lights as "on" when they are off)

Right now I manually inspect the dashboard but I would rather have it generate an alarm or have it auto-restart =)
Or maybe go looking for the cause.....but still would be nice to have this option =)

djarrard, 2019-05-22:
Is the state of the switch accurately reflected on the website as being on/off when it shouldn’t be?

If so, you could schedule a script to run that checks the state of the switch via the api and notifies you when it’s not working properly. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Natrixz, 2019-05-22:
No sadly not ☹
It shows it as “on” and even toggling it will not actually toggle it.
Im suspecting it’s losing the connection and doesn’t restore it

From: Dennis J
Sent: Wednesday, 22 May 2019 17:32
To: theyosh/TerrariumPI
Cc: Natrixz; Author
Subject: Re: [theyosh/TerrariumPI] Configurable Light Alarm (#292)

Is the state of the switch accurately reflected on the website as being on/off when it shouldn’t be?
If so, you could schedule a script to run that checks the state of the switch via the api and notifies you when it’s not working properly.
Sent with GitHawk

theyosh, 2019-05-22:
Is it a network version? I do get the feeling that it will not always keep the connection open. So I have a ping running to the  EG-PM2 power switch.

But, I think it should be possible to use the light sensors for this. But I do need some more thinking about this, as the current notification system is probably not suitable for this. So that needs also some extension.

Natrixz, 2019-05-22:
It actually is the USB-version.
Makes it even stranger because it always used to work fine….
It’s not until recently that it’s starting to do this and I just haven’t had the time yet to try and troubleshoot it :(

Might just be an USB-issue, never the less would be nice if we could get notifications from it =)
(might be a nice moment to add more extensions to the notifications)

theyosh, 2019-05-27:
In order to know if a light is on by a sensor, we have to have some threshold values. This can be entered manually, or can be calculated.

So I am not sure how to tackle this. I was thinking of using 24 hour data, pick the min and max in that period. Divide it by 2 get the average. And when the sensor is above it, the light is on. Below is off.

But then, the software needs at least 24 hours of data running the right schema. Could be a bit confusing to setup.

Manually setting the 'average' value could also be an option. But that will also need some data to figure out the right value.

Any idea's from your side about this?

Natrixz, 2019-05-29:
Why not just start with entering it manually?
That way you can specify the threshold yourself, above value X, lights are on.
Below value Y (or maybe even value X) lights are off

After that you/we can always progress into using a calculated value right?

theyosh, 2019-10-01:
Sorry for the late reply, but for now I put it on the wish list...

theyosh, 2023-06-11:
So after some years, I finally had time to add this feature. And it has become pretty basic, almost the same way you requested.

At a light sensor, you can now add a threshold value which indicates the value when the light is on. Or 'on enough'. 
[image]

Enter a value like the average of the on value that is measured. This can be done with every light sensor you have.

You will get a small light bulb indicating that the light threshold is set. And will light up yellow when the light is 'on' according the threshold value.

[image]

Next in the light area you can select the light sensors which are then used for checking if the lights are on when needed. You can select light sensors, or LDR buttons, which are also light sensors.

[image]

When the lights are off based on sensors and should be on, or vise versa, there will be a notification message. 

The notification message can be configured at the notification messages. Use the message type `Lights area wrong state`

[image]

And now you can make you message with the placeholders on the left.

When mutliple light sensors are selected, an average is calculated for all the threshold values and form all the current value. The averages are compared.
And when LDR buttons are used, at least half those sensors should be on to be on for all.

So with a combination of light sensors and LDR buttons, you need to be within the averages of the sensors, and at least the half of the LDR buttons needs to be on.

Is this something you where looking for al those years ;)

closed 2023-08-26 by theyosh