theyosh/TerrariumPI #293: Miflora light sensor too noisy

duenni, 2019-05-23:
Hi,

does anyone else experience a really noisy "average light" sensor from a MiFlora?
[image]
It's really unusable in this state. Is this a problem related to Miflora (is my Miflora faulty and should be replaced?) or ist this some issue with TerrariumPi and how it gets the data?
Also there are two y-scales, one for Lux one for uW/cm^2. When setting the "Limit Max" value in the sensors settings, it seems to correspond to the "UVB" value in the graph. There is also an "UVA" value which is the lower limit. What sense does that make? Why are there two scales?
Thanks!

theyosh, 2019-05-27:
Hmm, 

my problem is that I use existing libraries and code. So I can not tell why this is happening. I do see the same thing:

[image]

But I have also added a UV Index sensor, which uses the values of UVA and UVB and show that value on the graph. Therefore we have two scales and no min and max values. I know this is a bit odd comparing to other graphs, but you can measure lights in different ways,UV and Lux, so I thought this is the best way to show it in a single graph

But as you can see, my UV Index is also spiky. This is because my UV sensors are also spiky. And as I am not able to measure the power over the lines, or bluetooth packages, I must conclude that this is the normal way. I have thought of making the data more smooth, that involves some math which I am not that good in.

So for now, I have accepted this behaviour as expected and have no Idea how to change.

theyosh, 2019-07-01:
Closing due to no more response....

closed 2019-07-01 by theyosh