theyosh/TerrariumPI #297: use for aquarium

kklmn, 2019-06-06:
Hi,

Thanks for your package that looks very attractive and promising!

I am now trying to figure out how to use it for my aquarium. You write that it is applicable not only to terrariums but I don't see where to get away from the terrarium usage pattern. In particular, I don't need doors and sounds and am not sure about weather. But I do need other functions, such as to set off a few switches for a short time (switch off streams for 15 min for feeding), set a week day hourly schedule for lights (I want it on only when I am home in the morning and evening and on weekends) and implement filtration control (by 0-10 V, so far don't know how). Probably scheduling and delayed switching are available, I just don't see how to configure them.

So the main question: is it possible to remove from the web GUI what is not in use, in particular from `Top indicators` and from `Environment`? How about implementing usage scenarios, like 'terrarium', 'aquarium', 'pond' etc in the future?

Smaller issues:
- Prices are a good idea but I think if a unit price is not given, the cost should be hidden. E.g. my water flow is close-cycled, so no cost involved in the flow measurement but I always have to see "€0.00" in the Dashboard.
- Currency should be explicitly opted by the user. Mine is nether $ nor €. It can be inferred from `Weather location` but much easier is just a user string.
- Home/Profile is certainly about a large animal with personality, not about an aquarium or a pond.

If you are not going to look towards aquariums, please advise on where to start with the GUI customization.

Thank you!

Ultameciia, 2019-06-06:
I can't answer all the questions but I can answer a considerable amount of them for you. 

The program is essentially written for Terrariums, however you can use it to control other things. It just doesn't explitly have a "Aquarium Mode". For controlling pumps, devices etc you'd use a relay, meross power switch, or the other smart surge protector he supports. You could even request other relay/switch device to be added. Some things he adds support for. That in itself solves your light issue as well. Connect the lights to a relay or smart surge protector and either set a time schedule or use sunrise/sunset from your weather data. 

As for the GUI he's using [this](https://github.com/theyosh/gentelella) for the GUI. It's modified slightly to serve his needs for TerrariumPi. Without modification to the code there's no way to remove the unused Environment stuff. It'll just show as disabled until something is there like a sensor or lights. 

I agree about the currency stuff. I don't have anything for you there. Profile is definitely for an animal, without editing the code you won't be able to change those fields. However the profile has no bearing on the operation, you don't have to use it.

theyosh, 2019-07-01:
@kklmn Hi, sorry for the late delay. I must amid that the code is probably more suitable for terrarium then an aquarium. As I do not have one currently. So I was hoping that for aquarium use, the somewhat same logic does apply.

I do see some points about zero values like water flow or actual price. I could change the code to hide then when happening. But for now, I do not have a good testing environment (no aquarium running).

Also I do know that a saltwater aquarium needs a lot more tools and measurement, then a normal cold water one. So that is also something I probably will not support well.

It is possible to program power switches on a 24 hour basis. That will just repeat every day. But I think not as specific as you want. That could be part of #231 which is currently on the wishlist.

But on the other hand, I do not see why the profile page is not suitable for your aquarium. It is just a general page. If you are looking for tracking individual animal data, then that is not possible. That will also introduce more logic, and more interfaces.

So, for now, I have to admit that will probably not work well for an aquarium system. And due to the already long wish list, I am afraid that it will take a will to implement aquarium specific logic and features. You could see what is changing in time.. but for now, I am sorry that it is not suitable for your aquarium

kebs19, 2020-05-28:
With all my respect to the developers of Reef pi, that system has a lot of rough edges and it's very unstable. 
And it's not intended to be used on a freshwater aquarium anyway

kklmn, 2020-05-28:
Hello, that was a year ago when I was searching for a ready and good looking software. This is when I asked here. I saw reef-pi too, didn't like it.
Then finally I decided to make it myself to have the functions and the look the way I want.
And I did it: [github.com/kklmn/ArdquariumPi](https://github.com/kklmn/ArdquariumPi). It has been in operation for 6 months now. I see it as reliable and convenient. But this is because I myself made it. Not sure it would be as convenient for everyone.

theyosh, 2020-06-01:
Hi @kklmn 
Nice to see that you are creating your own setup. And a nice approach  to use two system to check each other.

I leave this issue open for the future. I still want to be able to support aquariums. But it does not have a priority right now.

theyosh, 2021-10-15:
Hmm, for now, I have to close it. I do not see time to make it more aquarium friendly. Sorry...

closed 2021-10-15 by theyosh