theyosh/TerrariumPI #298: TerrariumPi autofilling out switches for Wemo devices

furiousg3orge, 2019-06-10:
I am configuring all my devices but I'm noticing that when I reboot the PI that it will pull and populate my wemo switches (for my home) into the switches fields. I haven't set these up or have ever pointed terrariumPI to these switches. Can anyone help me figure this out or, if there is a fix, point me in that direction? My search came up empty.

theyosh, 2019-06-11:
Hi,

this is standard behaviour. My software will scan for certain powerswitches and will add them when found. There is an option to exclude those power switches AFTER you have saved them once: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-exclude-a-power-switch

theyosh, 2019-07-01:
Closing due to no response. So I expect that the excluding option is working

closed 2019-07-01 by theyosh