theyosh/TerrariumPI #299: Cannot change Hardware field of sensor

tvStatic, 2019-06-13:
Once a sensor is created, changing its hardware value to another type appears to have no effect. I changed address at the same time, clicked submit, waited for the operation to complete, then navigated away and back to the Sensor setting page. Address had been updated, but hardware was unchanged. Tried a few times to change Hardware field but was unable to change it.

theyosh, 2019-06-13:
Yes, that is annoying but it is needed :) As the unique ID is generated based on hardware type, and that it cause some internal problems when I do allow changing the hardware type.

A second argument is that you should not change the hardware type that often. Only during installation it could be needed. But once it is running and configured, the hardware is not able to change due to hardware device itself.

So, the only way is to remove the old sensor, and add a new one with the new hardware type.

tvStatic, 2019-06-13:
Understood. This particular use case was for testing the AM2320 sensor (switching it from the Script type that I implemented), though I'd noticed the same behaviour when I chose the incorrect sensor type when setting up as well.

It is a usability issue at the very least, since I was able to select the new type and submit it without any indication that the change wasn't committed. I propose disabling the Hardware type field after the sensor has been created so the user knows explicitly that it cannot be changed. I doubt I'll be the last person to come across this issue otherwise. :)

theyosh, 2019-06-26:
Changed a bit and merged it. Also added for switches and doors. Webcams do not have a hardware pulldown. So should be fixed

tvStatic, 2019-06-29:
Verified.

closed 2019-06-29 by tvStatic

theyosh, 2019-07-01:
Hi, this caused a bug with #307. So I had to change this code.... is it still working for you?

tvStatic, 2019-07-08:
Upgraded to latest, and reloaded UI disabling cache. Still looks ok to me.