theyosh/TerrariumPI #300: sensor by calculate difference of 2 sensor

funware84, 2019-06-16:
i need to switch on a pump if there is a diffenrece in temperature mor than x°C.

can u add a sensor which can be calculatet like, sensor 1 - or + sensor 2 
this i'd like to use in enviroment to toogle a pump on.

hope u can add this

theyosh, 2019-06-20:
Hmm, to get it right. Both temperature sensors can both go up and down in value, but as long as the are within the X degree difference it is all ok? So there is no min and max value, but a difference value between sensors... 

That is interesting, but not possible at the moment. And I need to think about this, as this makes everything more complex. Also I need to think how to create the screen for this. As the X amount of degrees should be entered somewhere.

funware84, 2019-06-21:
yes u are right,
have tried to find a way with the new script sensor,
i wasnt able to do becouse i am not a programmer

funware84, 2019-06-24:
For now i'd be lucky if someone ca post a Skript.
So i can add this Skript as a temp sensor.
Temp regulation will be needed if it s getting cold again. 
Right now i nees the regulation of differenz

theyosh, 2019-06-26:
Hmm, I think I can make something for you... give me some time ( few days)

theyosh, 2019-07-01:
Added a script in the '[contrib](https://github.com/theyosh/TerrariumPI/blob/master/contrib/sensors_diff.py)' folder. Do a git pull to get the latest code.

You should copy the file to a different name, so it will not conflict with updates. Then you have to change the two IDs in the 'SENSORS' variable. The comments in the code should be enough to figure those out.

Then add it as a type of sensor you want in the sensors, and it should give you the current difference as the actual value. Then you can work with min and max values to trigger alarms. And then use that sensor somewhere in the environment part.

theyosh, 2019-07-01:
Ah, and that will cost you some pictures!! :) #210

theyosh, 2019-07-01:
Update, rerun the installer after the update!! Else it will not start

theyosh, 2019-07-05:
Ok, have you been able to test this? At least have you created a new temperature sensor based on the script? So that you have a graph showing the actual temperature difference?

funware84, 2019-07-06:
Thanks a lot.
Yesterday i have set it up.
Works fine
I will Posr somw pics soon

theyosh, 2019-07-28:
Closing due to implemented

closed 2019-07-28 by theyosh