theyosh/TerrariumPI #306: Reboot pi from dashboard

funware84, 2019-06-24:
Is there a posibility to add an option to reboot the pi using the ui?

theyosh, 2019-07-01:
Ok, done! Do a `git pull` and then rerun the installer according to the https://github.com/theyosh/TerrariumPI#updating .

This will give you a reboot button on the system settings page. You will have to confirm the action, and you need to be authenticated to make the call to the backend.

This is an early version. The next version will have some kind of overlay, which will try to reconnect to the server while it is rebooting. And reload when the server is back again.

In the meantime, could you look at: #210 :)

theyosh, 2019-07-01:
Update, rerun the installer after the update!! Else it will not start

tvStatic, 2019-07-04:
Would it be possible to add an option to shutdown the Pi in the same place?

theyosh, 2019-07-05:
Done! And we have a fancy reboot screen ;)

theyosh, 2019-07-28:
Closing due to implemented

closed 2019-07-28 by theyosh